PX4/PX4-SITL_gazebo-classic #536: Error about nested models regarding gps0_joint

julianoes, 2020-06-26:
I get the following error on `gzserver` startup with `iris`:

```
[gzserver    ] [Err] [Model.cc:99] Error Code 16 Msg: Nested models are not yet supported by DOM objects, skipping model [iris].
[gzserver    ] [Err] [Model.cc:99] Error Code 18 Msg: Child link with name[gps0::link] specified by joint with name[gps0_joint] not found in model with name[iris].
[gzserver    ] [Err] [Model.cc:99] Error Code 23 Msg: FrameAttachedToGraph error, Non-LINK vertex with name [gps0_joint] is disconnected; it should have 1 outgoing edge in MODEL attached_to graph.
[gzserver    ] [Err] [Model.cc:99] Error Code 23 Msg: Graph has __model__ scope but sink vertex named [gps0_joint] does not have FrameType LINK when starting from vertex with name [gps0_joint].
[gzserver    ] [Err] [Model.cc:99] Error Code 18 Msg: Child link with name[gps0::link] specified by joint with name[gps0_joint] not found in model with name[iris].
[gzserver    ] [Err] [Model.cc:99] Error Code 26 Msg: PoseRelativeToGraph error, Non-MODEL vertex with name [gps0_joint] is disconnected; it should have 1 incoming edge in MODEL relative_to graph.
[gzserver    ] [Err] [Model.cc:99] Error Code 26 Msg: PoseRelativeToGraph frame with name [gps0_joint] is disconnected; its source vertex has name [gps0_joint], but its source name should be __model__.
[gzserver    ] [Err] [Model.cc:99] Error Code 8 Msg: A <script> element is missing a child <uri> element, or the <uri> element is empty.
```

FYI @TSC21 @Jaeyoung-Lim

TSC21, 2020-06-26:
You are using Gazebo 11 right?

julianoes, 2020-06-26:
Yes.

TSC21, 2020-06-26:
Then I would say that a solution needs to be find in the OSRF side to fix the fact `Nested models are not yet supported by DOM objects`. Nested models have been supported on DOM since a long time, so I don't get why is this a problem now. Have you looked in the gazebo issues for something similar?

bperseghetti, 2020-06-27:
@julianoes that was the initial issue I had posted till I changed the issue to be about the R1 since it evolved more into an R1 issue for us.

bperseghetti, 2020-06-27:
@julianoes @TSC21 Note The current version of libsdformat's DOM API does not support models defined directly inside parent models. As a workaround, such models can be accessed using the Element API. http://sdformat.org/tutorials?tut=composition

@TSC21 do we even need a nest GPS model, what previous problem does it exactly solve (just curious for background understanding)? 

Side note going forward it appears this might be an issue no matter what for any future gazebo or ignition sims so figuring out what works best for current versions of SDF I think is critical.

bperseghetti, 2020-06-27:
Also looks like work on it is added as a milestone for SDF v1.8, not sure if it will be released ever to v1.7. https://github.com/osrf/sdformat/issues/283

TSC21, 2020-06-27:
> @TSC21 do we even need a nest GPS model, what previous problem does it exactly solve (just curious for background understanding)?

https://github.com/PX4/sitl_gazebo/pull/517

Jaeyoung-Lim, 2020-08-13:
I can reproduce this on Ubuntu Focal, Gazebo 11.0.0

Jaeyoung-Lim, 2020-09-16:
Seems like https://github.com/osrf/gazebo/pull/2824 is relent to the discussion here

Jaeyoung-Lim, 2020-10-05:
@TSC21 Is there a reason why we are using nested models verses normal <sensor> tags for multiple gps models?

TSC21, 2020-10-05:
> normal tags

What do you mean with normal tags?

Jaeyoung-Lim, 2020-10-05:
@TSC21 Instead of 
```
    <include>
      <uri>model://gps</uri>
      <pose>0 0 0 0 0 0</pose>
      <name>gps</name>
    </include>
    <joint name='gps_joint' type='fixed'>
      <child>gps::link</child>
      <parent>base_link</parent>
    </joint>
```
we can do 
```
      <link name='link'>
        <pose>0 0 0 0 0 0</pose>
        <inertial>
          <pose>0 0 0 0 0 0</pose>
          <mass>0.01</mass>
          <inertia>
            <ixx>2.1733e-06</ixx>
            <ixy>0</ixy>
            <ixz>0</ixz>
            <iyy>2.1733e-06</iyy>
            <iyz>0</iyz>
            <izz>1.8e-07</izz>
          </inertia>
        </inertial>
        <visual name='visual'>
          <geometry>
            <cylinder>
              <radius>0.01</radius>
              <length>0.002</length>
            </cylinder>
          </geometry>
          <material>
            <script>
              <name>Gazebo/Black</name>
              <uri>__default__</uri>
            </script>
          </material>
        </visual>
        <sensor name='gps' type='gps'>
          <pose>0 0 0 0 0 0</pose>
          <plugin name='gps_plugin' filename='libgazebo_gps_plugin.so'>
            <robotNamespace/>
            <gpsNoise>1</gpsNoise>
            <gpsXYRandomWalk>2.0</gpsXYRandomWalk>
            <gpsZRandomWalk>4.0</gpsZRandomWalk>
            <gpsXYNoiseDensity>0.0002</gpsXYNoiseDensity>
            <gpsZNoiseDensity>0.0004</gpsZNoiseDensity>
            <gpsVXYNoiseDensity>0.2</gpsVXYNoiseDensity>
            <gpsVZNoiseDensity>0.4</gpsVZNoiseDensity>
          </plugin>
        </sensor>
      </link>
    <joint name='gps0_joint' type='fixed'>
      <parent>base_link</parent>
      <child>gps0::link</child>
    </joint>
```
This way we don't need to deal with the problems arising from nested models not being supported.

TSC21, 2020-10-05:
> @TSC21 Instead of
>
> ```
>     <include>
>       <uri>model://gps</uri>
>       <pose>0 0 0 0 0 0</pose>
>       <name>gps</name>
>     </include>
>     <joint name='gps_joint' type='fixed'> [...]

That's true, but what exactly is not supported? The `include` tags?

TSC21, 2020-10-05:
The problem with the above is that you need to repeat it to each sensor you add, which is really painful and starts to grow in a crazy way.

Jaeyoung-Lim, 2020-10-05:
> That's true, but what exactly is not supported? The include tags?

The include is supported, but nested models are not.

TSC21, 2020-10-05:
> > That's true, but what exactly is not supported? The include tags?
> 
> The include is supported, but nested models are not, as written in the error:
> 
> ```
> [gzserver    ] [Err] [Model.cc:99] Error Code 16 Msg: Nested models are not yet supported by DOM objects, skipping model [iris].
> ```

Since we removed the need for Xacro and replaced with jinja, we can now use `include` everywhere.

Jaeyoung-Lim, 2020-10-05:
@TSC21 The error @julianoes was reporting is not related to xacro. Its related to the sdfdom not supported nested models

Jaeyoung-Lim, 2020-10-06:
With discussion with @TSC21 this issue should have been fixed when transitioning to sdf templates.

@julianoes Please reopen if the issue persists

closed 2020-10-06 by Jaeyoung-Lim