PX4/PX4-SITL_gazebo-classic #538: Directional (yaw) stability without a vertical stabilizer

lum-supakorn, 2020-07-01:
I've been working on a simulation of a UAV which flies like a flying wing without any vertical stabilizer for directional stability.
With a virtual rudder (a `LiftDragPlugin` wing set up in a vertical configuration -- see the SITL `tailsitter` model for reference [here](https://github.com/PX4/sitl_gazebo/blob/1af7e29dbb1ecce7b0b191c9deb24ab1f13916ab/models/tailsitter/tailsitter.sdf#L545)), the vehicle is able to maintain satisfactory directional stability. However, the actual vehicle does this with differential thrust.

I have enabled `VT_FW_DIFTHR_EN`, set `VT_FW_DIFTHR_SC` from `0.3` to `0.1`, and removed the virtual elevator, but the vehicle did not seem to correct for yaw with differential thrust. In fact, there seemed to be no directional control at all. When the heading error got too large, the vehicle seemed to roll to correct for that instead of keeping the yaw error close to zero at all times. Loitering could not be achieved at all. Here is the [flight log](https://review.px4.io/plot_app?log=a79233a2-ac9c-4820-bc4e-b5be9047e9ca).

Is differential thrust in PX4 controller capable of maintaining yaw stability?

Jaeyoung-Lim, 2020-07-05:
@lum-supakorn I guess we need to define this in a mixer to be able to do this.

Jaeyoung-Lim, 2020-10-05:
@lum-supakorn Any updates?

Jaeyoung-Lim, 2020-10-06:
Closing, since the original author didn't come back.

Please reopen if this is still relevant @lum-supakorn

closed 2020-10-06 by Jaeyoung-Lim