PX4/PX4-SITL_gazebo-classic #539: Unable to integrate IMU into custom model

iwishiwasaneagle, 2020-07-03:
I'm trying to create a custom model for the Skywalker X8 using sitl_gazebo. However, the IMU isn't working.

The error message when trying to view the `gazebo_imi_plugin` in Gazebo: 
```
Topic [/data://world/default/model/skywalker/plugin/gazebo_imu_plugin/joint_cmd] is not valid.
[Err] [JointControlWidget.cc:393] Error advertising topic [/data://world/default/model/skywalker/plugin/gazebo_imu_plugin/joint_cmd]
```
The error in the PX4 logs:
```
WARN  [PreFlightCheck] Preflight Fail: Accel Range, hold still on arming
WARN  [PreFlightCheck] Primary accelerometer not found
```

I have essentially copied/followed the `plane.sdf` and I don't see any discrepancies. 

imu sections of the .sdf file:
```
<link name='skywalker/imu_link'>
      <pose>0 0 0 0 0 0</pose>
      <inertial>
        <pose>0 0 0 0 0 0</pose>
        <mass>0.015</mass>
        <inertia>
          <ixx>1e-05</ixx>
          <ixy>0</ixy>
          <ixz>0</ixz>
          <iyy>1e-05</iyy>
          <iyz>0</iyz>
          <izz>1e-05</izz>
        </inertia>
      </inertial>
    </link>
    <joint name='skywalker/imu_joint' type='revolute'>
      <child>skywalker/imu_link</child>
      <parent>base_link</parent>
      <axis>
        <xyz>1 0 0</xyz>
        <limit>
          <lower>0</lower>
          <upper>0</upper>
          <effort>0</effort>
          <velocity>0</velocity>
        </limit>
        <dynamics>
          <spring_reference>0</spring_reference>
          <spring_stiffness>0</spring_stiffness>
        </dynamics>
        <use_parent_model_frame>1</use_parent_model_frame>
      </axis>
    </joint>
```
```
    <plugin name='gazebo_imu_plugin' filename='libgazebo_imu_plugin.so'>
      <robotNamespace></robotNamespace>
      <linkName>skywalker/imu_link</linkName>
      <imuTopic>/imu</imuTopic>
      <gyroscopeNoiseDensity>0.0003394</gyroscopeNoiseDensity>
      <gyroscopeRandomWalk>3.8785e-05</gyroscopeRandomWalk>
      <gyroscopeBiasCorrelationTime>1000.0</gyroscopeBiasCorrelationTime>
      <gyroscopeTurnOnBiasSigma>0.0087</gyroscopeTurnOnBiasSigma>
      <accelerometerNoiseDensity>0.004</accelerometerNoiseDensity>
      <accelerometerRandomWalk>0.006</accelerometerRandomWalk>
      <accelerometerBiasCorrelationTime>300.0</accelerometerBiasCorrelationTime>
      <accelerometerTurnOnBiasSigma>0.196</accelerometerTurnOnBiasSigma>
    </plugin>
```
```
<plugin name='mavlink_interface' filename='libgazebo_mavlink_interface.so'>
      <robotNamespace></robotNamespace
      <imuSubTopic>/imu</imuSubTopic>
      <magSubTopic>/mag</magSubTopic>
      <baroSubTopic>/baro</baroSubTopic>
      <mavlink_addr>INADDR_ANY</mavlink_addr>
      <mavlink_udp_port>14560</mavlink_udp_port>
...
```

Jaeyoung-Lim, 2020-07-05:
@iwishiwasaneagle Are you sure you are using the latest master of the firmware and sitl_gazebo that is in sync?

iwishiwasaneagle, 2020-07-05:
> @iwishiwasaneagle Are you sure you are using the the firmware and sitl_gazebo that is in sync?

Yes, it's the latest PX4/Firmware with the only additions being the custom sdf file, config file, meshes and relevant changes to be able to run `make px4_sitl gazebo_skywalker`

Jaeyoung-Lim, 2020-07-05:
Looks like there is something wrong with your model name / topic path and `gazebo_mavlink_interface` is unable to receive the correct messages

iwishiwasaneagle, 2020-07-05:
> > @iwishiwasaneagle Are you sure you are using the the firmware and sitl_gazebo that is in sync?
> 
> Yes, it's the latest PX4/Firmware with the only additions being the custom sdf file, config file, meshes and relevant changes to be able to run `make px4_sitl gazebo_skywalker`

I have basically copy/pasted the `plane.sdf` file for IMU and body/link naming so I'm not sure where else to look

Jaeyoung-Lim, 2020-07-05:
@iwishiwasaneagle that doesn't mean that you did everything correctly. Why don't you try to reverse engeineer what went wrong by slowly increasing the diff with `plane.sdf` ?

iwishiwasaneagle, 2020-07-05:
> @iwishiwasaneagle that doesn't mean that you did everything correctly. Why don't you try to reverse engeineer what went wrong by slowly increasing the diff with `plane.sdf` ?

Okay I'll try that tomorrow and report back. Thanks anyway @Jaeyoung-Lim!

closed 2020-07-05 by iwishiwasaneagle