PX4/PX4-SITL_gazebo-classic #540: Controlling gimbals for multiple vehicles using ROS node in SITL

Jaroan, 2020-07-05:
I want to control the gimbal of two Typhoon H480 models in an offboard SITL setup using the topic /mavros/mount_control/command. I have understood how to implement it for a single model with help from this repo [Jaeyoung-Lim/mavros_humantracking](https://github.com/Jaeyoung-Lim/mavros_humantracking). My implementation : [Gif Link](https://imgur.com/gallery/qMO0CsH)
For multi-UAV simulation, I tried to extend using the appropriate launch file and corresponding topics but I was not able to control the gimbals. They seemed to be unsteady as well. [Video](https://imgur.com/gallery/qMO0CsH)
I am unable to move even one vehicle’s gimbal for a multi UAV simulation, like in the case of the single-vehicle. Any help and suggestion will be appreciated. 

I tried to implement the changes given in the pull request - https://github.com/PX4/sitl_gazebo/pull/469 but I didn’t find a difference in the jitter. Is there something I’m skipping? I've also raised this issue in a Forum topic https://discuss.px4.io/t/controlling-gimbal-for-multiple-vehicles-using-ros-node-in-sitl/17415?u=jaroan :slightly_frowning_face:

Jaeyoung-Lim, 2020-07-05:
@Jaroan The gimbal mavlink messages are being sent to the same udp port. You need to asign different ports for each vehicle as done with the xacro macros

Jaroan, 2020-07-05:
I have the following in the Typhoon SDF File. Where do I set it for multiple vehicles? I apologise for the silly doubt
```
   <plugin name='mavlink_interface' filename='libgazebo_mavlink_interface.so'>
      <robotNamespace></robotNamespace>
      <imuSubTopic>/imu</imuSubTopic>
      <gpsSubTopic>/gps</gpsSubTopic>
      <magSubTopic>/mag</magSubTopic>
      <baroSubTopic>/baro</baroSubTopic>
      <mavlink_addr>INADDR_ANY</mavlink_addr>
      <mavlink_udp_port>14560</mavlink_udp_port>
      <serialEnabled>false</serialEnabled>
      <serialDevice>/dev/ttyACM0</serialDevice>
      <baudRate>921600</baudRate>
      <qgc_addr>INADDR_ANY</qgc_addr>
      <qgc_udp_port>14550</qgc_udp_port>
      <sdk_addr>INADDR_ANY</sdk_addr>
      <sdk_udp_port>14540</sdk_udp_port>
      <hil_mode>false</hil_mode>
      <hil_state_level>false</hil_state_level>
      <enable_lockstep>false</enable_lockstep>
      <use_tcp>true</use_tcp>
      <motorSpeedCommandPubTopic>/gazebo/command/motor_speed</motorSpeedCommandPubTopic>
      <control_channels>
        <channel name="rotor0">
          <input_index>0</input_index>
          <input_offset>0</input_offset>
          <input_scaling>1500</input_scaling>
          <zero_position_disarmed>0</zero_position_disarmed>
          <zero_position_armed>100</zero_position_armed>
          <joint_control_type>velocity</joint_control_type>

        </channel>
        <channel name="rotor1">
          <input_index>1</input_index>
          <input_offset>0</input_offset>
          <input_scaling>1500</input_scaling>
          <zero_position_disarmed>0</zero_position_disarmed>
          <zero_position_armed>100</zero_position_armed>
          <joint_control_type>velocity</joint_control_type>
        </channel>
        <channel name="rotor2">
          <input_index>2</input_index>
          <input_offset>0</input_offset>
          <input_scaling>1500</input_scaling>
          <zero_position_disarmed>0</zero_position_disarmed>
          <zero_position_armed>100</zero_position_armed>
          <joint_control_type>velocity</joint_control_type>
        </channel>
        <channel name="rotor3">
          <input_index>3</input_index>
          <input_offset>0</input_offset>
          <input_scaling>1500</input_scaling>
          <zero_position_disarmed>0</zero_position_disarmed>
          <zero_position_armed>100</zero_position_armed>
          <joint_control_type>velocity</joint_control_type>
        </channel>
        <channel name="rotor4">
          <input_index>4</input_index>
          <input_offset>0</input_offset>
          <input_scaling>1500</input_scaling>
          <zero_position_disarmed>0</zero_position_disarmed>
          <zero_position_armed>100</zero_position_armed>
          <joint_control_type>velocity</joint_control_type>
        </channel>
       <channel name="rotor5">
          <input_index>5</input_index>
          <input_offset>0</input_offset>
          <input_scaling>1500</input_scaling>
          <zero_position_disarmed>0</zero_position_disarmed>
          <zero_position_armed>100</zero_position_armed>
          <joint_control_type>velocity</joint_control_type>
        </channel>
        <channel name="gimbal_roll">
          <input_index>6</input_index>
          <input_offset>0</input_offset>
          <input_scaling>-3.1415</input_scaling>
          <zero_position_disarmed>0</zero_position_disarmed>
          <zero_position_armed>0</zero_position_armed>
          <joint_control_type>position_gztopic</joint_control_type>
          <gztopic>/gimbal_roll_cmd</gztopic>
          <joint_name>typhoon_h480::cgo3_camera_joint</joint_name>
        </channel>
        <channel name="gimbal_pitch">
          <input_index>7</input_index>
          <input_offset>0</input_offset>
          <input_scaling>-3.1415</input_scaling>
          <zero_position_disarmed>0</zero_position_disarmed>
          <zero_position_armed>0</zero_position_armed>
          <joint_control_type>position_gztopic</joint_control_type>
          <gztopic>/gimbal_pitch_cmd</gztopic>
          <joint_name>typhoon_h480::cgo3_camera_joint</joint_name>
        </channel>
        <channel name="gimbal_yaw">
          <input_index>8</input_index>
          <input_offset>0</input_offset>
          <input_scaling>-3.1415</input_scaling>
          <zero_position_disarmed>0</zero_position_disarmed>
          <zero_position_armed>0</zero_position_armed>
          <joint_control_type>position_gztopic</joint_control_type>
          <gztopic>/gimbal_yaw_cmd</gztopic>
          <joint_name>typhoon_h480::cgo3_vertical_arm_joint</joint_name>
        </channel>
        <channel name="left_leg">
          <input_index>9</input_index>
          <input_offset>1</input_offset>
          <input_scaling>0.5</input_scaling>
          <zero_position_disarmed>0</zero_position_disarmed>
          <zero_position_armed>0</zero_position_armed>
          <joint_control_type>position</joint_control_type>
          <joint_control_pid>
            <p>3.5</p>
            <i>0.5</i>
            <d>0</d>
            <iMax>4</iMax>
            <iMin>-4</iMin>
            <cmdMax>6</cmdMax>
            <cmdMin>-6</cmdMin>
          </joint_control_pid>
          <joint_name>left_leg_joint</joint_name>
        </channel>
        <channel name="right_leg">
          <input_index>10</input_index>
          <input_offset>1</input_offset>
          <input_scaling>0.5</input_scaling>
          <zero_position_disarmed>0</zero_position_disarmed>
          <zero_position_armed>0</zero_position_armed>
          <joint_control_type>position</joint_control_type>
          <joint_control_pid>
            <p>3.5</p>
            <i>0.5</i>
            <d>0</d>
            <iMax>4</iMax>
            <iMin>-4</iMin>
            <cmdMax>6</cmdMax>
            <cmdMin>-6</cmdMin>
          </joint_control_pid>
          <joint_name>right_leg_joint</joint_name>
        </channel>
      </control_channels>
    </plugin>
```

Jaeyoung-Lim, 2020-07-05:
@Jaroan https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d-posix/6011_typhoon_h480.post#L10 This is where the mount orientation mavlink is being streamed. How are you specifying the sysID for each of the vehicles?

Jaroan, 2020-07-05:
Oh, I see. I am using _multi_uav_mavros_sitl_sdf.launch_ file with the command `roslaunch px4 multi_uav_mavros_sitl_sdf.launch vehicle:=typhoon_h480
` to spawn two typhoon_h480s

Jaeyoung-Lim, 2020-07-05:
And do they have different IDs? In this case it should have different ports assigned. You might see it in the shell which one is used

Jaroan, 2020-07-05:
I have provided different IDs in the launch file:

```
    <!-- UAV0 -->
    <group ns="uav0">
        <!-- MAVROS and vehicle configs -->
        <arg name="ID" value="0"/>
        <arg name="fcu_url" default="udp://:14540@localhost:14580"/>
        <!-- PX4 SITL and vehicle spawn -->
        <include file="$(find px4)/launch/single_vehicle_spawn_sdf.launch">
            <arg name="x" value="0"/>
            <arg name="y" value="0"/>
            <arg name="z" value="0"/>
            <arg name="R" value="0"/>
            <arg name="P" value="0"/>
            <arg name="Y" value="0"/>
            <arg name="vehicle" value="$(arg vehicle)"/>
            <arg name="mavlink_udp_port" value="14560"/>
            <arg name="mavlink_tcp_port" value="4560"/>
            <arg name="ID" value="$(arg ID)"/>
        </include>
        <!-- MAVROS -->
        <include file="$(find mavros)/launch/px4.launch">
            <arg name="fcu_url" value="$(arg fcu_url)"/>
            <arg name="gcs_url" value=""/>
            <arg name="tgt_system" value="$(eval 1 + arg('ID'))"/>
            <arg name="tgt_component" value="1"/>
        </include>
    </group>
    <!-- UAV1 -->
    <group ns="uav1">
        <!-- MAVROS and vehicle configs -->
        <arg name="ID" value="1"/>
        <arg name="fcu_url" default="udp://:14541@localhost:14581"/>
        <!-- PX4 SITL and vehicle spawn -->
        <include file="$(find px4)/launch/single_vehicle_spawn_sdf.launch">
            <arg name="x" value="1"/>
            <arg name="y" value="0"/>
            <arg name="z" value="0"/>
            <arg name="R" value="0"/>
            <arg name="P" value="0"/>
            <arg name="Y" value="0"/>
            <arg name="vehicle" value="$(arg vehicle)"/>
            <arg name="mavlink_udp_port" value="14561"/>
            <arg name="mavlink_tcp_port" value="4561"/>
            <arg name="ID" value="$(arg ID)"/>
        </include>
        <!-- MAVROS -->
        <include file="$(find mavros)/launch/px4.launch">
            <arg name="fcu_url" value="$(arg fcu_url)"/>
            <arg name="gcs_url" value=""/>
            <arg name="tgt_system" value="$(eval 1 + arg('ID'))"/>
            <arg name="tgt_component" value="1"/>
        </include>
    </group>
```

Jaeyoung-Lim, 2020-07-05:
Okay, you would need to figure out exactly where it went wrong.

The UDP / TCP ports are configured [here](https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d-posix/rcS#L110)

Jaroan, 2020-07-05:
@Jaeyoung-Lim On running netstat, I am able to see all the connections for TCP based on the launch are established as below. the UDP connections also are listed
```
$ netstat -tn | grep 4560
tcp       93     76 127.0.0.1:4560          127.0.0.1:33146         ESTABLISHED
tcp        0      0 127.0.0.1:33146         127.0.0.1:4560          ESTABLISHED
$ netstat -tn | grep 4561
tcp        0      0 127.0.0.1:33234         127.0.0.1:4561          ESTABLISHED
tcp       93      0 127.0.0.1:4561          127.0.0.1:33234         ESTABLISHED

$ netstat -au
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State      
udp        0      0 localhost:domain        0.0.0.0:*                                                    
udp        0      0 0.0.0.0:ipp             0.0.0.0:*                          
udp        0      0 0.0.0.0:14530           0.0.0.0:*                          
udp        0      0 0.0.0.0:14540           0.0.0.0:*                          
udp        0      0 0.0.0.0:14541           0.0.0.0:*                          
udp        0      0 0.0.0.0:14558           0.0.0.0:*                          
udp        0      0 0.0.0.0:14570           0.0.0.0:*                          
udp        0      0 0.0.0.0:14571           0.0.0.0:*                          
udp        0      0 0.0.0.0:14580           0.0.0.0:*                          
udp        0      0 0.0.0.0:14581           0.0.0.0:*                          
```
Is there any checks to ensure the MOUNT_ORIENTATION is set to the right ports based on this: https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d-posix/6011_typhoon_h480.post#L10

Jaeyoung-Lim, 2020-07-05:
Looks like the ports are correct. 

If it is not the ports, you need to find out where this is coming from

Question: does this work for a single vehicle?

Jaroan, 2020-07-05:
Yes it works for a single vehicle here is a video https://imgur.com/gallery/qMO0CsH
Here is the code snippet using a controller class object which instantiate mount control message as 
`	self.mount_control = MountControl()`
````
    # Single Quad Mount Command
    mountCnt = rospy.Publisher('/mavros/mount_control/command', MountControl, queue_size=10)

    gimbal_pitch_ = 0.0
    gimbal_yaw_ = 0.0
    gimbal_roll_ = 0.0
.
.
.
    while not rospy.is_shutdown():

	object.mount_control.header.stamp = rospy.Time.now()
	object.mount_control.header.frame_id = "map"
	object.mount_control.mode = 2
	object.mount_control.pitch = gimbal_pitch_
	object.mount_control.roll = gimbal_roll_
	if i%20 == 0:
		gimbal_yaw_ += 10
		print("yaw increased to", gimbal_yaw_)
	object.mount_control.yaw = gimbal_yaw_
	mountCnt.publish(object.mount_control)
	i+=1
        rate.sleep()
```

Jaeyoung-Lim, 2020-07-05:
@Jaroan Are you sure `/mavros/mount_control/command` is the right topic to publish? Each drone should have a different topic to receive this right?

Jaroan, 2020-07-05:
I am able to successfully arm, and create setpoints and takeoff both vehicles. The similar code for multiple vehicles using the appropriate namespace for topics e.g. uav0/mavros... and uav1/mavros... doesn't control the gimbal.

However running `rostopic echo uav0/mavros/mount_control/command` shows the yaw that is published correctly but no results on the vehicle as seen on Gazebo or Rviz image topic.

Jaeyoung-Lim, 2020-07-05:
Can you check if the mount control messages are received on the firmware side?

Jaeyoung-Lim, 2020-07-05:
@Jaroan Now I see the problem

You need separate sdf files for each typhoon_h480 which matches the tcp and udp ports

Jaroan, 2020-07-05:
Oh ok, I have made a copy of the typhoon_h480 sdf file and modified lines to the appropriate udp port https://github.com/PX4/sitl_gazebo/blob/1af7e29dbb1ecce7b0b191c9deb24ab1f13916ab/models/typhoon_h480/typhoon_h480.sdf#L1283
as well as     `  <qgc_udp_port>14551</qgc_udp_port>`
and      ` <sdk_udp_port>14541</sdk_udp_port>`

The single_vehicle_spawn_sdf.launch launch file was also duplicated appropriately to include the new sdf file and changes made to the multi_uav_mavros_sitl_sdf.launch file.

Do I need a separate startup file in Firmware/ROMFS/px4fmu_common/init.d-posix/ directory as well?

Jaeyoung-Lim, 2020-07-05:
@Jaroan No, you just need two different sdf files and not duplicate the launch file. No, the statupfile that is being used should not be duplicated. Please look at other examples for multivehicle spawning as written in https://dev.px4.io/master/en/simulation/multi_vehicle_simulation_gazebo.html

Jaroan, 2020-07-06:
@Jaeyoung-Lim  would I need to copy the model as well?

The instructions at https://dev.px4.io/master/en/simulation/multi_vehicle_simulation_gazebo.html#multiple-vehicles-using-sdf-models don't say how to incorporate different sdf files in the same model. 
It says the SITL/Gazebo port number is automatically inserted by xmstarlet for each spawned vehicle, and does not need to be specified in the SDF file.
The `multi_uav_mavros_sitl_sdf.launch` file calls the `single_vehicle_spawn_sdf.launch` https://github.com/PX4/Firmware/blob/master/launch/multi_uav_mavros_sitl_sdf.launch#L28
and vehicle is set only once
https://github.com/PX4/Firmware/blob/master/launch/multi_uav_mavros_sitl_sdf.launch#L61

It also says to add a new vehicle, modify the `single_vehicle_spawn_sdf.launch` file to point to the location of your model by changing the line below to point to your model:
`$(find px4)/Tools/sitl_gazebo/models/$(arg vehicle)/$(arg vehicle).sdf`
https://github.com/PX4/Firmware/blob/master/launch/single_vehicle_spawn_sdf.launch#L23
Wouldn't the `single_vehicle_spawn_sdf.launch` point to the same sdf both the times?

Jaeyoung-Lim, 2020-07-07:
Right, I think you need to verify where it is broken, including your code

Jaroan, 2020-07-08:
I rebuilt the latest Firmware from master. The gimbal of one vehicle out of two is controllable and follows the mount_contol messages.

Jaeyoung-Lim, 2020-07-08:
@Jaroan Cool, I guess you need to figure out what is wrong with instance 2. Maybe the first thing I would check is that your code is sending the control messages correctly and it reaches the firmware

swaroophangal, 2020-07-11:
Hello @Jaroan ,

IMHO the only changes that would be required would be, to add the following lines to `single_vehicle_spawn_sdf.launch` after [https://github.com/PX4/Firmware/blob/aec1b9243a90dd1b419745a3b086bbce07385ab9/launch/single_vehicle_spawn_sdf.launch#L23](https://github.com/PX4/Firmware/blob/aec1b9243a90dd1b419745a3b086bbce07385ab9/launch/single_vehicle_spawn_sdf.launch#L23) :

'<arg name="cmd" default="xmlstarlet ed -d '//plugin[@name=&quot;mavlink_interface&quot;]/mavlink_udp_port' -s '//plugin[@name=&quot;mavlink_interface&quot;]' -t elem -n mavlink_udp_port -v $(arg mavlink_udp_port) $(find px4)/Tools/sitl_gazebo/models/$(arg vehicle)/$(arg vehicle).sdf"/>
    <arg name="cmd" default="xmlstarlet ed -d '//plugin[@name=&quot;GstCameraPlugin&quot;]/udpPort' -s '//plugin[@name=&quot;GstCameraPlugin&quot;]/udpPort' -t elem -n udpPort -v $(arg video_udp_port) $(find px4)/Tools/sitl_gazebo/models/$(arg vehicle)/$(arg vehicle).sdf"/>`

I am stuck at how do I concatenate these xmlstarlet commands before the corrected xml being sent to spawn_model. 

@Jaeyoung-Lim please correct me if I am wrong.

Also, I am facing a weird issue since a few days. GPS messages are not received by PX4 SITL from gazebo. On commander gps status returns not running, while trying to start returns an error as `ERROR [gps] ERR: -1 (tcsetattr)`

Don't know my config issue or something else. I tried recloning Firmware multiple times.

Jaeyoung-Lim, 2020-07-11:
@swaroophangal When you describe this 'weird' error since a few days, what changed in these "few days"?

swaroophangal, 2020-07-11:
@Jaeyoung-Lim  I tried running sitl+iris model with just gazebo and it works. Just the running with ROS part does not work. Anyways, will open another issue for this. Sorry for hijacking the main issue.
About the multi-vehicle sitl with ROS+Gazebo, am I right in my approach?

Jaeyoung-Lim, 2020-07-12:
@swaroophangal That looks more or less correct, but you also need to stream MOUNT_ORIENTATION messages as done in https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d-posix/6011_typhoon_h480.post#L10

@Jaroan  It seems that the `single_vehicle_launch_sdf` starts the rcS directly, making the typhoon_h480.post not run as part the startup script. (therefore no gimbal controls when launching two)

Jaroan, 2020-07-13:
@swaroophangal Thank you for the suggestion, did you figure out how to concatenate these commands as the arg 'cmd' has already been declared in the launch file for `mavlink_tcp_port` and is not allowing the additional lines. The following is the error

```
RLException: while processing /src/Firmware/launch/single_vehicle_spawn_sdf.launch:
Invalid <arg> tag: arg 'cmd' has already been declared.

Arg xml is <arg default="xmlstarlet ed -d '//plugin[@name=&quot;mavlink_interface&quot;]/mavlink_udp_port' -s '//plugin[@name=&quot;mavlink_interface&quot;]' -t elem -n mavlink_udp_port -v $(arg mavlink_udp_port) $(find px4)/Tools/sitl_gazebo/models/$(arg vehicle)/$(arg vehicle).sdf" name="cmd"/>

```
@Jaeyoung-Lim I am able to control one vehicle's (say uav0 out of uav0 and uav1) gimbal accurately. I'm unable to figure out why only one of the two. I have modified the launch file as mentioned earlier to be able to launch the two sdf files.

> The single_vehicle_spawn_sdf.launch launch file was also duplicated appropriately and renamed to include the new sdf file and changes made to the multi_uav_mavros_sitl_sdf.launch file.

 ```
   <group ns="uav0">
        <!-- MAVROS and vehicle configs -->
        <arg name="ID" value="0"/>
        <arg name="fcu_url" default="udp://:14540@localhost:14580"/>
        <!-- PX4 SITL and vehicle spawn -->
        <include file="$(find px4)/launch/single_vehicle_spawn_sdf.launch">
.
.
.
    <group ns="uav1">
        <!-- MAVROS and vehicle configs -->
        <arg name="ID" value="1"/>
        <arg name="fcu_url" default="udp://:14541@localhost:14581"/>
        <!-- PX4 SITL and vehicle spawn -->
        <include file="$(find px4)/launch/second_vehicle_spawn_sdf.launch">
```

swaroophangal, 2020-07-13:
@Jaroan , exactly the error that I ran into when I tried. Current stuck there. Once these ports have been assigned in the sdf file, I guess we might not need multiple sdf files.
I tried running a simulation with two vehicles, and getting back the orientation of the mount. One peculiar issue that I observed was that the mount command was being reflected back correctly on the topic, but was not reflected in Gazebo.

swaroophangal, 2020-07-13:
@Jaroan , exactly the error that I ran into when I tried. Current stuck there. Once these ports have been assigned in the sdf file, I guess we might not need multiple sdf files.
I tried running a simulation with two vehicles, and getting back the orientation of the mount. One peculiar issue that I observed was that the mount command was being reflected back correctly on the topic, but was not reflected in Gazebo.
[image]

Jaroan, 2020-07-13:
@swaroophangal  Are any of the vehicles responding to the commands?

Jaeyoung-Lim, 2020-07-13:
A working example of multiuav (without gimbal) can be found here: https://github.com/Jaeyoung-Lim/mavros_controllers/pull/137

Jaeyoung-Lim, 2020-07-13:
@swaroophangal Can you maybe check https://github.com/PX4/sitl_gazebo/blob/master/src/gazebo_gimbal_controller_plugin.cpp#L331 ?

swaroophangal, 2020-07-13:
@Jaroan Yes, both vehicles respond to position and gimbal control commands as seen from the rostopics, but not in Gazebo.  That lead me to think that port wise everything is fine on PX4 end. 
In general multi-vehicle simulation with just vehicles(and not gimbal+camera)  works quite well.

@Jaeyoung-Lim I will check out the gimbal_controller_plugin and get back to you.

I will also check out the QGC issue. I tried searching for the same, but Google failed me. Will search in QGC git. Thanks.

Jaeyoung-Lim, 2020-07-13:
> @Jaroan Yes, both vehicles respond to position and gimbal control commands as seen from the rostopics, but not in Gazebo. That lead me to think that port wise everything is fine on PX4 end.
In general multi-vehicle simulation with just vehicles(and not gimbal+camera) works quite well.

I have verified this with the link I posted above. Seems like it is coming from the plugin

petertheprocess, 2020-07-14:
@Jaroan @Jaeyoung-Lim

Hi, I guess the proplem might lies in this section https://github.com/PX4/sitl_gazebo/blob/4ec5caad7f8d4fa98ed74a6acebd2b2fb0501724/models/typhoon_h480/typhoon_h480.sdf#L1428
```
<plugin name='gimbal_controller' filename='libgazebo_gimbal_controller_plugin.so'>
      <joint_yaw>typhoon_h480::cgo3_vertical_arm_joint</joint_yaw>
      <joint_roll>typhoon_h480::cgo3_horizontal_arm_joint</joint_roll>
      <joint_pitch>typhoon_h480::cgo3_camera_joint</joint_pitch>
```
when two vehicles are spawned in gazebo, only the first instance is called 'typhoon_h480', the second one might be 'typhoon_h480_1', so the plugin cannot find the correct joint to control.

So delete this namespace 'typhoon_h480' can solve this issue.
```
<plugin name='gimbal_controller' filename='libgazebo_gimbal_controller_plugin.so'>
      <joint_yaw>cgo3_vertical_arm_joint</joint_yaw>
      <joint_roll>cgo3_horizontal_arm_joint</joint_roll>
      <joint_pitch>cgo3_camera_joint</joint_pitch>
```

Jaeyoung-Lim, 2020-07-14:
@petertheprocess That was also my suspicion, but didnt work for me. Did you have any success?

Jaroan, 2020-07-14:
@petertheprocess Removing the namespaces didn't work for me as well.

swaroophangal, 2020-07-16:
@Jaeyoung-Lim Some observations:
1. Values of angle command, angle error and angle direction for all joints are logical and change alongwith commands from ROS. All these values are similar for all vehicles given similar commands.
2. While applying those forces at [https://github.com/PX4/sitl_gazebo/blob/master/src/gazebo_gimbal_controller_plugin.cpp#L601](https://github.com/PX4/sitl_gazebo/blob/master/src/gazebo_gimbal_controller_plugin.cpp#L601), calculated force values are going wonky.
3. Observation 2 can be seen on random vehicles(in a set of two vehicles, I saw this randomly on both but not at the same time)
This leads me to think that there probably might some issue in the PID calculations.

Also, i did away with fixed namespaces(typhoon_h480) in the sdf file defined for sonar plugin and all gimbal joints. The above observations are valid for both with and without the namespaces.

petertheprocess, 2020-07-17:
> @petertheprocess That was also my suspicion, but didnt work for me. Did you have any success?
> @petertheprocess Removing the namespaces didn't work for me as well.
I failed too.
But when I duplicate a new typhoon_h480_1.sdf like that
```
<sdf version='1.5'>
  <model name='typhoon_h480_1'>
...........
..........
........
      <sensor name="camera_imu_1" type="imu">
    <plugin name='gimbal_controller' filename='libgazebo_gimbal_controller_plugin.so'>
      <joint_yaw>typhoon_h480_1::cgo3_vertical_arm_joint</joint_yaw>
      <joint_roll>typhoon_h480_1::cgo3_horizontal_arm_joint</joint_roll>
      <joint_pitch>typhoon_h480_1::cgo3_camera_joint</joint_pitch>
 ..........
........
      <gimbal_imu>camera_imu_1</gimbal_imu>
    </plugin>
  </model>
</sdf>
```
I succed!
And if I don't change the <gimbal_imu> name, the gimbal of `typhoon_h480_1` can respons the mavros control cmd, but the orientation seems wrong.

BUT
What confused me is that in the  https://github.com/PX4/sitl_gazebo/blob/master/src/gazebo_gimbal_controller_plugin.cpp
forces are applied on the joints, and the JointPtr is set by
```
this->yawJoint = this->model->GetJoint(yawJointName);
```
and the `model` in the upper code section is a ModelPtr, which is set by gazebo plugin API (I am not sure.).
```
void GimbalControllerPlugin::Load(physics::ModelPtr _model,
  sdf::ElementPtr _sdf)
{
  this->model = _model;
..............
................
................
}
```
If each model instance in gazebo has a unique ModelPtr，then there is no difference between `this->model->GetJoint('cgo3_vertical_arm_joint')` and `this->model->GetJoint('typhoon_h480_1::cgo3_vertical_arm_joint')`.
So, there is no need to duplicate a new sdf, just remove the namespace could work. But we all faied. 
Is there anybody can explain the plugin code?

Jaeyoung-Lim, 2020-07-18:
@petertheprocess what do you mean by success? Were you able to control both gimbals separately?

petertheprocess, 2020-07-18:
> @petertheprocess what do you mean by success? Were you able to control both gimbals separately?

Yep.

petertheprocess, 2020-07-18:
And in http://osrf-distributions.s3.amazonaws.com/gazebo/api/11.0.0/group__gazebo__sensors.html, I find some description about the function `GetSensor()`.

SensorPtr gazebo::sensors::get_sensor (  const std::string &  _name )  
>This name should be fully scoped. This means _name = world_name::model_name::link_name::sensor_name. You may use the unscoped sensor name if that name is unique within the entire simulation. If the name is not unique a nullptr pointer is returned.

Though in  https://github.com/PX4/sitl_gazebo/blob/629e5f5c239d88421edab12b4a7e2e3c3ea7e8fc/src/gazebo_gimbal_controller_plugin.cpp#L338 `GetSensor()` was used, I think the description might work for it too. So does `GetJoint()`.

closed 2020-07-18 by Jaeyoung-Lim

Jaeyoung-Lim, 2020-07-20:
@petertheprocess @Jaroan I have fixed the issue in #549 Can you check if this solves the issue?

Jaroan, 2020-07-21:
@Jaeyoung-Lim I wasn't able to control both gimbals using the fix in #549. I don't understand why though. Should I rebuild after incorporating the changes in the `src/gazebo_gimbal_controller_plugin.cpp` file?

However, using @petertheprocess's method of using the second sdf file and renaming the model instances as mentioned worked.

> <model name='typhoon_h480_1'>
> ...........
> ..........
> ........
>       <sensor name="camera_imu_1" type="imu">
>     <plugin name='gimbal_controller' [...]

Jaeyoung-Lim, 2020-07-21:
@Jaroan of course you need to rebuild with the fix

Jaroan, 2020-07-23:
Yes, this fixed the issue. Thank you very much for your support throughout!

GuangyaoSHI, 2022-01-13:
> @Jaeyoung-Lim Yes, this fixed the issue. Thank you very much for your support throughout!

How do you fix the issue? Rebuild with the fix or use the second sdf? I tried the latest PX4 in which the #549 should be fixed. But I can still only control one gimbal.

GuangyaoSHI, 2022-01-14:
I am now using V1.12.1 PX4 and I notice that in the gimbal controller plugin, the camera_imu bug is fixed. But when I launch two typhoon_h480, only the gimbal of the first drone is controllable.  The /uav0/mount_control command   and /uav1/mout_control command are properly send, which can be observed in the rostopic. 
I notice that when the PX4 for uav1 is launching, it prompts a warning that bind failed, address already in use. I used the default launch file for multiple drones. So the address should be proper assigned. I am quite confused why the gimbal of uav1 is not controllable.

`INFO  [ekf2] starting instance 4, IMU:1 (1310996), MAG:1 (197644)
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 18570 remote port 14550
INFO  [ekf2] starting instance 5, IMU:2 (1311004), MAG:1 (197644)
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14280 remote port 14030
INFO  [mavlink] mode: Gimbal, data rate: 400000 B/s on udp port 13030 remote port 13280
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14558 remote port 14530
WARN  [mavlink] bind failed: Address already in use
`

Jaeyoung-Lim, 2022-01-14:
@GuangyaoSHI You can find the example in https://github.com/Jaeyoung-Lim/mavros_humantracking

GuangyaoSHI, 2022-01-14:
> @GuangyaoSHI You can find the example in https://github.com/Jaeyoung-Lim/mavros_humantracking

Thank you for your response. I tested this example in my desktop. I didn't make any changes to either this ros package (dependencies are installed ) or PX4 (v1.12.1). However, when I launch multiple drone example,  two drones can take off but they cannot follow the set trajectory. They will soon drop to the ground. Do I need tune any parameters?
 
I noticed that despite they can fly, actually only one gimbal is moving. 

So I did another test.  I deleted the geometric controller node because the gimbal control is independent of pose control. I observed the same phenomenon. uav0's gimbal will change its orientation but uav1 will only look at its default direction.

Do you have any idea what could go wrong?  Thank you for your help.

I observed the following warning information when I launched the file.
`WARN  [mavlink] bind failed: Address already in use
`

Jaeyoung-Lim, 2022-01-19:
@GuangyaoSHI Could you post a new issue on https://github.com/Jaeyoung-Lim/mavros_humantracking with the instructions on how to reproduce your issue? (how you are starting the simulation etc)

Jaeyoung-Lim, 2022-01-24:
@GuangyaoSHI I don't think reverting to an older PX4 version is a solution.

GuangyaoSHI, 2022-01-24:
> @GuangyaoSHI I don't think reverting to an older PX4 version is a solution.

Agreed. I will check their modifications to the gimbal controller plugin and try to find out how to fix the problem.

sbosleo, 2022-05-22:
Hello,  video streaming won't work for 

make px4_sitl gazebo_typhoon_h480

and qgroundcontrol.

Any suggestions? 
Thanks.

Jaeyoung-Lim, 2022-05-31:
@sbosleo Please do not hijack the topic of this thread

You can post a new issue it n this repository or ask questions on the public forum discuss.px4.io

BillSiriusLiu, 2022-07-02:
@Jaeyoung-Lim 
> I am now using V1.12.1 PX4 and I notice that in the gimbal controller plugin, the camera_imu bug is fixed. But when I launch two typhoon_h480, only the gimbal of the first drone is controllable. The [...]

The Address already in use is caused by this line:
https://github.com/PX4/PX4-Autopilot/blob/7f76761657bafd8b955e6b559932e750c5391975/ROMFS/px4fmu_common/init.d-posix/airframes/6011_typhoon_h480.post#L4
after adding px4_instance to the ports this warning is silenced. But the gimbal still won't work.
BTW, I was trying to launch two typhoon_h480 and the gimbal of one of them was not working. However, if I change the model to iris both gimbals are correct. Now I'm using the latest version of autopilot. 

And I found in px4-rc.mavlink there are two default gimbal ports 
`udp_onboard_gimbal_port_local=$((13030+px4_instance)) \n
udp_onboard_gimbal_port_remote=$((13280+px4_instance))`
which are not mentioned in the launch file.
And in https://github.com/PX4/PX4-SITL_gazebo/blob/5610c3fb441a2f3babc8ad7a63c8c4ce3e40abfa/src/gazebo_gimbal_controller_plugin.cpp#L751
it seems 13030 is the only port mentioned, so I added an editable port argument both in the launch file and typhoon_h480.sdf.jinja and set them to 13030 and 13031, and now both gimbals fails. LOL
Does this have anything to do with this issue?

aluminium8, 2023-03-01:
gazebo_gimbal_controller_plugin.cpp needs the arg "udp_gimbal_port_remote"(
https://github.com/PX4/PX4-SITL_gazebo-classic/blob/e3722bf9132567e8dd08b7ed6df2986e21a6ec18/src/gazebo_gimbal_controller_plugin.cpp#L360-L365.
but typhoon_h480.sdf don't have it.

`<udp_gimbal_port_remote>13031</udp_gimbal_port_remote>` under `<plugin name='gimbal_controller' filename='libgazebo_gimbal_controller_plugin.so'>` in **typhoon_h480_1.sdf**,  I think I need to write

ParamDeshpande, 2023-06-23:
@aluminium8  thanks for the update, this worked for me, I'm using PX4 version v1.13.3 I made the changes to the .sdf file and to the .sdf.jinja file with 

`<udp_gimbal_port_remote>{{ udp_onboard_gimbal_port_remote }}</udp_gimbal_port_remote>`

to typhoon_h480.sdf.jinja

```
    <plugin name='gimbal_controller' filename='libgazebo_gimbal_controller_plugin.so'>
      <udp_gimbal_port_remote>{{ udp_onboard_gimbal_port_remote }}</udp_gimbal_port_remote>
      <joint_yaw>typhoon_h480::cgo3_vertical_arm_joint</joint_yaw>
      <joint_roll>typhoon_h480::cgo3_horizontal_arm_joint</joint_roll>
      <joint_pitch>typhoon_h480::cgo3_camera_joint</joint_pitch>
```

I tested changes by also adding this to a separate .sdf file ( eg typhoon_h480_1.sdf ) changed the respective ports and this worked. ``
`      <udp_gimbal_port_remote>13031</udp_gimbal_port_remote>  `

after this i was able to control a gimbal for an arbitary ID, on px4 sitl, I launched all nodes separately and was controlling this without `mavros` on `px4>` shell 
```
px4>  gimbal test yaw 10
```

 Just updating this `SITL_gazebo-classic` repo didn't help, I also needed to make this change to individually control gimbals.  

@Jaeyoung-Lim  I had tried using [ human tracking repo ]( https://github.com/Jaeyoung-Lim/mavros_humantracking/ ) but was unable to control gimbals individually using mavros, only one was responsive.  I am yet to try if they are stable / responsive after adding the above changes  As @aluminium8 pointed out in this [issue](https://github.com/Jaeyoung-Lim/mavros_humantracking/issues/18#issue-1603640281) both the models also went haywire.

shupx, 2023-07-17:
@ParamDeshpande Thanks for your sharing. Your method works for the latest PX4 v1.14 but not work for v1.13.3. 

Because the [gazebo_gimbal_controller_plugin used by PX4 v1.13.3](https://github.com/PX4/PX4-SITL_gazebo-classic/blob/48440d7b5c78a21182415266334981f1163f4b2c/src/gazebo_gimbal_controller_plugin.cpp#L743-L757) does not have a `udp_gimbal_port_remote` element. Instead, it used a fixed udp port 13030. The `udp_gimbal_port_remote` element is added by the [commit cc8f33d](https://github.com/PX4/PX4-SITL_gazebo-classic/commit/cc8f33d543d1c480bb46c743430c2564507d6db3) for "gazebo_gimbal_controller_plugin.cpp", which is adpoted by the [gazebo simulation part of PX4 v1.14](https://github.com/PX4/PX4-Autopilot/tree/v1.14.0-beta2/Tools/simulation/gazebo-classic)

Similar issue discussion: https://github.com/PX4/PX4-SITL_gazebo-classic/issues/940