PX4/PX4-SITL_gazebo-classic #543: Models using includes don't behave as expected

iwishiwasaneagle, 2020-07-09:
Running `make px4_sitl gazebo_plane_cam` loads the plane w/ the camera. Then running `commander takeoff` results in the plane being disarmed. However, it barely moves and visually the puller propeller rotates about the z-axis, not the x-axis as expected. Removing all camera related content, leaving the plane.sdf to contain the following gives the exact same results:
```
<?xml version="1.0"?>
<sdf version='1.5'>
  <model name='plane_cam'>
    <include>
      <uri>model://plane</uri>
    </include>
  </model>
</sdf>
```
Furthermore, replacing the content of `plane_cam.sdf` with `plane.sdf` and running `make px4_sitl gazebo_plane_cam` gives the desired results.

To reproduce:
- Clone PX4/Firmware
- Run `make px4_sitl gazebo_plane_cam`

[Output of `make px4_sitl gazebo_plane_cam 2>&1 | tee debug.log`](https://github.com/PX4/sitl_gazebo/files/4897237/gazebo_plane.log)

[review.px4.io](https://logs.px4.io/plot_app?log=d9431ac3-1b24-416f-a631-b417660258d0)

Jaeyoung-Lim, 2020-07-09:
@iwishiwasaneagle I tried to reproduce this, but I can't. (When I run `make px4_sitl gazebo_plane_cam` and run `commander takeoff` makes the plane takeoff)

what system are you running with which gazebo version?

iwishiwasaneagle, 2020-07-09:
> @iwishiwasaneagle I tried to reproduce this on latest master, but I can't. (When I run `make px4_sitl gazebo_plane_cam` and run `commander takeoff` makes the plane takeoff)
> 
> what system are you running with which gazebo version?

Ubuntu 18.04 LTS and Gazebo 11.0.0

```$ uname -a
Linux ubuntu-bionic 4.15.0-109-generic #110-Ubuntu SMP Tue Jun 23 02:39:32 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

```$ gazebo -v
Gazebo multi-robot simulator, version 11.0.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

Gazebo multi-robot simulator, version 11.0.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org
```

Jaeyoung-Lim, 2020-07-09:
Could you try with gazebo9? We have some issues with gazebo11 and it is not fully supported yet

iwishiwasaneagle, 2020-07-09:
That's fixed the problem! Thank you very much

closed 2020-07-09 by iwishiwasaneagle