PX4/PX4-SITL_gazebo-classic #546: GPS does not work in SITL when launch from launch file

swaroophangal, 2020-07-11:
I am facing a weird issue since a few days. GPS messages are not received by PX4 SITL from gazebo. On commander gps status returns not running, while trying to start returns an error as `ERROR [gps] ERR: -1 (tcsetattr)`
Also, all parameters requests by mavros fail.

I am able to run SITL with just the gazebo shell script though.

Don't know my config issue or something else. I tried recloning Firmware multiple times.

Jaeyoung-Lim, 2020-07-11:
@swaroophangal Did this work before the few days?

For me this works. Have you tried setting the `GAZEBO_MODEL_PATH` and `GAZEBO_PLUGIN_PATH` as written in https://dev.px4.io/master/en/simulation/ros_interface.html ?

swaroophangal, 2020-07-11:
@Jaeyoung-Lim  Yes, it used to work before. I have added the required paths in bashrc.

Jaeyoung-Lim, 2020-07-11:
@swaroophangal That is odd. Could you identify exactly which commit it stopped working, and for which model?

swaroophangal, 2020-07-11:
A very silly step omitted by me.

A catkin build/catkin_make is required if the Firmware directory in catkin_ws/src is deleted and recreated or updated via git pull to master /any other branch. This step probably needs to be mentioned on the wiki page at [ https://dev.px4.io/master/en/simulation/ros_interface.html ]( https://dev.px4.io/master/en/simulation/ros_interface.html ).

I am closing the issue.

closed 2020-07-11 by swaroophangal