PX4/PX4-SITL_gazebo-classic #554: cannot find mavlink_sitl_gazebo

atieh-sahraei, 2020-08-02:
Hi, 
My Setup information is as follow:
Mavros: [Installed by, install_geographiclib_datasets.sh]
ROS: [melodic]
Gazebo: [9]
Ubuntu: [18.04]
and the installation did not have any issue and when running " make px4_sitl_default gazebo" the gazebo runs without error. I also set the paths.
However, recently when I want to run the sitl with mavros "roslaunch px4 mavros_posix_sitl.launch" I get the following error:
(It is strange because the day before I could run the launch file but now it gives error)
I am following this tutorial: [(https://gaas.gitbook.io/guide/software-realization-build-your-own-autonomous-drone/build-your-own-autonomous-drone-e01-offboard-control-and-gazebo-simulation)] 

Resource not found: The following package was not found in <arg default="$(find mavlink_sitl_gazebo)/worlds/empty.world" name="world"/>: mavlink_sitl_gazebo
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/name/catkin_ws/src/mavros/test_mavros
ROS path [2]=/home/name/catkin_ws/src/Firmware
ROS path [3]=/home/name/catkin_ws/src/mavros/mavros_extras
ROS path [4]=/home/name/catkin_ws/src/mavros/mavros
ROS path [5]=/home/name/catkin_ws/src/mavros/mavros_msgs
ROS path [6]=/home/name/catkin_ws/src/mavros/libmavconn
ROS path [7]=/home/name/catkin_ws/src/beginner_tutorials
ROS path [8]=/opt/ros/melodic/share
The traceback for the exception was written to the log file

This is  the launch file:
`<?xml version="1.0"?>
<launch>
    
    
    
    <arg name="x" default="0"/>
    <arg name="y" default="0"/>
    <arg name="z" default="0"/>
    <arg name="R" default="0"/>
    <arg name="P" default="0"/>
    <arg name="Y" default="0"/>
    
    <arg name="est" default="ekf2"/>
    <arg name="vehicle" default="iris"/>
    <arg name="world" default="$(find mavlink_sitl_gazebo)/worlds/empty.world"/>
    <arg name="sdf" default="$(find mavlink_sitl_gazebo)/models/$(arg vehicle)/$(arg vehicle).sdf"/>

    
    <arg name="gui" default="true"/>
    <arg name="debug" default="false"/>
    <arg name="verbose" default="false"/>
    <arg name="paused" default="false"/>
    <arg name="respawn_gazebo" default="false"/>
    
    <arg name="fcu_url" default="udp://:14540@localhost:14557"/>
    <arg name="respawn_mavros" default="false"/>
    
    <arg name="interactive" default="true"/>
    
    <include file="$(find px4)/launch/posix_sitl.launch">
        <arg name="x" value="$(arg x)"/>
        <arg name="y" value="$(arg y)"/>
        <arg name="z" value="$(arg z)"/>
        <arg name="R" value="$(arg R)"/>
        <arg name="P" value="$(arg P)"/>
        <arg name="Y" value="$(arg Y)"/>
        <arg name="world" value="$(arg world)"/>
        <arg name="vehicle" value="$(arg vehicle)"/>
        <arg name="sdf" value="$(arg sdf)"/>
        <arg name="gui" value="$(arg gui)"/>
        <arg name="interactive" value="$(arg interactive)"/>
        <arg name="debug" value="$(arg debug)"/>
        <arg name="verbose" value="$(arg verbose)"/>
        <arg name="paused" value="$(arg paused)"/>
        <arg name="respawn_gazebo" value="$(arg respawn_gazebo)"/>
    </include>
    
    <include file="$(find mavros)/launch/px4.launch">
        
        <arg name="gcs_url" value=""/>
        <arg name="fcu_url" value="$(arg fcu_url)"/>
        <arg name="respawn_mavros" value="$(arg respawn_mavros)"/>
    </include>
</launch>`

Jaeyoung-Lim, 2020-08-04:
It seems like your `GAZEBO_PLUGIN_PATH` is not set correctly

adbidwai, 2020-08-06:
Your package is not found in the path of ROS packages.
It seems have not sourced your current workspace in your terminal.
Build your packages again and source your workspace again after building it.

closed 2020-08-12 by Jaeyoung-Lim

HuaYuXiao, 2024-08-24:
> It seems like your `GAZEBO_PLUGIN_PATH` is not set correctly

You should check your `~/.bashrc`, and make sure `GAZEBO_PLUGIN_PATH` is set correctly.