PX4/PX4-SITL_gazebo-classic #560: Looks like the formula is incorrect

a6a3uh, 2020-08-04:
https://github.com/PX4/sitl_gazebo/blob/2858552ed70c379ac240113373e5f50310d98144/src/liftdrag_plugin/liftdrag_plugin.cpp#L259

As stated in comments above the correct formulae is:
projected = spanwiseI Xcross ( vector Xcross spanwiseI)
this vector triple product could be converted to dot product:
vel - vel.Dot(spanwiseI)*spanwiseI
instead we have following:
vel - vel.Dot(spanwiseI)*velI

Putting it another way. spanwise velocity is velocity projection onto spanwiseI multiplied by spanwise unit vector (not velocity unit vector).

Jaeyoung-Lim, 2020-08-05:
@a6a3uh Good catch, could you make a PR for the fix?

a6a3uh, 2020-08-05:
Done https://github.com/PX4/sitl_gazebo/pull/562

Jaeyoung-Lim, 2020-08-05:
Fixed in #562

closed 2020-08-05 by Jaeyoung-Lim