PX4/PX4-SITL_gazebo-classic #564: add_library cannot create target "gmock_main" because another target with the same name already exists.

adbidwai, 2020-08-06:
I am trying to build sitl_gazebo and while building (using `catkin build`), I'm facing errors
This is my error log :
```
Make Error at /usr/share/cmake-3.10/Modules/ExternalProject.cmake:2975 (add_custom_target):
  add_custom_target cannot create target "GMock" because another target with
  the same name already exists.  The existing target is a custom target
  created in source directory
  "/home/adbidwai/sop_ws/src/sitl_gazebo/external/OpticalFlow/external/klt_feature_tracker".
  See documentation for policy CMP0002 for more details.
Call Stack (most recent call first):
  /usr/share/cmake/GMock/GMockConfig.cmake:56 (ExternalProject_Add)
  /opt/ros/melodic/share/catkin/cmake/test/gtest.cmake:312 (find_package)
  /opt/ros/melodic/share/catkin/cmake/all.cmake:164 (include)
  /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:105 (find_package)

CMake Error at /usr/share/cmake/GMock/GMockConfig.cmake:65 (add_library):
  add_library cannot create target "gtest" because another target with the
  same name already exists.  The existing target is an interface library
  created in source directory
  "/home/adbidwai/sop_ws/src/sitl_gazebo/external/OpticalFlow/external/klt_feature_tracker".
  See documentation for policy CMP0002 for more details.
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/test/gtest.cmake:312 (find_package)
  /opt/ros/melodic/share/catkin/cmake/all.cmake:164 (include)
  /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:105 (find_package)

CMake Error at /usr/share/cmake/GMock/GMockConfig.cmake:70 (add_library):
  add_library cannot create target "gtest_main" because another target with
  the same name already exists.  The existing target is an interface library
  created in source directory
  "/home/adbidwai/sop_ws/src/sitl_gazebo/external/OpticalFlow/external/klt_feature_tracker".
  See documentation for policy CMP0002 for more details.
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/test/gtest.cmake:312 (find_package)
  /opt/ros/melodic/share/catkin/cmake/all.cmake:164 (include)
  /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:105 (find_package)

CMake Error at /usr/share/cmake/GMock/GMockConfig.cmake:74 (add_library):
  add_library cannot create target "gmock" because another target with the
  same name already exists.  The existing target is an interface library
  created in source directory
  "/home/adbidwai/sop_ws/src/sitl_gazebo/external/OpticalFlow/external/klt_feature_tracker".
  See documentation for policy CMP0002 for more details.
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/test/gtest.cmake:312 (find_package)
  /opt/ros/melodic/share/catkin/cmake/all.cmake:164 (include)
  /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:105 (find_package)

CMake Error at /usr/share/cmake/GMock/GMockConfig.cmake:78 (add_library):
  add_library cannot create target "gmock_main" because another target with
  the same name already exists.  The existing target is an interface library
  created in source directory
  "/home/adbidwai/sop_ws/src/sitl_gazebo/external/OpticalFlow/external/klt_feature_tracker".
  See documentation for policy CMP0002 for more details.
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/test/gtest.cmake:312 (find_package)
  /opt/ros/melodic/share/catkin/cmake/all.cmake:164 (include)
  /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:20 (include)
  CMakeLists.txt:105 (find_package)
```

Please help in reolving this issue.
TIA

Jaeyoung-Lim, 2020-08-12:
@adbidwai why are you trying to build this package with catkin build?

adbidwai, 2020-08-12:
> @adbidwai why are you trying to build this package with catkin build?

Should I try builidng using catkin_make instead of catkin build?

Jaeyoung-Lim, 2020-08-12:
@adbidwai No, if you are simply trying to run SITL with ROS, you need to build PX4 separately which includes sitl_gazebo as a submodule. 

please follow the instructions in https://dev.px4.io/master/en/simulation/ros_interface.html

Jaeyoung-Lim, 2020-08-15:
@adbidwai Have you checked if you have gtest/gmock installed?

adbidwai, 2020-08-19:
Yes I did. I am facing the same error still.

Jaeyoung-Lim, 2020-11-25:
Fixed in https://github.com/PX4/PX4-SITL_gazebo/pull/652

closed 2020-11-25 by Jaeyoung-Lim