PX4/PX4-SITL_gazebo-classic #579: compile error

tiemuhua, 2020-08-19:
`/home/zhuchen/px4/sitl_gazebo/src/gazebo_mavlink_interface.cpp: In member function ‘void gazebo::GazeboMavlinkInterface::GpsCallback(gazebo::GpsPtr&, const int&)’:
/home/zhuchen/px4/sitl_gazebo/src/gazebo_mavlink_interface.cpp:1015:15: error: ‘mavlink_hil_gps_t {aka struct __mavlink_hil_gps_t}’ has no member named ‘id’
   hil_gps_msg.id = id;
               ^
/home/zhuchen/px4/sitl_gazebo/src/gazebo_mavlink_interface.cpp: In member function ‘void gazebo::GazeboMavlinkInterface::LidarCallback(gazebo::LidarPtr&, const int&)’:
/home/zhuchen/px4/sitl_gazebo/src/gazebo_mavlink_interface.cpp:1048:14: error: ‘mavlink_distance_sensor_t {aka struct __mavlink_distance_sensor_t}’ has no member named ‘signal_quality’
   sensor_msg.signal_quality = lidar_message->signal_quality();
              ^
/home/zhuchen/px4/sitl_gazebo/src/gazebo_mavlink_interface.cpp: In member function ‘void gazebo::GazeboMavlinkInterface::SonarCallback(gazebo::SonarPtr&, const int&)’:
/home/zhuchen/px4/sitl_gazebo/src/gazebo_mavlink_interface.cpp:1120:14: error: ‘mavlink_distance_sensor_t {aka struct __mavlink_distance_sensor_t}’ has no member named ‘signal_quality’
   sensor_msg.signal_quality = sonar_message->signal_quality();
              ^
[ 91%] Built target gazebo_video_stream_widget
/home/zhuchen/px4/sitl_gazebo/src/gazebo_geotagged_images_plugin.cpp:739:22: error: too many arguments to function ‘uint16_t mavlink_msg_camera_capture_status_pack_chan(uint8_t, uint8_t, uint8_t, mavlink_message_t*, uint32_t, uint8_t, uint8_t, float, uint32_t, float)’
         _imageCounter);                         // total number of images
                      ^
`
can anybody tell me why the project can not compile successfully?

Jaeyoung-Lim, 2020-08-19:
@tiemuhua It looks like you have a outdated mavlink in firmware

tomlogan501, 2020-08-24:
Hi,
Got the same error just for /home/tsimon/media/Documents/sitl_gazebo/src/gazebo_geotagged_images_plugin.cpp with GCC 8.4.0
@Jaeyoung-Lim where do you check the mavlink firmware version ?

Jaeyoung-Lim, 2020-08-24:
@tomlogan501 You can check mavlink version in the Firmware : `mavlink/include/2.0`

tomlogan501, 2020-08-24:
Yeah sure, but it seems to require an upgraded version of the 2.0, i got some diffs since my previous pull

Jaeyoung-Lim, 2020-10-05:
This has been fixed on upstream firmware. Please re-open if otherwise

closed 2020-10-05 by Jaeyoung-Lim