PX4/PX4-SITL_gazebo-classic #581: PX4 SITL Gazebo Standard_VTOL will not yaw for takeoff and transition

UASOP, 2020-08-20:
Greetings, 

I am currently working on a project using the PX4 standard VTOL sitl gazebo and noticed a repeating behavior that in order to begin the transition to fixed wing for a VTOL takeoff and transition waypoint, the waypoint has to be directly in front of the SITL standard VTOL. If the waypoint is anywhere other then in front of the aircraft with an azimuth of 090 with some minor deviations in degrees from there, the aircraft will not yaw to face the heading for transition. I have checked the weathervane parameters as well just to see if the WV_YRATE_MAX was set to 0 and it is currently 90. If there any reason why the SITL aircraft is not following the command to yaw t a specific heading before the transition? 

Thanks in advance for he assistance. In the mean time I am going to try to disable the WV_EN parameter to see if that fixes the problem and follow up back here.

UASOP, 2020-08-20:
Disabling weathervane did not resolve the issue.

UASOP, 2020-08-20:
Alright so quick update here. I did a quick git checkout of version 1.10.1 and the heading changes on the VTOL Standard as needed. The version that I was running with the heading not working was the current master. I am going to look to see what differences i can find between the two after running a test on version 1.10.2 just to see if there was any issues with that version.

Jaeyoung-Lim, 2020-08-20:
@UASOP I think this is not an issue of the simulation, but the firmware. Please post a issue on the firmware sie

UASOP, 2020-08-20:
Posted in PX4 Firmware github issue as instructed, closing this issue out for now.

closed 2020-08-20 by UASOP