PX4/PX4-SITL_gazebo-classic #586: A weird parameter definition in LiftDrag_plugin.

petertheprocess, 2020-08-24:
In the LiftDrag_plugin, there is a parameter named `a0`, it is defined and used as follows:
```
 <plugin name="right_wing" filename="libLiftDragPlugin.so">
      <a0>0.05984281113</a0>
      .................
      ......................
      ...........................
 </plugin>
```
https://github.com/PX4/sitl_gazebo/blob/78bba3f25e7e240bf70878bd007dfd77b33440c1/src/liftdrag_plugin/liftdrag_plugin.cpp#L284
https://github.com/PX4/sitl_gazebo/blob/78bba3f25e7e240bf70878bd007dfd77b33440c1/src/liftdrag_plugin/liftdrag_plugin.cpp#L301-L303
 
And in http://gazebosim.org/tutorials?tut=aerodynamics&cat=physics, its meaning was declared as:
>Note in this example, the airfoil has non-zero camber, and has a net positive lift at zero angle of attack. To obtain equivalent representation using the current Gazebo LiftDragPlugin plugin parameters, shift the entire curve to the right such that the zero lift point corresponds to zero angle of attack. And we can label the original zero angle of attack location as a0 in the shifted curve, i.e. a0 is 5 degrees. Also shift the stall angle accordingly, i.e. alpha_stall is now 19.2 degrees.

[image]

I think it is weird for people who know aerodynamics, since a0 can be easily misunderstood as the `zero-lift AOA` .
So I suggest changing the code like this.
```
this->alpha =  acos(cosAlpha) - this->alpha0; 
```

Jaeyoung-Lim, 2020-08-24:
@petertheprocess Indeed the liftdrag model in OSRF is a bit odd for me. But the liftdrag plugin has been simply pulled from OSRF's liftdrag plugin

I think it makes sense to change the definition, but we need to document this very clearly otherwise people will get confused even more with the new definition.

If you are interested, please make a PR with the relevant changes

Jaeyoung-Lim, 2020-11-25:
Closing since this is stale.

If you would want to improve the representation in this repo, please reopen and create a PR

closed 2020-11-25 by Jaeyoung-Lim