PX4/PX4-SITL_gazebo-classic #587: The deflection of the rudder has no effect on the force of the plane.

petertheprocess, 2020-08-24:
Hi, I find a bug in plane.sdf
https://github.com/PX4/sitl_gazebo/blob/78bba3f25e7e240bf70878bd007dfd77b33440c1/models/plane/plane.sdf#L587-L626
For the elevator, <control_joint_name> and <control_joint_rad_to_cl> are defined to take the deflection of elevator into account.
But for the rudder, <control_joint_name> and <control_joint_rad_to_cl> are missed.
```
 <plugin name="rudder" filename="libLiftDragPlugin.so"> 
   <a0>0.0</a0> 
   <cla>4.752798721</cla> 
   <cda>0.6417112299</cda> 
   <cma>0.0</cma> 
   <alpha_stall>0.3391428111</alpha_stall> 
   <cla_stall>-3.85</cla_stall> 
   <cda_stall>-0.9233984055</cda_stall> 
   <cma_stall>0</cma_stall> 
   <cp>-0.5 0 0.05</cp> 
   <area>0.02</area> 
   <air_density>1.2041</air_density> 
   <forward>1 0 0</forward> 
   <upward>0 1 0</upward> 
   <link_name>base_link</link_name> 
   <control_joint_name> 
     rudder_joint 
   </control_joint_name> 
   <control_joint_rad_to_cl>4.0</control_joint_rad_to_cl>
   <robotNamespace></robotNamespace> 
   <windSubTopic>world_wind</windSubTopic> 
 </plugin> 
```
Since rudder_joint axis is [0 0 1], when the rudder rotate to the right( $$\delta > 0$$), the lift increase. So  <control_joint_rad_to_cl> should be an postive value.

Jaeyoung-Lim, 2020-08-24:
@petertheprocess Please feel free to make the fix and create a pull request

petertheprocess, 2020-08-24:
> @petertheprocess Please feel free to make the fix and create a pull request
thx
Honoured to do this.

Jaeyoung-Lim, 2020-10-05:
Fixed in #588

closed 2020-10-05 by Jaeyoung-Lim