PX4/PX4-SITL_gazebo-classic #592: Imu turn_on_bias repeatedly added

robin-shaun, 2020-08-29:
Hello~ As the code below shows, `gyroscope_turn_on_bias_` and `accelerometer_turn_on_bias_` are added every time when the measurement is updated. I think the turn_on_bias only takes effect at the beginning.

```
void GazeboImuPlugin::addNoise(Eigen::Vector3d* linear_acceleration,Eigen::Vector3d* angular_velocity, const double dt) 
{
  for (int i = 0; i < 3; ++i) {
    gyroscope_bias_[i] = phi_g_d * gyroscope_bias_[i] +
        sigma_b_g_d * standard_normal_distribution_(random_generator_);
    (*angular_velocity)[i] = (*angular_velocity)[i] +
        gyroscope_bias_[i] +
        sigma_g_d * standard_normal_distribution_(random_generator_) +
        gyroscope_turn_on_bias_[i];
  }

  for (int i = 0; i < 3; ++i) {
    accelerometer_bias_[i] = phi_a_d * accelerometer_bias_[i] +
        sigma_b_a_d * standard_normal_distribution_(random_generator_);
    (*linear_acceleration)[i] = (*linear_acceleration)[i] +
        accelerometer_bias_[i] +
        sigma_a_d * standard_normal_distribution_(random_generator_) +
        accelerometer_turn_on_bias_[i];
  }
}

```

Moreover, there isn't turn_on_bias for magnetometer in gazebo_magnetometer_plugin and GPS in gazebo_gps_plugin. I don't know why.

Jaeyoung-Lim, 2020-08-30:
@robin-shaun Thanks for looking into this. Please feel free to create a PR to fix it

Jaeyoung-Lim, 2020-10-05:
@robin-shaun Any updates?

closed 2020-10-07 by Jaeyoung-Lim