PX4/PX4-SITL_gazebo-classic #593: Libgstreamer1.0-package has unmet dependencies (Broken packages)

HarshilNaikk, 2020-09-02:
I am trying to get px4-sitl to work work with gazebo, and am following https://dev.px4.io/v1.9.0/en/ros/mavros_installation.html for the steps. 
However, after i try to install libgstreamer plugins, i get a message saying that the package is broken and has unmet dependencies.

> Reading package lists... Done
Building dependency tree       
Reading state information... Done
gstreamer1.0-clutter-3.0 is already the newest version (3.0.26-1).
gstreamer1.0-packagekit is already the newest version (1.1.9-1ubuntu2).
gstreamer1.0-plugins-good is already the newest version (1.14.0-1ubuntu1).
gstreamer1.0-pulseaudio is already the newest version (1.14.0-1ubuntu1).
gstreamer1.0-libav is already the newest version (1.14.0-1).
gstreamer1.0-plugins-ugly is already the newest version (1.14.0-1).
gstreamer1.0-alsa is already the newest version (1.14.1-1ubuntu1~ubuntu18.04.2).
gstreamer1.0-plugins-base is already the newest version (1.14.1-1ubuntu1~ubuntu18.04.2).
gstreamer1.0-plugins-base-apps is already the newest version (1.14.1-1ubuntu1~ubuntu18.04.2).
gstreamer1.0-tools is already the newest version (1.14.1-1~ubuntu18.04.2).
gstreamer1.0-x is already the newest version (1.14.1-1ubuntu1~ubuntu18.04.2).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:
The following packages have unmet dependencies:
 gstreamer1.0-plugins-base-dbg : Depends: libgstreamer-plugins-base1.0-0 (= 1.14.1-1ubuntu1~ubuntu18.04.2) but 1.14.5-0ubuntu1~18.04.1 is to be installed
 gstreamer1.0-python3-dbg-plugin-loader : Conflicts: gstreamer1.0-python3-plugin-loader but 1.14.0-1 is to be installed
 gstreamer1.0-python3-plugin-loader : Conflicts: gstreamer1.0-python3-dbg-plugin-loader but 1.14.0-1 is to be installed
E: Unable to correct problems, you have held broken packages.

I tried looking up issues on google, but none seem to help. I reinstalled ROS twice, but the issue persists. 
Any help would be greatly appreciated!

I am using ROS Melodic on Ubuntu 18.04

Jaeyoung-Lim, 2020-09-28:
@HarshilNaikk I think reinstalling ROS is not the correct solution.

Could you share what exact error you are having?

HarshilNaikk, 2020-09-28:
Actually it turns out that while installing ROS, my wifi rebooted. so some packages might not have been installed properly. 
I reinstalled ROS and it works just fine.

Jaeyoung-Lim, 2020-09-28:
@HarshilNaikk Thanks for the update, closing

closed 2020-09-28 by Jaeyoung-Lim