PX4/PX4-SITL_gazebo-classic #603: Windy world in SITL no longer building

UASOP, 2020-09-20:
There seems to be a new build problem when trying to use the make px4_sitl gazebo_standard_vtol__windy command for wind simulation environments. Output is below. I am using the latest master. It is stating there is a permission denied for the windy.world file, which is strand as none of the world files are executable and empty.world responds without issue. 

/Tools/sitl_run.sh: line 111: /home/uasop/Firmware/Tools/sitl_gazebo/worlds/windy.world: Permission denied
FAILED: platforms/posix/CMakeFiles/gazebo_standard_vtol__windy 
cd /home/uasop/Firmware/build/px4_sitl_default/tmp && /home/uasop/Firmware/Tools/sitl_run.sh /home/uasop/Firmware/build/px4_sitl_default/bin/px4 none gazebo standard_vtol windy /home/uasop/Firmware /home/uasop/Firmware/build/px4_sitl_default
ninja: build stopped: subcommand failed.
Makefile:224: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1

Jaeyoung-Lim, 2020-09-20:
@UASOP Are you sure you have the submodule updated correctly? which version of PX4 firmware are you running?

UASOP, 2020-09-20:
@Jaeyoung-Lim I am currently using the origin master release. Firmware has been downloaded with --recursive and also ran git submodule update --init --recursive. Issue still persists. Thank you for the quick response.

Jaeyoung-Lim, 2020-09-20:
@UASOP I have tried on my system and I can run it without the issue. Looks like you have permission issues on your system

UASOP, 2020-09-20:
@Jaeyoung-Lim Thanks I will look further into whats causing this on ubuntu18.04LTS with the latest updates. I didnt change any system permissions recently so I am curious as to what caused that.

UASOP, 2020-09-20:
I have tried the latest build on another computer, fresh install of both ubuntu 18.04 LTS and also with 20.04. Both have given the same issue.

UASOP, 2020-09-20:
There is an additional message reported 

[48/110] Building CXX object CMakeFile...c/gazebo_geotagged_images_plugin.cpp.o
/home/uasop/Gemini01/Tools/sitl_gazebo/src/gazebo_geotagged_images_plugin.cpp: In member function ‘void gazebo::GeotaggedImagesPlugin::OnNewFrame(const unsigned char*)’:
/home/uasop/Gemini01/Tools/sitl_gazebo/src/gazebo_geotagged_images_plugin.cpp:245:11: warning: ignoring return value of ‘int system(const char*)’, declared with attribute warn_unused_result [-Wunused-result]
     system(gps_tag_command);
     ~~~~~~^~~~~~~~~~~~~~~~~
At global scope:
cc1plus: warning: unrecognized command line option ‘-Wno-address-of-packed-member’
[110/110] Linking CXX shared library libgazebo_video_stream_widget.so

UASOP, 2020-09-20:
Addtional Output FAILED: platforms/posix/CMakeFiles/gazebo_standard_vtol__windy

Jaeyoung-Lim, 2020-09-20:
@UASOP please share the full output. The outputs you are sharing are irrelevant

UASOP, 2020-09-20:
@Jaeyoung-Lim Attached are two files of the full output. Please let me know if you need more information. 
[makecleanoutput1.txt](https://github.com/PX4/sitl_gazebo/files/5252822/makecleanoutput1.txt)
[output.txt](https://github.com/PX4/sitl_gazebo/files/5252823/output.txt)

Jaeyoung-Lim, 2020-09-23:
@UASOP Okay, now I see the issue, I can reproduce it on Ubuntu 18.04

a-tab-a, 2020-09-28:
Not a fix, but windy world does load. I just changed the default world to windy world in sitl_run.sh file line 110. So make px4_sitl gazebo_iris__empty is also failing saying the permission denied issue. However the default make px4_sitl gazebo works perfect.

Jaeyoung-Lim, 2020-09-28:
@a-tab-a @UASOP Created a fix in https://github.com/PX4/Firmware/pull/15844

closed 2020-09-28 by dagar