PX4/PX4-SITL_gazebo-classic #604: Incorrect distance sensor readings with iris_obs_avoid

mortenfyhn, 2020-09-21:
I'm trying out simulation with multiple distance sensors in gazebo, but the readings I'm seeing are wrong. I'm running PX4/Firmware@v1.11.0, and its version of sitl_gazebo. The downward facing distance sensor seems good, but the forward/left/right ones all measure approx. 20 cm no matter what they actually point towards.

Reproduce with
* `git clone git@github.com:PX4/Firmware --recursive --branch v1.11.0`
* `cd Firmware`
* `make px4_sitl gazebo_iris_obs_avoid`
* Launch QGroundControl (v4.0.10).
* Set parameter `COM_OBS_AVOID` to 0, to allow flying without the avoidance system.
* Set position mode, then arm and take off.
* Observe in Gazebo that the forward/left/right distance sensor visualizations in Gazebo stop at around 20 cm.
* Run `listener distance_sensor` in the PX4 prompt, and see the same ~20 cm distances forward/left/right.
* Downwards distance is good.

I've removed the asphalt plane in the world for this screenshot to make it easier to see:
[image]

I think it may be caused by the distance sensor plugin seeing the drone itself. If i modify the sdf file to move the sensors 5 cm further down (they are underneath the drone), like so:
```diff
diff --git a/models/iris_obs_avoid/iris_obs_avoid.sdf b/models/iris_obs_avoid/iris_obs_avoid.sdf
index 9f4332b..5d7b51b 100644
--- a/models/iris_obs_avoid/iris_obs_avoid.sdf
+++ b/models/iris_obs_avoid/iris_obs_avoid.sdf
@@ -36,7 +36,7 @@
     <!--forward-facing lidar-->
     <include>
       <uri>model://lidar</uri>
-      <pose>0 0 -0.05 0 -1.57079633 0</pose>
+      <pose>0 0 -0.10 0 -1.57079633 0</pose>
       <name>lidar1</name>
     </include>
 
@@ -48,7 +48,7 @@
     <!--right-facing lidar-->
     <include>
       <uri>model://lidar</uri>
-      <pose>0 0 -0.05 -1.57079633 0 0</pose>
+      <pose>0 0 -0.10 -1.57079633 0 0</pose>
       <name>lidar2</name>
     </include>
 
@@ -60,7 +60,7 @@
     <!--left-facing lidar-->
     <include>
       <uri>model://lidar</uri>
-      <pose>0 0 -0.05 1.57079633 0 0</pose>
+      <pose>0 0 -0.10 1.57079633 0 0</pose>
       <name>lidar3</name>
     </include>
```
Then they seem to work. Here's an example where I put some guys to block the sensor beams. I can see from `listener distance_sensor` that the values are good, approx 4-5 meters.

[image]

I also noticed that with the same fix, and nothing blocking the sensors, the visualization shows around 35 meters, while the uorb (`listener distance_sensor`) shows current distance = 0.2 (the same as minimum distance). The value in this case should probably be inf, or some value larger than the max distance, so that this can be detected as an invalid / out of range measurement. I think the 35 meter visualization comes from `lidar.sdf` using different limit values for the sdf ray sensor config and the laser plugin config:

[image]

https://github.com/PX4/sitl_gazebo/blob/master/models/lidar/model.sdf#L39-L54

Jaeyoung-Lim, 2020-09-28:
Thanks @mortenfyhn 

I think your observation is correct, could you add a PR to fix this issue?

mortenfyhn, 2020-09-29:
Yeah, can do! But do you know why it happens? Is it because the collision geometry is different from the visual geometry, so that the laser hits the collision box, while the visible 3D model is different, so it looks wrong because of that?

Jaeyoung-Lim, 2020-09-29:
@mortenfyhn I have seen the same on other models, (e.g. on the `plane_lidar` model the laser is attached quite low.

I was assuming the laster only collides with the collision geometry (correct me if I am wrong) probably when the distance sensors are inside the collision geometry, it sees the inside of the box. But haven't looked more into it afterwards

PR #618 (2020-10-01): Adjust iris obs avoid distance sensor location
Fix #604.

I've moved the horizontal (forward, left, right, back) distance sensors down from -5 cm to -10 cm. This makes them mostly work as they should, but they still sometimes give incorrect (very short) readings. See log below, and gazebo collision box image:

[image]
[image]
closed 2020-10-01 by Jaeyoung-Lim