PX4/PX4-SITL_gazebo-classic #608: Mount orientation control on typhoon_h480 has disparity between gazebo and mavros

swaroophangal, 2020-09-25:
Hello @Jaeyoung-Lim ,

I am trying to command the gimbal on typhoon_h480 to -90 in pitch. But the camera is never at -90 in pitch...i would have understood even if it stopped at -89.0. But there is more than 10deg offset in the commanded value and seen orientation in gazebo. Please let me know if I am going wrong somewhere. I have attached a screenshot for your reference. 
I have tried the following:
1. Changed the limits of the pitch joint in the sdf at https://github.com/PX4/sitl_gazebo/blob/master/models/typhoon_h480/typhoon_h480.sdf#L308
2. Tried commanding more than -90 to the joint via mavros
3. Tried changing the MNT_OFF_PITCH parameter to compensate for the offset

Swaroop

[image]

Jaeyoung-Lim, 2020-09-25:
@swaroophangal While I can see that the camera orientation does indeed seem wrong, could you confirm that the gimbal orientation is also not tracking properly?

swaroophangal, 2020-09-25:
@Jaeyoung-Lim the screenshot top right pane is the output of mount_control/orientation topic echo. Mavros supplies the exact same value that has been commanded, whereas gazebo does not reflect that for pitch command -90 as can be seen in the screenshot left region. I did not try commanding the complete range to understand where exactly the travel reduces per degree of command. I also did not check with the roll axis. Will do them later as pitch is a high priority for me right now.

swaroophangal, 2020-09-28:
@Jaeyoung-Lim could you please point me to the direction where I should start looking for the issue?

Jaeyoung-Lim, 2020-09-29:
@swaroophangal You need to debug where the difference is coming for. Have you looked into the `gimbal_controller_plugin`?

Jaeyoung-Lim, 2020-10-07:
@swaroophangal Any updates?

zapurva, 2020-10-15:
At https://github.com/PX4/sitl_gazebo/blob/b943edaddee2b5e84e7bb489c43ebb13dd70495d/models/typhoon_h480/typhoon_h480.sdf#L424

Changing `<lower>` to `-2.09` solves this issue

Jaeyoung-Lim, 2020-10-15:
@zapurva Okay, seems like there was a joint limit on the pitch axis. Could you make a PR for this?

Jaeyoung-Lim, 2020-10-16:
Fixed in https://github.com/PX4/sitl_gazebo/pull/636

closed 2020-10-16 by Jaeyoung-Lim