PX4/PX4-SITL_gazebo-classic #620: Iris (quadrotor) Gazebo (PX4) Model Modification

noman0511, 2020-10-05:
Hello Everyone,

I am new to gazebo/px4 and I am trying to simulate iris model based on my own parameters of the quad-rotor which has weight of about 1350-1400 grams, using 1045 propeller, 920kV motor and 3s Li-Po battery. How can i set my simulations based on my drone's parameter. The idea is to stabilize the model in simulation before coming to the real life testing . I will be using Pixhawk controller. I have gone through a lot of discussions on PX4 and GitHub. and couldn't get a proper answer that what files do i need to modify and what parameters do i need to modify to get my simple simulation properly running in the gazebo.  So can someone guide me/help me to get my simulation working. 

Regards,

Jaeyoung-Lim, 2020-10-05:
@noman0511 You can by modifying the parameters in the sdf file. https://github.com/PX4/sitl_gazebo/blob/master/models/iris/iris.sdf.jinja

You can look at the motor_model plugin to and also change the inertial properties of the vehicle by modifying `<inertial>` properties of the base_link. If you are unfamiliar with sdf files, there are a lot of tutorials availabe (e.g. http://gazebosim.org/tutorials?tut=build_model )

noman0511, 2020-10-05:
Thanks for the response, so sdf is the only file that i need to modify for this simple simulation? In the motor_model plugin we need to have some calculations forexample,
     

       <timeConstantUp>0.0125</timeConstantUp>
      <timeConstantDown>0.025</timeConstantDown>
      <maxRotVelocity>1100</maxRotVelocity>
      <motorConstant>5.84e-06</motorConstant>
      <momentConstant>0.06</momentConstant>
      <rotorDragCoefficient>0.000175</rotorDragCoefficient>
      <rollingMomentCoefficient>1e-06</rollingMomentCoefficient>

These constants have their own calculations/equtions, i dont have much data about my motors, except that i have given already. How can i calculate e.g, max thurst, max motor efficency, and the above constants using a very limited data.

Jaeyoung-Lim, 2020-10-05:
@noman0511 You need to infer them from the datasheet or measure them directly

deepak-1530, 2020-10-05:
@noman0511 if you have a CAD file of your drone, you can generate the SDF file from that (get URDF from this http://wiki.ros.org/sw_urdf_exporter and convert to SDF (https://answers.ros.org/question/53752/how-can-i-transform-an-urdf-file-to-an-sdf-file/) ).  That way, you'll get all the meshes and body dimensions as well. Then you can vary the properties as mentioned by @Jaeyoung-Lim in the SDF file.

noman0511, 2020-10-06:
Thanks all for the support. @deepak-1530 I don't have the CAD file the only information I have right now is about the Voltage that I am applying, the size of the propeller, and the max thrust in grams generated by each motor, revolution per minute of the rotor the weight of the motor, and the overall weight of the drone. That's why I am stuck to this point. Any suggestions?

Jaeyoung-Lim, 2020-10-28:
@noman0511 You need to either measure it yourself, or infer the information from the specs of the components you use. 

CAD only contains geometric information of the vehicle so you need to come up with this information yourself

engcang, 2020-11-03:
@noman0511 @Jaeyoung-Lim 

Hi guys. Please refer [here](https://github.com/engcang/mavros-gazebo-application/tree/master/px4_model_making#-motor), which is just from [here](https://github.com/PX4/PX4-SITL_gazebo/issues/110) and [here](https://github.com/PX4/PX4-SITL_gazebo/issues/110)

But I think the discussion could not end yet and people are still waiting for the original author/programmer who coded the 'motor_model_plugin' first. I believe @Jaeyoung-Lim must know something better maybe.

Jaeyoung-Lim, 2020-11-03:
@engcang Is there something unclear about the thrust calculations? It is simply a proportional to the rotational velocity with the added rotor drag term

Jaeyoung-Lim, 2020-11-19:
@noman0511 Any updates?

Jaeyoung-Lim, 2020-11-19:
Closing since this seems stale. Please feel free to reopen if you are still struggling with this issue

closed 2020-11-19 by Jaeyoung-Lim

bilalmubdir, 2024-01-17:
Dears @Jaeyoung-Lim, and

I'm working on Simulating the IRIS+ quadrotor but I struggled with its parameters, I didn't find any published work that reports its parameters, such as the moment of inertia around three axes or the motor parameters. Could you help me with that, please? and if the parameters are already verified and published in any paper I would appreciate it.

Thank you in advance