PX4/PX4-SITL_gazebo-classic #638: geotagging plugin not working

igorsgcampos, 2020-10-21:
I am trying to use PX4 SITL to validate an enhancement for survey missions, but I need to have the images geotagged, unfortunately, they are coming without it.

I tried to use the log file and the images but it didn’t work either.

I noticed that the instructions that mention geotagged images refer to the plane_cam model, but the typhoon_h480 (which I am using) uses the same plugin:

```
<plugin name="GeotaggedImagesPlugin" filename="libgazebo_geotagged_images_plugin.so">
    <robotNamespace>typhoon_h480</robotNamespace>
    <interval>1</interval>
    <width>3840</width>
    <height>2160</height>
    <maximum_zoom>8.0</maximum_zoom>
</plugin>
```

I would deeply appreciate if someone could help me to get this working. I plan to run the geotagged images of mcmillan airfield with different parameters on a photogrammetry software and compare the results, hence the use of SITL, it gives me an 100% controlled environment to evaluate the performance differences.

igorsgcampos, 2020-10-21:
There is a warning, but I think it is not related to it:

[Wrn] [Publisher.cc:185] Queue limit reached for topic /gazebo/default/user_camera/pose, deleting message. This warning is printed only once.

igorsgcampos, 2020-10-21:
I tried with both the PX4 recommended version of sitl_gazebo and the latest master of this submodule. I am running it with the following command:

make px4_sitl gazebo_typhoon_h480__mcmillan_airfield

Jaeyoung-Lim, 2020-10-22:
@igorsgcampos The geotagging should be working. Are you sure you have exif tools installed? ( as written in the readme)
```
sudo apt-get install libimage-exiftool-perl
```

igorsgcampos, 2020-10-23:
Yes, they are properly installed. All dependencies listed in the readme are.

Jaeyoung-Lim, 2020-10-23:
@igorsgcampos What does https://github.com/PX4/sitl_gazebo/blob/9f832e596a9aaee724d2fce9859e473624e3f87e/src/gazebo_geotagged_images_plugin.cpp#L230 return?

igorsgcampos, 2020-10-23:
the messages around the image capture messages are as follows:

```
[Dbg] [gazebo_geotagged_images_plugin.cpp:478] Handle Start Capture
[Msg] Took picture: frames/DSC00001.jpg
[Dbg] [gazebo_geotagged_images_plugin.cpp:271] Done with image capture
[Dbg] [gazebo_geotagged_images_plugin.cpp:704] Send capture status
    1 image files updated
```

igorsgcampos, 2020-10-23:
These are the messages around the image captures:

```
[Dbg] [gazebo_geotagged_images_plugin.cpp:478] Handle Start Capture
[Msg] Took picture: frames/DSC00011.jpg
[Dbg] [gazebo_geotagged_images_plugin.cpp:271] Done with image capture
[Dbg] [gazebo_geotagged_images_plugin.cpp:704] Send capture status
    1 image files updated
```

So it is not returning -1.

igorsgcampos, 2020-10-23:
I did some further investigation on my own, it seems that the image is being geotagged, but without coordinates.

The gps topic is being published, it works normally if I subscribe to it with: 
`gz topic -e /gazebo/default/typhoon_h480/gps/link/gps`

I also uncommented/fixed the following line:
`gzdbg << "got gps pos: " << _lastGpsPosition.X() << ", " << _lastGpsPosition.Y() <<endl;`
in: 
`GeotaggedImagesPlugin::OnNewGpsPosition`

and it never enters the scope of this function.

igorsgcampos, 2020-10-24:
Found the issue: the publisher appends a /link/ to the topic before gps, but the subscriber wasn't appending it, so it wasn't being called. What I couldn't figure out is why I could echo the topic with `gz topic -e /gazebo/default/typhoon_h480/gps/link/gps`, but couldn't with `gz topic -e /gazebo/default/typhoon_h480/link/gps`

https://github.com/PX4/sitl_gazebo/blob/9f832e596a9aaee724d2fce9859e473624e3f87e/src/gazebo_gps_plugin.cpp#L219

https://github.com/PX4/sitl_gazebo/blob/9f832e596a9aaee724d2fce9859e473624e3f87e/src/gazebo_geotagged_images_plugin.cpp#L144

PR #640 (2020-10-24): fix #638 - geotagging plugin not working: subscribed to wrong topic
The geotagging plugin was subscribing to the wrong topic, and the callback was never getting called. This fixes #638.
commit 2451437c (2020-10-24): fix #638 - geotagging plugin not working: subscribed to wrong topic
closed 2020-10-24 by TSC21