PX4/PX4-SITL_gazebo-classic #639: Cannot use UAV motor test command

leocencetti, 2020-10-21:
Sending the MAVLINK command `MAV_CMD_DO_MOTOR_TEST` with  `id=209`  results in the following warning:
```
[ WARN] [1603312628.991594762, 48.276000000]: CMD: Unexpected command 209, result 3
```
Reproducible using QGroundControl's Motor Setup function.

Full log here: https://pastebin.com/kt44QzNS

Jaeyoung-Lim, 2020-10-22:
@leocencetti Is this regarding SITL? or PX4 in general?

leocencetti, 2020-10-22:
> @leocencetti Is this regarding SITL? or PX4 in general?

It is regarding SITL only.

> I don't think the motor test command is supposed to work in SITL Gazebo

That was my conclusion as well... do you know if is there any other way to mimic the same behavior (one motor spinning at the time)?

Jaeyoung-Lim, 2020-10-22:
@leocencetti What are you trying to achieve?

If you are interested in adding the functionality, contributions are welcome :smile:

leocencetti, 2020-10-22:
> @leocencetti What are you trying to achieve?

We are developing a custom calibration routine that requires each motor to spin at a certain speed for a given time (we want to validate it in SITL before deploying on hardware). `motor_test` seemed a good starting point, although I know that the test time is not customizable.

> If you are interested in adding the functionality, contributions are welcome

:wink:

Jaeyoung-Lim, 2020-10-28:
@leocencetti Would you be able to move this forward?

leocencetti, 2020-10-28:
> @leocencetti Would you be able to move this forward?

@Jaeyoung-Lim unfortunately no, not in a reasonable time frame.

Jaeyoung-Lim, 2020-10-28:
@leocencetti Okay, closing then

closed 2020-10-28 by Jaeyoung-Lim