PX4/PX4-SITL_gazebo-classic #642: Gazebo simulation showing a shade.

igorsgcampos, 2020-10-28:
I am having an issue with the simulation in sonoma raceway where the image appears too dark. The shade moves as I change the camera orientation. Please help me fix it as I am trying to get images to run in metashape for comparison of surveying techniques. This is what I am getting:

[image]

igorsgcampos, 2020-10-28:
adding the following after the sun is included helps, but then I get no shadows from the objects... What I don't know is what is causing this massive shadow.
```
    <scene>
      <ambient>0.9 0.9 0.9 1</ambient>
      <shadows>false</shadows>
    </scene>
```

engcang, 2020-10-29:
Can you attach full world file here?
I guess there is an improper ***light source***. You should remove/change it/them.

Jaeyoung-Lim, 2020-10-29:
@igorsgcampos I am aware of this problem, and this got worse with gazebo11

bperseghetti, 2020-11-01:
@igorsgcampos @Jaeyoung-Lim @engcang by default when you run the world, the pose of the sun model is not given, it is simply "included" meaning it should spawn at [0,0,0], cross fingers. The sun models light source is offset 10 meters in +Z inside it's sdf, this means that you have a light source at around ~10 meters from the ground, light sources 10m from the ground can make some pretty impressive shadows. Furthermore, there is no scene values being set, so it can either inherit them from an existing model or go with an internal default. Lastly, the vector given inside the sun models sdf for direction makes for large shadows in the "X-dimension".

This is not a gazebo 11 issue or bug, it's simply a poor choice of hiding the light source by wrapping it in another model in addition to not setting the default scene settings in the world file. This is most easily solved by creating a different sun model "sun_2" in this case with better params for "sunny solar apex near the equator day". This still results in shadows but they are more realistic (unless you live at one of the poles).
Here it is with the scenes ambient and background set to zero but with correct light source from above, as you can see the sides of the buildings are not illuminated as there is no ambient and background:
[image]

To fix this simply add a reasonable scene ambient and background values to the world file as you can see below:
[image]

I will shortly push up this fix with the sun_2 model and updated world files to the [pr-hitl-capable-jinja](https://github.com/PX4/PX4-SITL_gazebo/tree/pr-hitl-capable-jinja) branch.

igorsgcampos, 2020-11-01:
@engcang In the screenshot I was using the standard sonoma_racewa world with the standard sun.

@Jaeyoung-Lim Ubuntu 18.04 / Gazebo 9

https://github.com/PX4/PX4-SITL_gazebo/blob/75a8b031f69f8994661e2f1b2ee4d9905353117a/models/sun/model.sdf#L5
@bperseghetti Actually, The standard sun is set to 0,0,10. In the screenshot I attached it is possible to see that there is shadow where there wasn't supposed to be any. Furthermore, the shadow changes when I move the camera around.

I tried moving it to 1000 to see if the behavior would change, but nothing changed.

bperseghetti, 2020-11-01:
@igorsgcampos did you get a chance to read everything I posted above?

bperseghetti, 2020-11-01:
Try using the sun_2 model from that branch and the updated world file where I have the scene values modified. Note that I am on 20.04 and using Gazebo 11. I know that Gazebo in earlier versions struggled with scene settings and light sources. Most of the annoying issues I encountered with Gazebo 9 were fixed in Gazebo 11.

igorsgcampos, 2020-11-01:
I did read, but what I could not understand is why when I moved the sun up, it didn`t change, and why does the shadow change when I move the camera? I will try the sun_2 from the branch you mentioned later and report back.

bperseghetti, 2020-11-01:
As a way to eliminate the massive calculations required otherwise, when zoomed out the shadows disappear, In perspective mode they are also "warped" by the lens you are viewing them through. The thing you need to make sure you also have is the "scene block" that I mentioned adding to the world file.

Don't forget to copy the world file from my branch that I mentioned above.

bperseghetti, 2020-11-06:
@igorsgcampos this fixed the issue on my side as you can see from my previous screen shots, does this appear to fix the issue on your side as well?

Jaeyoung-Lim, 2020-11-21:
Also related to https://github.com/osrf/gazebo/issues/2854

closed 2020-12-30 by Jaeyoung-Lim