PX4/PX4-SITL_gazebo-classic #643: Getting higher rate of sensor message topics and position topics

engcang, 2020-10-29:
First of all, I really appreciate your great work.
I am trying to get higher rate of ROS topics for sensor messages and position messages 
and I guess many other people tried, but could not make it.

1: I edited
***Firmware/ROMFS/px4fmu_common/init.d-posix/rcS*** file's ***GCS link*** block
into 
~~~xml
mavlink stream -r 100 -s LOCAL_POSITION_NED -u $udp_gcs_port_local
mavlink stream -r 100 -s HIGHRES_IMU -u $udp_gcs_port_local
mavlink stream -r 100 -s ATTITUDE -u $udp_gcs_port_local
mavlink stream -r 100 -s ATTITUDE_QUATERNION -u $udp_gcs_port_local
mavlink stream -r 100 -s ATTITUDE_TARGET -u $udp_gcs_port_local
~~~

and successfully checked that rates (Hz) of those are changed via ***Mavlink Inspector*** of ***QGroundControl***

But still getting fixed 50.0 hz for ***/mavros/imu/data***, ***/mavros/imu/data_raw***, and 30.0 hz for ***/mavros/local_position/pose***

I first thought this is one of issues from ***MAVROS*** but saw [this comment from TSC21](https://github.com/mavlink/mavros/issues/980#issuecomment-386744106).

2: I also tried to run
~~~shell
$ rosrun mavros mavsys rate --all 100
or
$ rosservice call /mavros/set_stream_rate "stream_id: 30 message_rate: 100 on_off: true"
30~32, 105 for stream_id and true/false for on_off both.
~~~
But failed to increase the rates of topics.

3: I checked the gazebo_mavlink source files and header files,
***Firmware/Tools/sitl_gazebo/src/gazebo_imu_plugin.cpp*** and ***Firmware/Tools/sitl_gazebo/src/gazebo_mavlink_interface.cpp***, and saw they are trying to send data at full rate as [here](https://github.com/PX4/PX4-SITL_gazebo/blob/master/src/gazebo_mavlink_interface.cpp#L521), which makes me more confused.

I believe many other people struggled to make it but could not get any good news.
I want to know if there is specific way to increase the rates of ROS topics of sensor msgs and positions of SITL-Gazebo.
Thank you in advance.

Jaeyoung-Lim, 2020-10-29:
@engcang What are you trying to achieve? How high are you trying to subscribe to imu topics? the `gcs_port_local` is the wrong port to touch, rather you need to change `offboard_port_remote`

engcang, 2020-10-29:
> @engcang What are you trying to achieve? How high are you trying to subscribe to imu topics?
> 
> * the `gcs_port_local` is the wrong port to touch, rather you need to change `offboard_port_remote` or the `sdk_port_remote`
> * QGC and mavros use different udp ports(or it should be on different ports) to subscribe to mavlink messages, so you shouldn't be mixing the two

@Jaeyoung-Lim 
Thanks for fast reply.
I am trying to get 100 Hz of ***/mavros/imu/data_raw*** and 100 Hz of ***/mavros/local_position/pose***

For the cases of actual drones, I successfully changed those rates via SD card's ***etc/extras.txt*** file.

Jaeyoung-Lim, 2020-10-29:
@engcang Which part do you not understand?

engcang, 2020-10-29:
Gotcha!!!!

I did not understand the difference among `gcs_port_local`, `offboard_port_remote`, `sdk_port_remote`.

`offboard_port_remote` did not work, but instead `offboard_port_local` worked for me.
(and `sdk_port_remote` does not exist in the ***rcS*** file for me)

For someone else who may need this in the future, I attached my
***Firmware/ROMFS/px4fmu_common/init.d-posix/rcS*** file 
[here](https://github.com/engcang/mavros-gazebo-application/blob/master/rcS)

Thank you @Jaeyoung-Lim !
감사합니다!!

Jaeyoung-Lim, 2020-10-29:
@engcang Glad to hear that it worked for you!

engcang, 2020-11-09:
@Jaeyoung-Lim 
Sorry for invoking the same issue again.

Recently, I newly installed Ubuntu and Firmware, SITL clearly again and I used the same ***rcS file*** as [here](https://github.com/engcang/mavros-gazebo-application/blob/master/rcS), however, this does not work suddenly.

Using the same rcS file publishes the default hz for `imu/data` and `local_position/pose`, even no messages from `imu/data_raw`.

Is there any changes from recent commits of `SITL` pacakge?

Thank you.

Jaeyoung-Lim, 2020-11-09:
@engcang Probably you are hitting the change https://github.com/PX4/PX4-Autopilot/pull/15714 which, the rcS is now being copied into the ROMFS when running SITL.

Therefore to apply your changes in rcS you need to
1. Rebuild everytime you edit your rcS so that it gets re-copied into the ROMFS
2. edit the rcS file which SITL uses inside the ROMFS, which is `${FIRMWARE_DIR}/build/px4_sitl_default/etc/init.d-posix/rcS`

engcang, 2020-11-10:
@Jaeyoung-Lim  Thank you! It partially works!

As I edited `${FIRMWARE_DIR}/build/px4_sitl_default/etc/init.d-posix/rcS` file same as [here](https://github.com/engcang/mavros-gazebo-application/blob/master/rcS).

1. I got 100 hz of `/mavros/imu/data` and `/mavros/local_position/pose`
2. But, No messages come from `/mavros/imu/data_raw`, is there any change for advertising ***imu data***??

Thank you again!!!

Jaeyoung-Lim, 2020-11-10:
@engcang Probably, have you compared the diff of the current rcS and before?

engcang, 2020-11-10:
@Jaeyoung-Lim I think I used [this](https://github.com/PX4/PX4-Autopilot/blob/458420f9cbaa55b84243d2f13f2e69d9236c37d8/ROMFS/px4fmu_common/init.d-posix/rcS) commit before, when ROMFS/px4fmu_common/init.d-posix did not have `airframes` folder.
It seems that some [parameters](https://github.com/PX4/PX4-Autopilot/blob/458420f9cbaa55b84243d2f13f2e69d9236c37d8/ROMFS/px4fmu_common/init.d-posix/rcS#L124) are different with now.

otherwise, I could not find diff

Jaeyoung-Lim, 2020-11-10:
Yes exactly, that is the change that I sent you the link above that is changing the behavior

engcang, 2020-11-10:
@Jaeyoung-Lim Oh, I got that, thanks. Then what should I do now?
The changed behavior makes `/mavros/imu/data_raw` disabled? or I still can get the data through the specific way?

Jaeyoung-Lim, 2020-11-10:
No, you need to adapt your rcS to the recent changes

engcang, 2020-11-10:
I am so sorry to bother you continuously @Jaeyoung-Lim ,
Thanks to your comment, I got the newest [rcS file](https://github.com/PX4/PX4-Autopilot/blob/master/ROMFS/px4fmu_common/init.d-posix/rcS) from original repo and then changed it again as like [this](https://github.com/engcang/mavros-gazebo-application/blob/master/rcS), which just added this block

~~~sh
mavlink stream -r 100 -s LOCAL_POSITION_NED -u $udp_offboard_port_local
mavlink stream -r 100 -s HIGHRES_IMU -u $udp_offboard_port_local
mavlink stream -r 100 -s ATTITUDE -u $udp_offboard_port_local
mavlink stream -r 100 -s ATTITUDE_QUATERNION -u $udp_offboard_port_local
~~~

and not getting **data_raw** yet. I think I adapt the rcS to the recent change. Did I miss something?

I really appreciate!!!

Jaeyoung-Lim, 2020-11-10:
@engcang If you are still confused, feel free to hop on https://discuss.px4.io/t/korean-regional-dev-call-november-11-2020/19526 so we can discuss it online

Jaeyoung-Lim, 2020-11-19:
@engcang Any updates?

engcang, 2020-11-19:
@Jaeyoung-Lim Thanks for your attention. Well, actually I can just use `/mavros/imu/data` instead of `/mavros/imu/data_raw`.
So I just gave up haha.
I don't know how to use the link you gave me above. Do I just reply for the question there?

Jaeyoung-Lim, 2020-11-19:
@engcang No, that is the link for the Korean regional dev call, which happens every Wednesday 4pm(KST) since I know that you are korean, you can come up to the call and we can discuss it face to face to speed up the discussion.

Otherwise, your problem is that you are copying the rcS to the wrong location

engcang, 2020-11-19:
@Jaeyoung-Lim Ahhhh that sounds great. Let me jump in that dev call on the next Wednesday. Thank you!

Jaeyoung-Lim, 2020-12-15:
@engcang Any updates regarding this issue?

engcang, 2020-12-15:
@Jaeyoung-Lim Hi Jaeyoung.

Well, It is weird.
Sometimes `/mavros/imu/data_raw' comes in and sometimes it does not.
Even with same `iris.sdf`, same `rcS` file, same computer.

Now I just gave up and am using `/mavros/imu/data` instead!

Jaeyoung-Lim, 2020-12-15:
Okay, closing then

closed 2020-12-15 by Jaeyoung-Lim