PX4/PX4-SITL_gazebo-classic #645: mavlink_interface compile warning, possible bug?

George-avy, 2020-10-30:
Hello everyone!

When I'm compiling the mavlink_interface object, I get an error for this line:
https://github.com/PX4/PX4-SITL_gazebo/blob/06e801fe8b5267e561f00d8847a6343cb2386ab6/src/mavlink_interface.cpp#L439

The error message is:
```
<project_dir>/src/mavlink_interface.cpp: In member function ‘void MavlinkInterface::send_mavlink_message(const mavlink_message_t*)’:
<project_dir>/src/mavlink_interface.cpp:513:21: warning: comparison of unsigned expression < 0 is always false [-Wtype-limits]
  513 |             if (len < 0)
      |                 ~~~~^~~
```
Should < be == perhaps? What is the error result of the preceding `send` function?

dagar, 2020-10-30:
Both send and sendto return a ssize_t, so we need to fix `len`.

dagar, 2020-10-30:
https://github.com/PX4/PX4-SITL_gazebo/pull/646

PR #646 (2020-11-01): mavlink interface len is ssize_t
 - fixes #645
closed 2020-11-01 by Jaeyoung-Lim