PX4/PX4-SITL_gazebo-classic #647: Gazebo Motor Model is reading wrong joint velocities

aihamalharbat, 2020-10-31:
Hi everyone, 

I am facing a problem with the `gazebo_motor_model`.  

### Setup 

My setup: 

- Ubuntu 18.04 LTS 

- ROS Melodic  

- Gazebo v9.15.0 

- Firmware v1.11.1 

- LockStep is disabled from `Firmware/boards/px4/sitl/default.cmake` & UAV `.sdf` file 

- All tests are on a modified version of Typhoon_h480, but the problem exists with the original typhoon model. But this problem does not exist when testing with iris. 

- [This](https://github.com/aihamalharbat/sitl_gazebo/blob/peach_sitl/models/typhoon_h480/typhoon_h480.sdf) is the modified typhoon_h480 model (the UAV is called peach) 

## Problem Description  

I first noticed that the rotors are not generating the thrust that I am expecting, I started investigating it and ended up with these conclusions: 

- `gazebo_motor_model` loads all the parameters correctly from the `sdf` file 

- The problem is with reading the joint velocity in: 

https://github.com/PX4/PX4-SITL_gazebo/blob/06e801fe8b5267e561f00d8847a6343cb2386ab6/src/gazebo_motor_model.cpp#L196 

The above line does not get the correct rotational velocity of the joint compared to three other sources: 

-  The value that this method gets 

https://github.com/PX4/PX4-SITL_gazebo/blob/06e801fe8b5267e561f00d8847a6343cb2386ab6/src/gazebo_motor_model.cpp#L34-L38 

- The reference rotational velocities of the joints from: 

https://github.com/PX4/PX4-SITL_gazebo/blob/06e801fe8b5267e561f00d8847a6343cb2386ab6/src/gazebo_motor_model.cpp#L184-L189 

- What `gazebo` publishes in `/gazebo/link_states` 

## Make sure 

To verify that I added a simple `cout` after both calls of `joint_->GetVelocity(0)` in [line:35](https://github.com/PX4/PX4-SITL_gazebo/blob/06e801fe8b5267e561f00d8847a6343cb2386ab6/src/gazebo_motor_model.cpp#L35) and [Line:196](https://github.com/PX4/PX4-SITL_gazebo/blob/06e801fe8b5267e561f00d8847a6343cb2386ab6/src/gazebo_motor_model.cpp#L196) respectively as follows: 

``` 

std::cout << "[MotModel-Publish]omega_z: " << joint_->GetVelocity(0) << std::endl; 

std::cout << "[MotModel-UpdateFM]motor_rot_vel_: " << joint_->GetVelocity(0) << std::endl; 

``` 

and this is the streamed output, with the thrust command set at 0.3: 

``` 

[MotModel-UpdateFM]motor_rot_vel_: 36.6667 

[MotModel-Publish]omega_z: 55 

``` 

  

With a `MaxRotVelocity` of 1500 and `idle_speed` of 100, the rotational speed of the joint at thrust = 0.3 is indeed = omega_z = 550 rad/s (after accounting for `rotorVelocitySlowdownSim`) not 366. 

and with the thrust command set at 1.0: 

``` 

[MotModel-UpdateFM]motor_rot_vel_: 100 

[MotModel-Publish]omega_z: 150 

``` 

When testing with the original typhoon model, this is the streamed output, with the thrust command set at 0.3: 

``` 

[MotModel-UpdateFM]motor_rot_vel_: 51.2 

[MotModel-Publish]omega_z: 55 

``` 

and with the thrust command set at 1.0: 

``` 

[MotModel-UpdateFM]motor_rot_vel_: 139.717 

[MotModel-Publish]omega_z: 150 

``` 

So the problem exist with the original typhoon model, but it is much worse with my modified model. **However, this problem does not exist when using `iris` model.**

## Temp Solution 

So I temporarily solved the problem by using the value from [line:35](https://github.com/PX4/PX4-SITL_gazebo/blob/06e801fe8b5267e561f00d8847a6343cb2386ab6/src/gazebo_motor_model.cpp#L35) for `UpdateForcesAndMoments` calculations, instead of the value from [Line:196](https://github.com/PX4/PX4-SITL_gazebo/blob/06e801fe8b5267e561f00d8847a6343cb2386ab6/src/gazebo_motor_model.cpp#L196). 

However, that means that the force is always generated based on the rotational velocity of the previous iteration, because in each iteration, `Publish` is called after `UpdateForcesAndMoments`: 

https://github.com/PX4/PX4-SITL_gazebo/blob/06e801fe8b5267e561f00d8847a6343cb2386ab6/src/gazebo_motor_model.cpp#L176-L182 

I tried to change their order, but the lead to both of them reading the wrong value of the rotational velocity. 

## My SDF file

Since I did not see this problem when I was using iris, I am thinking that my `sdf` model might be part of the problem, even though I think that it is used only to load the parameters of the plugin. So here is a summary of what is modified in in Typhoon H480 [model](https://github.com/aihamalharbat/sitl_gazebo/blob/peach_sitl/models/typhoon_h480/typhoon_h480.sdf):
- The mass, inertia, and pose.
- Removed the gimbal (and the controller plugin), except (cgo3_camera_link) this is detached from the UAV, but the model does not build without it(!).
- Disable lockstep.
- Collisions and meshes
- Motors pose, mass, inertia, ordering.

### Question 

I don't want to just rely on my temp solution, but find out what went wrong. Can anyone help with this? Any pointers to potential causes or suggestions?

Thanks in advance.

bperseghetti, 2020-11-01:
Without looking over it too much, a few suggestions to try:
First make it so the joint names match the motor channel and the rotor names/numbering, this makes it much easier to understand what things are connected to eachother. 

Second, create this as a new model. Create a "peach" folder and copy over the relevant meshes, modify the name in the .config to the new model name, get rid of the entire cgo3 link block, and add it to the firmware's posix init.d.

This will help others to copy/try it without having to worry about other locally installed versions of the Typhoon and the associated saved models/namespace collisions.

Jaeyoung-Lim, 2020-11-01:
@aihamalharbat The rotor joint velocities are being slowed down with the [`rotorVelocitySlowdownSim`](https://github.com/PX4/PX4-SITL_gazebo/blob/06e801fe8b5267e561f00d8847a6343cb2386ab6/src/gazebo_motor_model.cpp#L137), since gazebo has a hard time simulating rotor velocities at realtime

Therefore here is the real calculation is happening in https://github.com/PX4/PX4-SITL_gazebo/blob/06e801fe8b5267e561f00d8847a6343cb2386ab6/src/gazebo_motor_model.cpp#L201

aihamalharbat, 2020-11-01:
@Jaeyoung-Lim 
Thank you for your reply.
As I said in "Make sure" section:
> With a MaxRotVelocity of 1500 and idle_speed of 100, the rotational speed of the joint at thrust = 0.3 is indeed = omega_z = 550 rad/s (after accounting for rotorVelocitySlowdownSim) not 366.

I am accounting for `rotorVelocitySlowdownSim`, and the problem is still there.

aihamalharbat, 2020-11-01:
@bperseghetti 
Thanks for the suggestions.
1- I wanted to fix the names-matching issue, I will do it soon.
2- The reason why I ended up editing the typhoon model instead of simply adding a new model as you mention is because when I tried these [steps](https://discuss.px4.io/t/create-custom-model-for-sitl/6700/4), I was never able to `make px4_sitl gazebo_peach`, I always had problems, so I will try that again as well.

bperseghetti, 2020-11-01:
Use: `DONT_RUN=1 make px4_sitl_default gazebo`

aihamalharbat, 2020-11-02:
@bperseghetti 

1. I cloned the Firmware into a new catkin ws, then I checked out the the v1.11.1 release
2. Then, I removed the files in `.ros/eeprom` & `DONT_RUN=1 make px4_sitl_default gazebo`
3. I tested the Firmware with the iris and typhoon model both in `make px4_sitl gazebo_iris` & `roslaunch px4 mavros_posix_sitl`
4. Iris and typhoon models can takeoff and land.
5. I added a model `peach_uav` following these {steps](https://discuss.px4.io/t/create-custom-model-for-sitl/6700/4) with these files 
[Peach_gazebo_sitl files.zip](https://github.com/PX4/PX4-SITL_gazebo/files/5473975/Peach_gazebo_sitl.files.zip)
6. `make clean` & removed the files in `.ros/eeprom`, then `DONT_RUN=1 make px4_sitl_default gazebo` & `make px4_sitl gazebo_iris` all works fine
7. Finally, `make px4_sitl gazebo_peach_uav`, it starts building and freezes here:

> $ make px4_sitl gazebo_peach_uav
> [0/4] Performing build step for 'sitl_gazebo'
> ninja: no work to do.
> [3/4] cd /home/aalharbat/sec_catkin_ws/src/Firmware/build/px4_sitl_default/tmp && ..._ws/src/Firmware /home/aalharbat/sec_catkin_ws/src/Firmware/build/px4_sitl_default
> SITL ARGS
> sitl_bin: /home/aalharbat/sec_catkin_ws/src/Firmware/build/px4_sitl_default/bin/px4
> debugger: none
> program: gazebo
> model: peach_uav
> world: none
> src_path: /home/aalharbat/sec_catkin_ws/src/Firmware
> build_path: /home/aalharbat/sec_catkin_ws/src/Firmware/build/px4_sitl_default
> GAZEBO_PLUGIN_PATH :/home/aalharbat/sec_catkin_ws/src/Firmware/build/px4_sitl_default/build_gazebo:/home/aalharbat/sec_catkin_ws/src/Firmware/build/px4_sitl_default/build_gazebo:/home/aalharbat/sec_catkin_ws/src/Firmware/build/px4_sitl_default/build_gazebo:/home/aalharbat/sec_catkin_ws/src/Firmware/build/px4_sitl_default/build_gazebo
> GAZEBO_MODEL_PATH :/home/aalharbat/sec_catkin_ws/src/Firmware/Tools/sitl_gazebo/models:/home/aalharbat/sec_catkin_ws/src/Firmware/Tools/sitl_gazebo/models:/home/aalharbat/sec_catkin_ws/src/Firmware/Tools/sitl_gazebo/models:/home/aalharbat/sec_catkin_ws/src/Firmware/Tools/sitl_gazebo/models
> LD_LIBRARY_PATH /home/aalharbat/sec_catkin_ws/devel/lib:/opt/ros/melodic/lib:/home/aalharbat/sec_catkin_ws/src/Firmware/build/px4_sitl_default/build_gazebo:/home/aalharbat/sec_catkin_ws/src/Firmware/build/px4_sitl_default/build_gazebo:/home/aalharbat/sec_catkin_ws/src/Firmware/build/px4_sitl_default/build_gazebo:/home/aalharbat/sec_catkin_ws/src/Firmware/build/px4_sitl_default/build_gazebo
> empty world, default world peach_uav.world for model found

I would wait for around 20 min, still frozen.

bperseghetti, 2020-11-02:
Why are you running make px4_sitl gazebo_iris (and also peach_uav)? 
All you need to do is `DONT_RUN=1 make px4_sitl_default gazebo` and then simply spawn the desired model you want, in this case peach_uav. You never need to run a separate make again. Simply just start the gz-server spawn the model and launch the gz-client. It seems like you are running this from a ros workspace? Why not just use catkin_make or catkin build?

Jaeyoung-Lim, 2020-11-12:
@bperseghetti 
If you run `DONT_RUN=1 make px4_sitl_default gazebo` you are just building the firmware and not running it. Therefore even if you spawn the model into gazebo, PX4 SITL binary will not be running

closed 2020-12-17 by Jaeyoung-Lim