PX4/PX4-SITL_gazebo-classic #648: Real time factor is only 0.6 when the usage of CPU and GPU is not great

robin-shaun, 2020-11-02:
Hello, as I know, Gazebo did computation mainly with only one CPU core in the past. However, when I use Gazebo 9.15, I'm glad to see that all CPU cores are used equally. However, for large scale simulation (30 UAVs), the real time factor is only 0.6 when the usage of CPU and GPU is not great. I don't know why.
[image]
[image]

bperseghetti, 2020-11-06:
The ODE solver needs to be allowed to create threads for each new device. My branch pr-hitl-capable-jinja has jinja world files that show what that would look like.

Jaeyoung-Lim, 2020-11-06:
@robin-shaun With lockstep, I believe your computer is not able to run 30 px4 instances in parallel in realtime with the gazebo simulation. You can try turning off lockstep and see if it helps. On my powerful desktop this is not the case

closed 2020-11-25 by Jaeyoung-Lim