PX4/PX4-SITL_gazebo-classic #656: Adding a sensor to read the swing angles of a suspended load

eldewany, 2020-11-16:
Hello, i'm trying to simulate a sensor to read the swing angles of a load that is attached to the center of the quadrotor and use this sensor to send the measured angles to a certain topic on the px4 system. I can't find any documentation on publishing gazebo data to mavros and the code in the mavlink interface is a bit complicated for me to track. It's worth noting that i modified the jinja file for the Iris quadrotor and i will add an imu to the link of the suspension wires but i don't know what to do from there.

Jaeyoung-Lim, 2020-11-17:
@eldewany You can look into tutorials on how to implement gazebo plugins: http://gazebosim.org/tutorials/?tut=plugins_hello_world and implement a sensor that reads out the swing angles.

Since the mavlink interface sends gazebo sensor values mavlink messages to firmware, you would need to possibly define a new mavlink interface to send the sensor values to the firmware and implement it in the gazebo mavlink interface

eldewany, 2020-11-18:
@Jaeyoung-Lim Thanks for the reply, the mavlink interface with gazebo is where i can't find any documentation. if you can link me to one or a tutorial concerning this matter i would be thankful.

Jaeyoung-Lim, 2020-11-18:
@eldewany This document explains how simulation / mavlink HIL messaging works: https://dev.px4.io/master/en/simulation/

Otherwise, you would need to look at the code

closed 2020-11-25 by Jaeyoung-Lim