PX4/PX4-SITL_gazebo-classic #660: No satellite when loading robot with xacro

linuxsocke, 2020-11-22:
Hi, 

I am currently loading in my robot description  with xacro and my mavlink doesnt see any satellites. I know xacro is now deprecated (why?) but is there any chance that it can be used aswell in the future? 

Loading in the libgazebo gps plugin looks like this in my xacros:

[image]
 
Looking into diagnotics I see:

    level: 0
    name: "convecdro_0/mavros: FCU connection"
    message: "connected"
    hardware_id: "udp://:14540@localhost:14580"
    values: 
      - 
        key: "Received packets:"
        value: "38390"
      - 
        key: "Dropped packets:"
        value: "0"
      - 
        key: "Buffer overruns:"
        value: "0"
      - 
        key: "Parse errors:"
        value: "0"
      - 
        key: "Rx sequence number:"
        value: "246"
      - 
        key: "Tx sequence number:"
        value: "0"
      - 
        key: "Rx total bytes:"
        value: "1238559"
      - 
        key: "Tx total bytes:"
        value: "268575"
      - 
        key: "Rx speed:"
        value: "2413.000000"
      - 
        key: "Tx speed:"
        value: "537.000000"
      - 
    level: 2
    name: "convecdro_0/mavros: GPS"
    message: "No satellites"
    hardware_id: "udp://:14540@localhost:14580"
    values: 
      - 
        key: "Satellites visible"
        value: "0"
      - 
        key: "Fix type"
        value: "0"
      - 
        key: "EPH (m)"
        value: "Unknown"
      - 
        key: "EPV (m)"
        value: "Unknown"
      - 
    level: 0
    name: "convecdro_0/mavros: Heartbeat"
    message: "Normal"
    hardware_id: "udp://:14540@localhost:14580"
    values: 
      - 
        key: "Heartbeats since startup"
        value: "495"
      - 
        key: "Frequency (Hz)"
        value: "1.000000"
      - 
        key: "Vehicle type"
        value: "Hexarotor"
      - 
        key: "Autopilot type"
        value: "PX4 Autopilot"
      - 
        key: "Mode"
        value: "MANUAL"
      - 
        key: "System status"
        value: "Standby"
      - 
    level: 2
    name: "convecdro_0/mavros: System"
    message: "Sensor health"
    hardware_id: "udp://:14540@localhost:14580"
    values: 
      - 
        key: "Sensor present"
        value: "0x1008002C"
      - 
        key: "Sensor enabled"
        value: "0x1021002C"
      - 
        key: "Sensor health"
        value: "0x000E002C"
      - 
        key: "3D magnetometer"
        value: "Ok"
      - 
        key: "absolute pressure"
        value: "Ok"
      - 
        key: "GPS"
        value: "Ok"
      - 
        key: "rc receiver"
        value: "Fail"
      - 
        key: "AHRS subsystem health"
        value: "Fail"

If you want I can also show the launch files and how I am loading the mavlink interface plugin. 

**System Information**
Ubuntu: 20.04
ROS: Neotic
PX4: 1.11.2

Jaeyoung-Lim, 2020-11-22:
@linuxsocke Because it is unable to talk to the `gazebo_mavlink_interface`, which sends the gps values to PX4 throught the HIL_GPS message. 

You need to attach your gps plugin to a sensor so that the mavlink interface plugin can figure out that it exists in your model. You can check the code how this works. https://github.com/PX4/PX4-SITL_gazebo/blob/a6698f5e7b4faa68b1ea297f821b15020aa4545e/src/gazebo_mavlink_interface.cpp#L382

linuxsocke, 2020-11-22:
@Jaeyoung-Lim I want to spawn a robot in ros-gazebo by providing a description using xacro. So when adding a gps it expects a gps_joint in the set of joints of the robot? I already tried to add a gps_link with a joint which I called like ${namespace}/gps_joint or simply gps_joint. Or am I just completely getting something wrong?

Jaeyoung-Lim, 2020-11-22:
@linuxsocke I don't think xacro will work at all with this repo, unless you convert the generated urdf file into a sdf file. 

Have you looked at the code I have sent you?

linuxsocke, 2020-11-22:
Ah ok I didnt see. Its expecting only sdf format.

So a sdf template for jinja is the only way.

closed 2020-11-22 by linuxsocke

Jaeyoung-Lim, 2020-11-22:
@linuxsocke You can also create a `urdf` file from xacro, then covert the urdf file to a sdf file through `gz sdf -p`.

However, I am not sure if this would be worth the effort

linuxsocke, 2020-11-22:
I think so too. It would be another story if the conversion can be somehow done also in the launch process.

So with a jinja template my custom mav has satellite now.

What are the advantages of jinja over simply using sdf files? I know it can be used to replace some parameters. But how do I e.g. set namespaces for my links?

crazy-robotics, 2025-03-27:
@linuxsocke  Have you solved it? I am getting `No valid position estimation` error because I don't have GPS,  following plugin is called in `libgazebo_gps_plugin.so` . Here is the [xacro file ](https://github.com/crazy-robotics/dronedoc/blob/master/dronedoc/models/iris_depth_camera/xacro_sitl/urdf/component_snippets.xacro#L127)

crazy-robotics, 2025-03-27:
@linuxsocke I was facing the same error [ref1](https://discuss.px4.io/t/no-valid-position-estimate/44476) [ref2](https://discuss.px4.io/t/gps-plugin-not-working-with-xacro-file-libgazebo-gps-plugin-so/44453) I was using default gps plugin 
```
  <xacro:macro name="gps_plugin_macro" params="namespace gps_noise">
    <gazebo>
      <plugin name='gps_plugin' filename='libgazebo_gps_plugin.so'>
        <robotNamespace/>
        <gpsNoise>1</gpsNoise>
        <linkName>gps0</linkName>
        <gpsSubTopic>/gps</gpsSubTopic>
      </plugin>
    </gazebo>
  </xacro:macro>
```
I was getting position error 

[image]

then I changed to following 
```
  <xacro:macro name="gps_plugin_macro" params="namespace gps_noise">
    <gazebo>
    <sensor name='gps' type='gps'>
    <pose>0 0 0 0 0 0</pose>
    <update_rate>5.0</update_rate>
    <always_on>true</always_on>
    <visualize>false<visualize>
      <plugin name='gps_plugin' filename='libgazebo_gps_plugin.so'>
        <robotNamespace/>
        <gpsNoise>1</gpsNoise>
        <gpsXYRandomWalk>2.0</gpsXYRandomWalk>
        <gpsZRandomWalk>4.0</gpsZRandomWalk>
        <gpsXYNoiseDensity>0.0002</gpsXYNoiseDensity>
        <gpsZNoiseDensity>0.0004</gpsZNoiseDensity>
        <gpsVXYNoiseDensity>0.2</gpsVXYNoiseDensity>
        <gpsZNoiseDensity>0.4</gpsZNoiseDensity>
        <linkName>gps0</linkName>
        <gpsSubTopic>/gps</gpsSubTopic>
      </plugin>
    </gazebo>
  </xacro:macro>
``` 
the error was solved 
[image]
also GPS count 

[image]