PX4/PX4-SITL_gazebo-classic #662: gazebo models for ros2 version

ldg810, 2020-12-03:
Hi, I am developing PX4 avoidance module for ROS2(dashing and foxy).

Currently, there is only gazebo models for ROS1. The .sdf files should be ported appropriately([https://discourse.ros.org/t/migrating-gazebo-plugins-to-ros2/10433](https://discourse.ros.org/t/migrating-gazebo-plugins-to-ros2/10433)).

I made .sdf of iris_depth_camera(actually it would not changed but depth_camera.sdf would be changed) for ROS2 version, but is there any rule/name conventions about considering ROS2 version files?

Jaeyoung-Lim, 2020-12-08:
@ldg810 We need to support models for ROS2, but also support ROS1 at the same time. I suggest we need different plugins for each

Jaeyoung-Lim, 2020-12-16:
@ldg810 Any updates?

ldg810, 2020-12-17:
> @ldg810 Any updates?

Hi, did you mean new branch('ros2') for PX4-SITL_gazebo plugin? (as you mentioned "we need different plugins for each").

I am trying to make px4.launch.py based on [this file](https://github.com/osrf/ros2_serial_example/blob/master/ros2_serial_example/launch/px4.launch.py) but there are sophisticated problems to implement it.. like
- build and running "micrortps_agent" in PX4 firmware is not stable (so I made totally different micrortps_agent with @stmoon  and use internally.)
- I don't have good idea for launching PX4 in launch.py file. Can you suggest some idea about ros2 launch file and implement it to ros2 branch?

Jaeyoung-Lim, 2020-12-17:
> build and running "micrortps_agent" in PX4 firmware is not stable (so I made totally different micrortps_agent with @stmoon and use internally.)

Obviously, It is hard to support you if I don't have access to what you are running :smile: Therefore everything in this repo will be against the standard implementation and you are on your own.

>  don't have good idea for launching PX4 in launch.py file. Can you suggest some idea about ros2 launch file and implement it to ros2 branch?

I don't think having a separate ros2 branch is a good idea, but maybe having ros2 specific plugins would make sense: http://gazebosim.org/tutorials?tut=ros2_overview This does not require a launchfile

ldg810, 2020-12-17:
> Obviously, It is hard to support you if I don't have access to what you are running 😄 Therefore everything this repo will be against the standard implementation and you are on your own with yours. Is there something stopping you from fixing the standard `micrortps_agent`?

The current micrortps_agent is completely depends on fastrtps. For example, the [Subscriber.cpp.em](https://github.com/PX4/PX4-Autopilot/blob/master/msg/templates/urtps/Subscriber.cpp.em) includes fastrtps headers like this

```
#include <fastrtps/participant/Participant.h>
#include <fastrtps/attributes/ParticipantAttributes.h>
#include <fastrtps/subscriber/Subscriber.h>
#include <fastrtps/attributes/SubscriberAttributes.h>

#include <fastrtps/Domain.h>
```

In result, if fastrtps version was changed, some configuration variables are newly created or disappeared and make build fail. For example, domainId in following parts.
```
@[if version.parse(fastrtps_version) < version.parse('2.0')]@
    PParam.rtps.builtin.domainId = 0;
@[else]@
    PParam.domainId = 0;
```

Thus, I and @stmoon  made micrortps_agent solely depends on ros2 (i.e. include rclcpp/rclcpp.hpp)

We didn't elaborate the code for making pull requests but maybe we will make PR about this. (we didn't have much time to do it :( )

> > don't have good idea for launching PX4 in launch.py file. Can you suggest some idea about ros2 launch file and implement it to ros2 branch?
> 
> This repo in general does not have any ROS dependency except for some specific plugins. Therefore, I don't think having a separate ros2 branch is a good idea. Maybe having ros2 specific plugins would make sense: http://gazebosim.org/tutorials?tut=ros2_overview This does not require a launchfile.

I don't have much knowledge about ros2 specific plugins so I would study it :) Thank you.

Jaeyoung-Lim, 2020-12-17:
> Thus, I and @stmoon made micrortps_agent solely depends on ros2 (i.e. include rclcpp/rclcpp.hpp)
> We didn't elaborate the code for making pull requests but maybe we will make PR about this. (we didn't have much time to do it :( )

Still, I cannot help you on your proprietary software

> I don't have much knowledge about ros2 specific plugins so I would study it :) Thank you.

What exact plugin are you looking for?

Jaeyoung-Lim, 2021-02-01:
@ldg810 Are there any progress on this front? I would be willing to provide the CI test infrastructure if there are any progress on this front

Jaeyoung-Lim, 2021-02-05:
Discussed offline with @ldg810 

- @ldg810 will create a ROS2 plugin, which will check if the package is being built in a ROS1/ROS2 environment, and then configure the jinja template in order to select which one to select

@TSC21 FYI

Jaeyoung-Lim, 2021-02-22:
@ldg810 Could you make the cmake fix as mentioned in #707 ?

ldg810, 2021-02-22:
> @ldg810 Could you make the cmake fix as mentioned in #707 ?

Yes, I am in short vacation(kindergarten is in vacation..) now so I am planning to fix it until next week. Sorry for delay :)

Jaeyoung-Lim, 2021-02-22:
@ldg810 Not in a rush! Just didn't want this to be forgotten :smiley:

Jaeyoung-Lim, 2021-03-22:
Fixed in https://github.com/PX4/PX4-SITL_gazebo/pull/707

closed 2021-03-22 by Jaeyoung-Lim