PX4/PX4-SITL_gazebo-classic #664: 'make px4_sitl_default gazebo' error

cuizhi, 2020-12-08:
when 'make px4_sitl_default gazebo', gazebo can't be started. while 'make px4_sitl_default ' can be made successfully.
my environment is ubuntu 20.04 + noetic+gazebo11, i install the ros and gazebo by 'sudo apt install ros-noetic-desktop-full'.

this is my .bashrc:
source /usr/share/gazebo-11/setup.sh 
source /home/ustb/Firmware/Tools/setup_gazebo.bash /home/ustb/Firmware /home/ustb/Firmware/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:/home/ustb/Firmware
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:/home/ustb/Firmware/Tools/sitl_gazebo

this is the 'make px4_sitl_default gazebo':
[0/4] Performing build step for 'sitl_gazebo'
ninja: no work to do.
[3/4] cd /home/ustb/Firmware/build/px4_sitl_default/tmp && /...ome/ustb/Firmware /home/ustb/Firmware/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/ustb/Firmware/build/px4_sitl_default/bin/px4
debugger: none
program: gazebo
model: none
world: none
src_path: /home/ustb/Firmware
build_path: /home/ustb/Firmware/build/px4_sitl_default
empty model, setting iris as default
GAZEBO_PLUGIN_PATH /usr/lib/x86_64-linux-gnu/gazebo-11/plugins::/home/ustb/Firmware/build/px4_sitl_default/build_gazebo:/home/ustb/Firmware/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH /usr/share/gazebo-11/models::/home/ustb/Firmware/Tools/sitl_gazebo/models:/home/ustb/Firmware/Tools/sitl_gazebo/models
LD_LIBRARY_PATH /home/ustb/px4_ws/devel/lib:/opt/ros/noetic/lib:/usr/lib/x86_64-linux-gnu/gazebo-11/plugins:/home/ustb/Firmware/build/px4_sitl_default/build_gazebo:/home/ustb/Firmware/build/px4_sitl_default/build_gazebo
empty world, setting empty.world as default
gzserver: /usr/include/boost/smart_ptr/shared_ptr.hpp:734: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = gazebo::event::Connection; typename boost::detail::sp_member_access<T>::type = gazebo::event::Connection*]: Assertion `px != 0' failed.
/home/ustb/Firmware/Tools/sitl_run.sh: line 124: 29902 Aborted                 (core dumped) gzserver "${src_path}/Tools/sitl_gazebo/worlds/empty.world"
SITL COMMAND: "/home/ustb/Firmware/build/px4_sitl_default/bin/px4" "/home/ustb/Firmware"/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t "/home/ustb/Firmware"/test_data
INFO  [px4] Creating symlink /home/ustb/Firmware/ROMFS/px4fmu_common -> /home/ustb/Firmware/build/px4_sitl_default/tmp/rootfs/etc

______  __   __    ___
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [param] selected parameter default file eeprom/parameters_10016
[param] parameter file not found, creating eeprom/parameters_10016
  SYS_AUTOCONFIG: curr: 0 -> new: 1
* SYS_AUTOSTART: curr: 0 -> new: 10016
  BAT_N_CELLS: curr: 0 -> new: 3
  CAL_ACC0_ID: curr: 0 -> new: 1311244
  CAL_ACC_PRIME: curr: 0 -> new: 1311244
  CAL_GYRO0_ID: curr: 0 -> new: 2294028
  CAL_GYRO_PRIME: curr: 0 -> new: 2294028
  CAL_MAG0_ID: curr: 0 -> new: 197388
  CAL_MAG_PRIME: curr: 0 -> new: 197388
  COM_DISARM_LAND: curr: 2.0000 -> new: 0.5000
  COM_OBL_ACT: curr: 0 -> new: 2
  COM_RC_IN_MODE: curr: 0 -> new: 1
  EKF2_ANGERR_INIT: curr: 0.1000 -> new: 0.0100
  EKF2_GBIAS_INIT: curr: 0.1000 -> new: 0.0100
  COM_ARM_EKF_AB: curr: 0.0017 -> new: 0.0050
  EKF2_REQ_GPS_H: curr: 10.0000 -> new: 0.5000
  MC_PITCH_P: curr: 6.5000 -> new: 6.0000
  MC_PITCHRATE_P: curr: 0.1500 -> new: 0.2000
  MC_ROLL_P: curr: 6.5000 -> new: 6.0000
  MC_ROLLRATE_P: curr: 0.1500 -> new: 0.2000
  MPC_HOLD_MAX_Z: curr: 0.6000 -> new: 2.0000
  MPC_Z_VEL_I: curr: 0.1000 -> new: 0.1500
  MPC_Z_VEL_P: curr: 0.2000 -> new: 0.6000
  MPC_XY_P: curr: 0.9500 -> new: 0.8000
  MPC_XY_VEL_P: curr: 0.0900 -> new: 0.2000
  MPC_XY_VEL_D: curr: 0.0100 -> new: 0.0160
  MPC_SPOOLUP_TIME: curr: 1.0000 -> new: 0.5000
  MPC_TKO_RAMP_T: curr: 3.0000 -> new: 1.0000
  NAV_ACC_RAD: curr: 10.0000 -> new: 2.0000
  NAV_DLL_ACT: curr: 0 -> new: 2
  RTL_DESCEND_ALT: curr: 30.0000 -> new: 5.0000
  RTL_LAND_DELAY: curr: -1.0000 -> new: 5.0000
  RTL_RETURN_ALT: curr: 60.0000 -> new: 30.0000
  SDLOG_MODE: curr: 0 -> new: 1
  SDLOG_PROFILE: curr: 3 -> new: 131
  SDLOG_DIRS_MAX: curr: 0 -> new: 7
  SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
  SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
  TRIG_INTERFACE: curr: 4 -> new: 3
  COM_CPU_MAX: curr: 90.0000 -> new: -1.0000
* RTL_DESCEND_ALT: curr: 5.0000 -> new: 10.0000
* RTL_LAND_DELAY: curr: 5.0000 -> new: 0.0000
  PWM_MAX: curr: 2000 -> new: 1950
  PWM_MIN: curr: 1000 -> new: 1075
  GPS_UBX_DYNMODEL: curr: 7 -> new: 6
* SYS_AUTOCONFIG: curr: 1 -> new: 0
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
Gazebo multi-robot simulator, version 11.3.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Err] [ConnectionManager.cc:109] Failed to connect to master in 30 seconds.
[Err] [gazebo_shared.cc:77] Unable to initialize transport.
[Err] [gazebo_client.cc:56] Unable to setup Gazebo

Jaeyoung-Lim, 2020-12-08:
I think your bashrc is the problem. Try doing something like https://dev.px4.io/master/en/simulation/ros_interface.html

cuizhi, 2020-12-11:
> I think your bashrc is the problem. Try doing something like https://dev.px4.io/master/en/simulation/ros_interface.html

ths for reply. i tried but nothing works. Finally, i decided try ubuntu 18.04 + gazebo 9.

closed 2020-12-15 by Jaeyoung-Lim