PX4/PX4-SITL_gazebo-classic #666: gstreamer is linked to all plugins

Jaeyoung-Lim, 2020-12-09:
Seems like gstreamer is linked unnecessarily to all the plugins
```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
GSTREAMER_LIBRARIES (ADVANCED)
    linked by target "gazebo_uuv_plugin" in directory /home/jaeyoung/src/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_lidar_plugin" in directory /home/jaeyoung/src/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_groundtruth_plugin" in directory /home/jaeyoung/src/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_sonar_plugin" in directory /home/jaeyoung/src/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_geotagged_images_plugin" in directory /home/jaeyoung/src/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_controller_interface" in directory /home/jaeyoung/src/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_opticalflow_mockup_plugin" in directory /home/jaeyoung/src/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "mav_msgs" in directory /home/jaeyoung/src/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "nav_msgs" in directory /home/jaeyoung/src/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "std_msgs" in directory /home/jaeyoung/src/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_opticalflow_plugin" in directory /home/jaeyoung/src/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_airspeed_plugin" in directory /home/jaeyoung/src/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_airship_dynamics_plugin" in directory /home/jaeyoung/src/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "sensor_msgs" in directory /home/jaeyoung/src/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_vision_plugin" in directory /home/jaeyoung/src/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_gimbal_controller_plugin" in directory /home/jaeyoung/src/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_imu_plugin" in directory /home/jaeyoung/src/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "physics_msgs" in directory /home/jaeyoung/src/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_wind_plugin" in directory /home/jaeyoung/src/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_mavlink_interface" in directory /home/jaeyoung/src/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_motor_model" in directory /home/jaeyoung/src/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_catapult_plugin" in directory /home/jaeyoung/src/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_multirotor_base_plugin" in directory /home/jaeyoung/src/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_magnetometer_plugin" in directory /home/jaeyoung/src/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_usv_dynamics_plugin" in directory /home/jaeyoung/src/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_barometer_plugin" in directory /home/jaeyoung/src/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_parachute_plugin" in directory /home/jaeyoung/src/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_gst_camera_plugin" in directory /home/jaeyoung/src/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_irlock_plugin" in directory /home/jaeyoung/src/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_gps_plugin" in directory /home/jaeyoung/src/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "gazebo_video_stream_widget" in directory /home/jaeyoung/src/PX4-Autopilot/Tools/sitl_gazebo
    linked by target "LiftDragPlugin" in directory /home/jaeyoung/src/PX4-Autopilot/Tools/sitl_gazebo

-- Configuring incomplete, errors occurred!

```

dayjaby, 2021-03-08:
Is this still an issue? I could not verify that the plugins are linked against gstreamer:

```
$ ldd build/px4_sitl_default/build_gazebo/libgazebo_mavlink_interface.so  | grep gst
<nothing>
$ ldd build/px4_sitl_default/build_gazebo/libgazebo_gst_camera_plugin.so  | grep gst
libgstreamer-1.0.so.0 => /usr/local/lib/x86_64-linux-gnu/libgstreamer-1.0.so.0 (0x00007ff72be22000)
```

Jaeyoung-Lim, 2021-03-10:
@dayjaby Probably not, but we still can't compile the project without it which might not be related to linking. I will close this issue for now.

closed 2021-03-10 by Jaeyoung-Lim