PX4/PX4-SITL_gazebo-classic #668: Unable to add LIDARs to Multi UAV simulations using MAVROS in SITL

Jaroan, 2020-12-11:
I have been trying to install LIDARS on two Multirotor Models based on [this link](https://github.com/PX4/PX4-SITL_gazebo/blob/0823f5e006011d36858c2dd39881874c1edac10b/models/iris_obs_avoid/iris_obs_avoid.sdf#L36) of Iris Obstacle Avoidance SDF file

The launch file used was [PX4-Autopilot/launch/multi_uav_mavros_sitl_sdf.launch](https://github.com/PX4/PX4-Autopilot/blob/master/launch/multi_uav_mavros_sitl_sdf.launch)

However, the lidar only appeared on one of the spawned UAVs as shown in the image below.
[image]
[image]
The command rostopic list gave the following details of the topics for the lidars.

    /typhoon_h480_0/laser/scan
    /typhoon_h480_1/laser/scan

While rostopic echo /typhoon_h480_1/laser/scan provided the following readings of the obstacle:

    header:
    seq: 56
    stamp:
    secs: 583
    nsecs: 908000000
    frame_id: “typhoon_h480_1/rplidar_link”
    angle_min: -3.1400001049
    angle_max: 3.1400001049
    angle_increment: 0.0174930356443
    time_increment: 0.0
    scan_time: 0.0
    range_min: 0.20000000298
    range_max: 6.0
    ranges: [inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, 2.846505880355835, 2.722485303878784, 2.663222074508667, 2.617100715637207, 2.5575616359710693, 2.5479671955108643, 2.5439600944519043, 2.5301525592803955, 2.5229380130767822, 2.525747776031494, 2.5125815868377686, 2.513965129852295, 2.5295307636260986, 2.5325605869293213, 2.5703623294830322, 2.602731704711914, 2.6433591842651367, 2.683734655380249, 2.7767276763916016, inf,…] (shortened)
    intensities: [0.0, …0.0] (shortened)

The command rostopic echo /typhoon_h480_0/laser/scan provided no change in values

```
 header: 
  seq: 1280
  stamp: 
    secs: 552
     nsecs: 108000000
   frame_id: "typhoon_h480_0/rplidar_link"
 angle_min: -3.1400001049
 angle_max: 3.1400001049
 angle_increment: 0.0174930356443
 time_increment: 0.0
 scan_time: 0.0
 range_min: 0.20000000298
 range_max: 6.0
 ranges: [inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf, inf]
 intensities: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
```

What could be the problem that made UAV1 have a laser reading but not UAV0?
Any help will be appreciated!
Thank you!

Jaeyoung-Lim, 2020-12-11:
@Jaroan You should check what is happening in https://github.com/PX4/PX4-Autopilot/blob/fd18ba649678a8aa5f1b871b2f6b3d770c1d87d3/launch/single_vehicle_spawn_sdf.launch#L23

Jaroan, 2020-12-11:
@Jaeyoung-Lim I customized the model name in the SDF file as 
` <model name='typhoon_h480_0'>`
And added the same to the lidar model inclusion in my sdf file
```
    <include>
      <uri>model://rplidar</uri>
      <pose>0.00 0 -0.05 0 0 0</pose>
      <name>lidar0</name>
    </include>
    <joint name="lidar_joint0" type="fixed">
      <child>lidar0::link</child>
      <parent>typhoon_h480_0::base_link</parent>
    </joint>
```
This enabled both UAVs to have a functional lidar! I am guessing the appropriate namespace to be provided for all models helped solve this.