PX4/PX4-SITL_gazebo-classic #677: Unable to build release/1.10 on Gazebo11 Ubuntu 20.04

iftahnaf, 2020-12-20:
I'm using Ubuntu 20.04 with release/1.10 and gazebo 11.

When trying to run SITL with default models (iris and empty world), I'm getting the following error:

> INFO] [1608484890.690956, 0.176000]: Calling service /gazebo/spawn_sdf_model
> [INFO] [1608484891.031511, 0.372000]: Spawn status: SpawnModel: Successfully spawned entity
> gzserver: /usr/include/boost/smart_ptr/shared_ptr.hpp:734: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = gazebo::event::Connection; typename boost::detail::sp_member_access<T>::type = gazebo::event::Connection*]: Assertion `px != 0' failed.
> ERROR [mavlink] [timesync] Time jump detected. Resetting time synchroniser.
> Aborted (core dumped)
> [gazebo-3] process has died [pid 19985, exit code 134, cmd /opt/ros/noetic/lib/gazebo_ros/gzserver -e ode /home/iftach/PX4-Autopilot/Tools/sitl_gazebo/worlds/empty.world __name:=gazebo __log:=/home/iftach/.ros/log/cb8e546c-42e7-11eb-899a-998fb2c7f009/gazebo-3.log].
> log file: /home/iftach/.ros/log/cb8e546c-42e7-11eb-899a-998fb2c7f009/gazebo-3*.log

I tried different types of models and worlds but I'm getting the same error.
release/1.10 is missing the iris.sdf file so I took it from the master branch.

When I ran release/1.11 the error was gone and the SITL works without any problem.

Jaeyoung-Lim, 2020-12-20:
> I tried different types of models and worlds but I'm getting the same error.
> release/1.10 is missing the iris.sdf file so I took it from the master branch.

I think this is the problem. You need to fix what the problem was that prevents the `iris.sdf` from being created. If you take `iris.sdf` from master it won't be compatible with v1.10

iftahnaf, 2020-12-21:
Hi @Jaeyoung-Lim, thanks for the answer.
I deleted the `iris.sdf` file, and then I ran:
`make clean`
`make px4_sitl_default`
It generated another `iris.sdf` file, but I'm still getting the same error.
 I'm following the tutorial page when I'm running the simulation:

> cd <Firmware_clone>
> DONT_RUN=1 make px4_sitl_default gazebo
> source ~/catkin_ws/devel/setup.bash    # (optional)
> source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default
> export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)
> export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/sitl_gazebo
> roslaunch px4 posix_sitl.launch

Any other thoughts?

Jaeyoung-Lim, 2020-12-21:
Why are you running ` roslaunch px4 posix_sitl.launch` when trying to run `make px4_sitl_default` ?

Also you need to run gazebo e.g.
```
make px4_sitl gazebo
```

iftahnaf, 2020-12-21:
Ok so I clean everything again with `make clean`, then build with `make px4_sitl gazebo` , and ran:
```
cd ~/PX4-Autopilot
source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/sitl_gazebo
roslaunch px4 mavros_posix_sitl.launch
```
Still, I'm getting the same error. What else should I try?

Jaeyoung-Lim, 2020-12-21:
As I said above, why are you running roslaunch?

iftahnaf, 2020-12-21:
@Jaeyoung-Lim  Thanks again for the help.

I'm trying to run ROS-gazebo SITL, and I'm following the instructions from here [](https://dev.px4.io/master/en/simulation/ros_interface.html)

When I try to run `roslaunch px4 mavros_posix_sitl.launch ` (I'm exporting the `ROS_PACKAGE_PATH` and source the nessecery files as written in the instructions before `roslaunch`)get the error:

> gzserver: /usr/include/boost/smart_ptr/shared_ptr.hpp:734: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = gazebo::event::Connection; typename boost::detail::sp_member_access<T>::type = gazebo::event::Connection*]: Assertion `px != 0' failed.
> Aborted (core dumped)
> [gazebo-3] process has died [pid 5010, exit code 134, cmd /opt/ros/noetic/lib/gazebo_ros/gzserver -e ode /home/iftach/PX4-Autopilot/Tools/sitl_gazebo/worlds/empty.world __name:=gazebo __log:=/home/iftach/.ros/log/9fce7f24-439f-11eb-a5fe-213a4b37e415/gazebo-3.log].
> log file: /home/iftach/.ros/log/9fce7f24-439f-11eb-a5fe-213a4b37e415/gazebo-3*.log

And when I try to run only the `make px4_sitl gazebo` I'm getting:

> [Err] [Model.cc:1137] Exception occured in the Load function of plugin with name[mavlink_interface] and filename[libgazebo_mavlink_interface.so]. This plugin will not run.
> gzserver: /usr/include/boost/smart_ptr/shared_ptr.hpp:734: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = gazebo::event::Connection; typename boost::detail::sp_member_access<T>::type = gazebo::event::Connection*]: Assertion `px != 0' failed.
> /home/iftach/PX4-Autopilot/Tools/sitl_run.sh: line 93:  5806 Aborted                 (core dumped) gzserver --verbose "${src_path}/Tools/sitl_gazebo/worlds/${model}.world"
> SITL COMMAND: "/home/iftach/PX4-Autopilot/build/px4_sitl_default/bin/px4" "/home/iftach/PX4-Autopilot"/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t "/home/iftach/PX4-Autopilot"/test_data
> INFO  [px4] Creating symlink /home/iftach/PX4-Autopilot/ROMFS/px4fmu_common -> /home/iftach/PX4-Autopilot/build/px4_sitl_default/tmp/rootfs/etc
> 0 WARNING: setRealtimeSched failed (not run as root?)

closed 2020-12-28 by LorenzMeier

RamanaBotta, 2022-02-21:
Any answers?