PX4/PX4-SITL_gazebo-classic #679: Mac OS Big Sur SITL Gazebo build error

kadirozgunn, 2020-12-21:
Hi everyone,

I get error when I build gazebo on Mac OS Big Sur. I successfully did Jmavsim SITL however when I build SITL_Gazebo I get error:

```
[765/769] Performing build step for 'sitl_gazebo'
ninja: error: '/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/usr/lib/libm.tbd', needed by 'libmav_msgs.dylib', missing and no known rule to make it
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build 
cd /Users/kadirozgun/PX4-Autopilot/build/px4_sitl_default/build_gazebo && /usr/local/Cellar/cmake/3.19.2/bin/cmake --build /Users/kadirozgun/PX4-Autopilot/build/px4_sitl_default/build_gazebo -- -j1
ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1
```

Anyone get this error and recommendation?

Jaeyoung-Lim, 2020-12-21:
@kadirozgunn As you can see in the macOS build results, currently brew installation is failing on macOS BigSur

canberkdurmus, 2020-12-27:
After an accidental update to the Big Sur, I downgraded by system manually to Catalina 10.15.7 back but the error is the same.

LorenzMeier, 2020-12-28:
I'm currently looking into this (running on Big Sur).

LorenzMeier, 2020-12-28:
I'm running successfully on Big sur after fixing a brew dependency: https://github.com/PX4/PX4-Autopilot/commit/3d809134f5eec566e8159987debcb4190c973d06

I'm closing this.

closed 2020-12-28 by LorenzMeier

mrpollo, 2021-02-08:
I ran into this issue this morning on a fresh macOS Big Sur install, I was missing the Command Line Tools for Xcode, worked great afterward.

mrischmu, 2021-02-15:
hi @mrpollo, could you briefly explain what you did please? 
Thanks!

mrpollo, 2021-02-15:
On a fresh macOs Big Sur install, you don't have any of the developer tools you need to build the project, Apple doesn't bundle them on the OS anymore, you need to download them manually, although it requires an Apple ID (free) you can download the latest from here https://developer.apple.com/downloads/index.action

mrischmu, 2021-02-15:
Hey, thanks for the fast answer! I did downloaded the command line Tools for Xcode and installed it (as well as reinstalled Xcode), but I still have the error mentioned above:
_ninja: error: '/usr/lib/libm.dylib', needed by 'libmav_msgs.dylib', missing and no known rule to make it
[1/5] Generating ../../logs
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build
cd /Users/XXX/Workspace/PX4-Autopilot/build/px4_sitl_default/build_gazebo && /usr/local/Cellar/cmake/3.18.2/bin/cmake --build /Users/XXX/Workspace/PX4-Autopilot/build/px4_sitl_default/build_gazebo -- -j1_
Any tips?
Thanks!

mrpollo, 2021-02-15:
if you installed the command line tools after you had already tried building the project you might need to clean up your environment, try `make distclean`, and then try building again