PX4/PX4-SITL_gazebo-classic #683: Camera Manager Issues in multiple SITL Instances

mzahana, 2020-12-29:
Hi,
I would like to describe some of the issues related to the camera manager which simulates mavlink camera protocol in SITL. This is implemented in [gazebo_geotagged_images_plugin.cpp](https://github.com/PX4/PX4-SITL_gazebo/blob/master/src/gazebo_geotagged_images_plugin.cpp).

The issue appeared when I tried to get video streams from multiple simulated `typhoon_h480`models in Gazebo. The streaming works correctly for a single drone simulation. However, in multi-typhoon simulation, when I switch between vehicles in QGC, I don't get the corresponding video stream in QGC video display.

The problem seems to be in the camera manager implemented in [gazebo_geotagged_images_plugin.cpp](https://github.com/PX4/PX4-SITL_gazebo/blob/master/src/gazebo_geotagged_images_plugin.cpp). In  particular,
* The mavlink ID (vehicle ID) seems to be hard-coded in the sent mavlink messages, for example [here](https://github.com/PX4/PX4-SITL_gazebo/blob/master/src/gazebo_geotagged_images_plugin.cpp#L240), [here](https://github.com/PX4/PX4-SITL_gazebo/blob/master/src/gazebo_geotagged_images_plugin.cpp#L362), [here](https://github.com/PX4/PX4-SITL_gazebo/blob/master/src/gazebo_geotagged_images_plugin.cpp#L549), [here](https://github.com/PX4/PX4-SITL_gazebo/blob/master/src/gazebo_geotagged_images_plugin.cpp#L585), [here](https://github.com/PX4/PX4-SITL_gazebo/blob/master/src/gazebo_geotagged_images_plugin.cpp#L612), [here](https://github.com/PX4/PX4-SITL_gazebo/blob/master/src/gazebo_geotagged_images_plugin.cpp#L648), [here](https://github.com/PX4/PX4-SITL_gazebo/blob/master/src/gazebo_geotagged_images_plugin.cpp#L717), [here](https://github.com/PX4/PX4-SITL_gazebo/blob/master/src/gazebo_geotagged_images_plugin.cpp#L743). This ID should be different for different vehicles. Therefore, it should be passed to as a parameter to the the plugin
* Another issue is related to the camera streaming `URI` which is currently hard-coded in [here](https://github.com/PX4/PX4-SITL_gazebo/blob/master/src/gazebo_geotagged_images_plugin.cpp#L644). This should be different for different cameras and should be parametrized, and passed to the plugin when called from an `.xml` file
* Another issue is the port number at which the plugin receives data from QGC. Currently, it is hard-coded [here](https://github.com/PX4/PX4-SITL_gazebo/blob/master/src/gazebo_geotagged_images_plugin.cpp#L459) and should be different for different instances of the plugin. Therefore, it should be parametrized as well
* Another issue is in the case of having multiple simulated cameras on the same drone. In this case each simulated camera requires a separate plugin instance which should have unique ports and URI as mentioned above, and also unique `MAV_COMP_ID_CAMERA` e.g. [here](https://github.com/PX4/PX4-SITL_gazebo/blob/master/src/gazebo_geotagged_images_plugin.cpp#L550).

I hope this is useful, and hopefully we have a PR that fixes these issues soon.

Jaeyoung-Lim, 2021-01-01:
As discussed with @mzahana offline, @mzahana will be looking into a fix for this issue

mzahana, 2021-01-03:
@Jaeyoung-Lim please check PR https://github.com/PX4/PX4-SITL_gazebo/pull/687

Jaeyoung-Lim, 2021-01-10:
Fixed in https://github.com/PX4/PX4-SITL_gazebo/pull/687 thanks to @mzahana !

closed 2021-01-10 by Jaeyoung-Lim