PX4/PX4-SITL_gazebo-classic #686: Controlling a suspended load

eldewany, 2021-01-02:
Hello, i'm developing a simulation that including a mass hung by a rod from the iris quadrotor, i added the load to the iris sdf file and it's working fine. The issue is i want to induce sudden displacements to the load in terms of load swing angles and impulses. Googling led me to ros controllers but they require URDF which is not present for the current SDF format used by sitl. What should i do ?

Jaeyoung-Lim, 2021-01-03:
This seems like a general gazebo question than related to this repository. For general ROS questions, you can also try posting your question on [https://answers.gazebosim.org/questions/](https://answers.gazebosim.org/questions/)

If you are interested in contributing the feature to this repository, this repo does not depend on ROS therefore it would be better with having no ROS dependencies. You can try writing a gazebo plugin that applies the impulses you want

eldewany, 2021-01-03:
Okay, thanks for your reply. I will post the question there and i will look into the gazebo plugin thing. Thanks again

closed 2021-01-03 by eldewany