PX4/PX4-SITL_gazebo-classic #693: Use sdf files in rviz

AliaChe, 2021-01-18:
Hello, 

I'm trying to use rviz to visualize a simulated iris drone in gazebo. Since the urdf and xacro files are deprecated, is there any method to visualize the drone, TF and the lazerscans in rviz using the sdf files instead of the urdf files? 

Thank you.

Jaeyoung-Lim, 2021-01-19:
@AliaChe You can define markers that can be visualized in Rviz: http://wiki.ros.org/rviz/DisplayTypes/Marker

AliaChe, 2021-01-19:
@Jaeyoung-Lim Thank you for your reply,  do you have any concrete example where markers are used to visualize the drone, the lidar and the lazerscans?

Jaeyoung-Lim, 2021-01-19:
@AliaChe You can check how this is being done in the [PX4/Avoidance](https://github.com/PX4/PX4-Avoidance) project.

closed 2021-01-19 by Jaeyoung-Lim