PX4/PX4-SITL_gazebo-classic #694: BlueROV2 Thruster chaotic Movement

Zarbokk, 2021-01-19:
Does someone know the reason why the propeller/thruster are moving around in the simulation if the motors are actuated?
This happens in Gazebo 11 started with:
`make px4_sitl_default gazebo_uuv_bluerov2_heavy`
If the motors are not actuated they are at the correct position. 

[image]

Jaeyoung-Lim, 2021-01-19:
@lennartalff Would it be possible for you to have a look?

lennartalff, 2021-01-25:
Should be solved in PX4/PX4-Autopilot#16637. If the Problem still occurs, please let me know.

Jaeyoung-Lim, 2021-01-26:
@lennartalff I have just tried and the problem is not solved.

I don't think the problems are related, since the props actually detach when running with gazebo11

Zarbokk, 2021-01-26:
Currently, I was not able to solve that problem.
The direction of the propeller is exactly the direction of the force that is acting on the body/propeller.
It behaves like a spring, Big force Big movement, low force low movement.
In the Gazebo forum, I was not able to find something useful only small related problems, without a real solution.

Maybe it's related to the Gazebo version? I am currently using Gazebo 11.

lennartalff, 2021-01-29:
@Jaeyoung-Lim I think it is related to the solver settings. Increasing the `<iters>` value in `uuv_bluerov2_heavy.world` to at least 20 reduces the errenous deflection of the rotors. Is there a reason for the chosen value of 10? Otherwise I would propose to increase it to 20 or maybe even use the default value of 50?

Zarbokk, 2021-02-02:
solved by hub.com/PX4/PX4-SITL_gazebo/pull/701

closed 2021-02-02 by Zarbokk