PX4/PX4-SITL_gazebo-classic #697: Creation of new models

ricardorodrigues09, 2021-01-22:
I know this isn't really an issue but I'm trying to build my own model and launch a simulation using mavros by running roslaunch px4 mavros_posix_sitl.launch. The difference is that instead of using iris.sdf I would be using a model created by me. When I do that I'm getting this output:

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
process[sitl-1]: started with pid [27566]
Error: Unknown model ''
ERROR [px4] Startup script returned with return value: 256
process[gazebo-2]: started with pid [27580]
process[gazebo_gui-3]: started with pid [27588]
process[vehicle_spawn_ubuntu_27550_3046093602485569898-4]: started with pid [27590]
================================================================================REQUIRED process [sitl-1] has died!
process has died [pid 27566, exit code 255, cmd /home/ricardorodrigues/PX4_Firmware/build/px4_sitl_default/bin/px4 /home/ricardorodrigues/PX4_Firmware/ROMFS/px4fmu_common -s etc/init.d-posix/rcS __name:=sitl __log:=/home/ricardorodrigues/.ros/log/ede81232-5cf0-11eb-8349-000c29b10c35/sitl-1.log].
log file: /home/ricardorodrigues/.ros/log/ede81232-5cf0-11eb-8349-000c29b10c35/sitl-1*.log
Initiating shutdown!

I have also tried changing iris.sdf to irismodified.sdf in order to know if the problem was related to the content of my sdf file or to its name. I guess the problem is that the simulation is not recognizing my model exists. Would you be kind to help me and tell me what should be the steps I have to perform? Do I need to change smth on the CMakeLists.txt file?

ricardorodrigues09, 2021-01-22:
I forgot to add that this model is a combination of differential drive robot which I created and an Iris model.

Jaeyoung-Lim, 2021-01-23:
@ricardorodrigues09 We already have a differential rover in https://github.com/PX4/PX4-SITL_gazebo/tree/master/models/r1_rover

Probably this helps?

ricardorodrigues09, 2021-01-25:
@Jaeyoung-Lim the thing is that I am creating a vehicle which is basically a drone mounted on top of a rover, so both connected by a fixed joint as if it was a single vehicle

Jaeyoung-Lim, 2021-01-28:
@ricardorodrigues09 You need to write a sdf file with your robot configuration then

Jaeyoung-Lim, 2021-01-30:
Closing, since this is a usage question and not an issue of this repo. 

If you have more questions, please post it on discuss.px4.io

closed 2021-01-30 by Jaeyoung-Lim