PX4/PX4-SITL_gazebo-classic #699: Support for Gazebo 11

McKayCyborg, 2021-01-28:
Hi there, 

Currently, I am seeing issues similar to: _https://github.com/PX4/PX4-SITL_gazebo/issues/419_ whereby the mavlink interface plugin fails to load if run it using Gazebo 11. 

Are there any plans to update the simulation tools to run on Gazebo 11 in the near future? My issue is that I would like to use the libgazebo_mavlink_interface with ros_control and need to use Noetic (because I am using Ubuntu 20.04), but ROS Noetic only supports Gazebo 11 as far as I can tell.

Any help would be awesome!

Thanks

Jaeyoung-Lim, 2021-01-28:
@McKayCyborg I run gazebo11 with ROS Noetic without any issues, and it is the standard environment where the PX4 SITL tests run in CI

Therefore it is suppoeted

McKayCyborg, 2021-01-28:
Thanks @Jaeyoung-Lim,

Given that I am running into the same issue as posted in: https://github.com/PX4/PX4-SITL_gazebo/issues/419, what ended up being the solution?

Jaeyoung-Lim, 2021-01-28:
@McKayCyborg Could you explain what your issue is? The issue you are referring to is gazebo11 with Ubuntu 18.04 and the errors are usually different

McKayCyborg, 2021-01-28:
@Jaeyoung-Lim to begin with, I'm not using the most up-to-date version of PX4 and I am using the mavlink interface plugins and the various sensor plugins in a custom ROS package so if you feel like answering this question is too much of a stretch, I understand if you just want to just close out the issue. Although, I opened the thread because I feel like this issue is has everything to do with the mavlink plugin itself.

Some details:
Ubuntu 20.04
PX4 version: v1.10.0-beta4 (cloned Oct 28 2019)
Gazebo11

What I've done:
I brought across the mavlink interface plugin and the sensor plugins (gps, baro, compass etc) from that PX4 clone into a custom package and am running the gazebo simulation independently of the PX4 SITL daemon. The Gazebo model includes the plugins in exactly the same way that PX4 does, and I've had this running successfully on Ubuntu 18.04 with Gazebo 9.

The error:
After booting my launch file, I essentially receive this error:

`[Err] [Model.cc:1137] Exception occured in the Load function of plugin with name[mavlink_interface] and filename[libgazebo_mavlink_interface.so]. This plugin will not run.
gzserver: /usr/include/boost/smart_ptr/shared_ptr.hpp:734: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = gazebo::event::Connection; typename boost::detail::sp_member_access<T>::type = gazebo::event::Connection*]: Assertion `px != 0' failed.
`

My Question:
Apart from the issue itself, I'm wondering at what point support for Gazebo11 was added? Because I feel like it's possible that I simply just have outdated plugins and need to update them to include support for Gazebo11

Jaeyoung-Lim, 2021-01-28:
@McKayCyborg It was added by this PR: https://github.com/PX4/PX4-SITL_gazebo/pull/497

The error you are showing simply means gazebo was unable to find one of the resources, so it is hard to tell what is wrong. 

If you are using custom code I would recommend you to try running PX4 master with Gazebo11 first to check if everything else is configured properly on your system

McKayCyborg, 2021-01-28:
Thankyou @Jaeyoung-Lim, I'll do that and make sure my system is configured properly.

McKayCyborg, 2021-01-29:
@Jaeyoung-Lim after cloning the master branch, it appears that the Gazebo rover sim loads without errors (except for being unable to load a bunch of visuals), although it doesn't connect to QGroundControl so I can't verify if the vehicle will actually run a mission. The Gazebo environment loads but the daemon just shows this output:

```
______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
Info: found model autostart file as SYS_AUTOSTART=10016
INFO  [param] selected parameter default file eeprom/parameters_10016
[param] parameter file not found, creating eeprom/parameters_10016
  SYS_AUTOCONFIG: curr: 0 -> new: 1
  SYS_AUTOSTART: curr: 0 -> new: 10016
  BAT_N_CELLS: curr: 0 -> new: 4
  CAL_ACC0_ID: curr: 0 -> new: 1310988
  CAL_ACC1_ID: curr: 0 -> new: 1310996
  CAL_ACC2_ID: curr: 0 -> new: 1311004
  CAL_GYRO0_ID: curr: 0 -> new: 1310988
  CAL_GYRO1_ID: curr: 0 -> new: 1310996
  CAL_GYRO2_ID: curr: 0 -> new: 1311004
  CAL_MAG0_ID: curr: 0 -> new: 197388
  CAL_MAG1_ID: curr: 0 -> new: 197644
  CBRK_SUPPLY_CHK: curr: 0 -> new: 894281
  COM_RC_IN_MODE: curr: 0 -> new: 1
  EKF2_REQ_GPS_H: curr: 10.0000 -> new: 0.5000
  EKF2_MULTI_IMU: curr: 0 -> new: 3
  SENS_IMU_MODE: curr: 1 -> new: 0
  EKF2_MULTI_MAG: curr: 0 -> new: 2
  SENS_MAG_MODE: curr: 1 -> new: 0
  SDLOG_MODE: curr: 0 -> new: 1
  SDLOG_PROFILE: curr: 1 -> new: 131
  SDLOG_DIRS_MAX: curr: 0 -> new: 7
  SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
  SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
  TRIG_INTERFACE: curr: 4 -> new: 3
  COM_CPU_MAX: curr: 90.0000 -> new: -1.0000
  IMU_INTEG_RATE: curr: 200 -> new: 250
  NAV_ACC_RAD: curr: 10.0000 -> new: 2.0000
  RTL_RETURN_ALT: curr: 60.0000 -> new: 30.0000
  RTL_DESCEND_ALT: curr: 30.0000 -> new: 10.0000
  PWM_MAX: curr: 2000 -> new: 1950
  PWM_MIN: curr: 1000 -> new: 1075
  GPS_UBX_DYNMODEL: curr: 7 -> new: 6
* SYS_AUTOCONFIG: curr: 1 -> new: 0
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
Gazebo multi-robot simulator, version 11.3.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 10.0.0.21
[Err] [ModelDatabase.cc:386] No <model> element in model.config for model[http://models.gazebosim.org/lamp_post]
```

My QGC version is v4.1.1

Jaeyoung-Lim, 2021-01-29:
@McKayCyborg I am not sure what you are saying is accurate
```
Info: found model autostart file as SYS_AUTOSTART=10016
```
The log you are showing is trying to load the iris model. Probably something else is wrong?

McKayCyborg, 2021-01-29:
@Jaeyoung-Lim Sorry for the cconfusion - I ran it again with the iris model to see if it was something rover-specific and have sent that output to you instead of the rover output

Jaeyoung-Lim, 2021-01-29:
@McKayCyborg And you don't have connection to QGC on both? Now I am confused - qgc connection should not be related to gazebo versions

Jaeyoung-Lim, 2021-02-01:
@McKayCyborg Any updates?

McKayCyborg, 2021-02-01:
@Jaeyoung-Lim after a deeper dive yesterday it looked like the module I was using was failing inside the gazebo_mavlink_interface plugin Load method during this call:

```
presetManager->GetCurrentProfileParam("max_step_size", param);
const double max_step_size = boost::any_cast<double>(param);
```

In the updated plugins, there is a conditional define that reworks this to:

```
template <class T>
T our_any_cast(const boost::any &val) {
#if GAZEBO_MAJOR_VERSION >= 11
  return gazebo::physics::PhysicsEngine::any_cast<T>(val);
#else
  return boost::any_cast<T>(val);
#endif
}
```

Clearly, my system was not able to locate the boost::any_cast<>() resource. My simulation appears to be working alright now, although the vehicle position seems to drift alot. 

Thankyou for your guidance

closed 2021-02-01 by McKayCyborg