PX4/PX4-SITL_gazebo-classic #705: Simulation starting error for gazebo_boat

bigbellmercy, 2021-02-05:
Hi~ Thanks for the `gazebo_boat`!

I'm trying to use `gazebo_boat`, for the first time, for my research project in Korea.

While I am trying to use `gazebo_boat`, I cannot run it by following error:
`[Err] [Connection.cc:546] Connection[0] Closed during Read`

I tried with several PX4 versions. v1.11.0 and maser(as of Feb 5): the same error. v1.10.2 and v1.9.2 does not recognize `gazebo_boat`.

It'll be good if some solution is given.

The all the messages are here:
```
`$:~/PX4/Firmware$ make px4_sitl gazebo_boat
[0/5] Performing build step for 'sitl_gazebo'
ninja: no work to do.
[4/5] cd /home/jong/PX4/Firmware/build...ng/PX4/Firmware/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/jong/PX4/Firmware/build/px4_sitl_default/bin/px4
debugger: none
program: gazebo
model: boat
world: none
src_path: /home/jong/PX4/Firmware
build_path: /home/jong/PX4/Firmware/build/px4_sitl_default
GAZEBO_PLUGIN_PATH :/home/jong/PX4/Firmware/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH :/home/jong/PX4/Firmware/Tools/sitl_gazebo/models
LD_LIBRARY_PATH /home/jong/catkin_ws2/devel/lib:/home/jong/catkin_ws/devel/lib:/opt/ros/melodic/lib:/home/jong/PX4/Firmware/build/px4_sitl_default/build_gazebo
empty world, default world boat.world for model found
SITL COMMAND: "/home/jong/PX4/Firmware/build/px4_sitl_default/bin/px4" "/home/jong/PX4/Firmware"/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t "/home/jong/PX4/Firmware"/test_data
INFO  [px4] Creating symlink /home/jong/PX4/Firmware/ROMFS/px4fmu_common -> /home/jong/PX4/Firmware/build/px4_sitl_default/tmp/rootfs/etc

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [param] selected parameter default file eeprom/parameters_1070
[param] parameter file not found, creating eeprom/parameters_1070
  SYS_AUTOCONFIG: curr: 0 -> new: 1
  SYS_AUTOSTART: curr: 0 -> new: 1070
  BAT_N_CELLS: curr: 0 -> new: 4
  CAL_ACC0_ID: curr: 0 -> new: 1311244
  CAL_ACC1_ID: curr: 0 -> new: 1311500
  CAL_GYRO0_ID: curr: 0 -> new: 1311244
  CAL_GYRO1_ID: curr: 0 -> new: 1311500
  CAL_MAG0_ID: curr: 0 -> new: 197388
  CAL_MAG1_ID: curr: 0 -> new: 197644
  CBRK_SUPPLY_CHK: curr: 0 -> new: 894281
  COM_RC_IN_MODE: curr: 0 -> new: 1
  EKF2_ANGERR_INIT: curr: 0.1000 -> new: 0.0100
  EKF2_GBIAS_INIT: curr: 0.1000 -> new: 0.0100
  COM_ARM_EKF_AB: curr: 0.0022 -> new: 0.0050
  EKF2_REQ_GPS_H: curr: 10.0000 -> new: 0.5000
  MC_PITCH_P: curr: 6.5000 -> new: 6.0000
  MC_PITCHRATE_P: curr: 0.1500 -> new: 0.2000
  MC_ROLL_P: curr: 6.5000 -> new: 6.0000
  MC_ROLLRATE_P: curr: 0.1500 -> new: 0.2000
  MPC_Z_VEL_P_ACC: curr: 4.0000 -> new: 12.0000
  MPC_Z_VEL_I_ACC: curr: 2.0000 -> new: 3.0000
  MPC_XY_P: curr: 0.9500 -> new: 0.8000
  MPC_XY_VEL_P_ACC: curr: 1.8000 -> new: 4.0000
  MPC_XY_VEL_D_ACC: curr: 0.2000 -> new: 0.3200
  RTL_RETURN_ALT: curr: 60.0000 -> new: 30.0000
  SDLOG_MODE: curr: 0 -> new: 1
  SDLOG_PROFILE: curr: 1 -> new: 131
  SDLOG_DIRS_MAX: curr: 0 -> new: 7
  SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
  SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
  TRIG_INTERFACE: curr: 4 -> new: 3
  COM_CPU_MAX: curr: 90.0000 -> new: -1.0000
  IMU_INTEG_RATE: curr: 200 -> new: 250
  MIS_LTRMIN_ALT: curr: -1.0000 -> new: 0.0100
  MIS_TAKEOFF_ALT: curr: 2.5000 -> new: 0.0100
  NAV_ACC_RAD: curr: 10.0000 -> new: 2.0000
  NAV_FW_ALT_RAD: curr: 10.0000 -> new: 1000.0000
Gazebo multi-robot simulator, version 9.16.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
  GND_SPEED_IMAX: curr: 1.0000 -> new: 0.1250
  GND_SPEED_P: curr: 2.0000 -> new: 0.2500
  GND_SPEED_TRIM: curr: 3.0000 -> new: 4.0000
  GND_THR_CRUISE: curr: 0.1000 -> new: 0.3000
  GND_THR_MAX: curr: 0.3000 -> new: 0.5000
* NAV_ACC_RAD: curr: 2.0000 -> new: 0.5000
  NAV_LOITER_RAD: curr: 50.0000 -> new: 2.0000
  CBRK_AIRSPD_CHK: curr: 0 -> new: 162128
  GND_MAX_ANG: curr: 0.7854 -> new: 0.6000
* SYS_AUTOCONFIG: curr: 1 -> new: 0
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
[Err] [Connection.cc:546] Connection[0] Closed during Read
[Err] [ConnectionManager.cc:135] Unable to read from master
[Err] [gazebo_shared.cc:77] Unable to initialize transport.
[Err] [gazebo_client.cc:56] Unable to setup Gazebo
^CERROR [px4] Startup script returned with return value: 2
/home/jong/PX4/Firmware/Tools/sitl_run.sh: line 178:  2122 Segmentation fault      (core dumped) gzserver "${src_path}/Tools/sitl_gazebo/worlds/${model}.world"  (wd: ~/PX4/Firmware/build/px4_sitl_default/tmp)
/home/jong/PX4/Firmware/Tools/sitl_run.sh: line 187: kill: (2122) - No such process
/home/jong/PX4/Firmware/Tools/sitl_run.sh: line 189: kill: (2457) - No such process
ninja: build stopped: interrupted by user.
Makefile:224: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Interrupt
```
`

LorenzMeier, 2021-02-05:
[image]

Gazebo 9 is not supported, please update to 10 or 11.

LorenzMeier, 2021-02-05:
Here is where Gazebo 9 crashed:
```
/home/jong/PX4/Firmware/Tools/sitl_run.sh: line 178:  2122 Segmentation fault      (core dumped) gzserver "${src_path}/Tools/sitl_gazebo/worlds/${model}.world"  (wd: ~/PX4/Firmware/build/px4_sitl_default/tmp)
```

closed 2021-02-05 by LorenzMeier

Jaeyoung-Lim, 2021-02-05:
@bigbellmercy Actually, I can run boat with gazebo9

It looks like you have somekind of a submodule mismatch with your firmware