PX4/PX4-SITL_gazebo-classic #706: thermal camera model (.sdf file)

addy1997, 2021-02-07:
I am trying to integrate a thermal camera with my IRIS quadcopter and simulate it in the ignition gazebo with PX4. If this feature is not possible, can you guide me on how to implement the file?

Jaeyoung-Lim, 2021-02-07:
@addy1997 There are two big questions in your question.

1. Using ignition gazebo is not implemented in this repository. You will need to port part the implementation of this repo into ignition gazebo. Classical "gazebo" and "ignition gazebo" are not compatible with different architecture and API

2. For thermal cameras, you need to add a "thermal camera" plugin. For reference, you can look at something like: http://wiki.ros.org/hector_gazebo_thermal_camera

addy1997, 2021-02-07:
But porting requires permission from the repo owner right? I just can't take away a chunk of code.

Jaeyoung-Lim, 2021-02-07:
@addy1997 If you comply with the license of this repository, there is nothing preventing you from doing so

addy1997, 2021-02-07:
Indubitably, yes I will comply. @Jaeyoung-Lim thanks for your help. really appreciated !!!

closed 2021-02-07 by addy1997