PX4/PX4-SITL_gazebo-classic #709: Sonar plugin not publishing topic?

weiyuhe, 2021-02-15:
Not sure why but the sonar sensor is not publishing any topics. Is the sonar plugin in _PX4-SITL_gazebo/models/sonar/model.sdf_  still unfinished?

Jaeyoung-Lim, 2021-02-16:
Where are you expecting it to publish?

weiyuhe, 2021-02-16:
Since it has:
```
 const string scopedName = _parent->ParentName();
  string topicName = "~/" + scopedName + "/sonar";
  boost::replace_all(topicName, "::", "/");

  sonar_pub_ = node_handle_->Advertise<sensor_msgs::msgs::Range>(topicName, 10);
```
I expected it to publish a ros topic, but it did not.

weiyuhe, 2021-02-16:
It looks like I was using an old version of the plugin, and now I updated it. However, I am getting some make errors:

```
FAILED: CMakeFiles/gazebo_sonar_plugin.dir/src/gazebo_sonar_plugin.cpp.o 
/usr/bin/c++  -DLIBBULLET_VERSION=2.87 -DLIBBULLET_VERSION_GT_282 -DROSCONSOLE_BACKEND_LOG4CXX -DROS_BUILD_SHARED_LIBS=1 -DROS_PACKAGE_NAME=\"mavlink_sitl_gazebo\" -Dgazebo_sonar_plugin_EXPORTS -isystem /usr/include/gazebo-9 -isystem /usr/include/bullet -isystem /usr/include/simbody -isystem /usr/local/include -isystem /usr/include/sdformat-6.2 -isystem /usr/include/ignition/math4 -isystem /usr/include/OGRE -isystem /usr/include/OGRE/Terrain -isystem /usr/include/OGRE/Paging -isystem /usr/include/ignition/transport4 -isystem /usr/include/ignition/msgs1 -isystem /usr/include/ignition/common1 -isystem /usr/include/ignition/fuel_tools1 -I/home/afrl/px4/Firmware/Tools/sitl_gazebo/include -I. -I/usr/include/eigen3 -I/usr/include/eigen3/eigen3 -I/usr/include/gazebo-9/gazebo/msgs -I/home/afrl/px4/Firmware/mavlink/include -I/usr/include/Paging -isystem /usr/include/opencv -I/home/afrl/px4/Firmware/Tools/sitl_gazebo/external/OpticalFlow/include -I/opt/ros/melodic/include -I/opt/ros/melodic/share/xmlrpcpp/cmake/../../../include/xmlrpcpp -isystem /usr/include/uuid -isystem /usr/include/x86_64-linux-gnu -Wno-deprecated-declarations -fPIC   -I/usr/include/uuid -I/usr/include/x86_64-linux-gnu -std=gnu++1z -MD -MT CMakeFiles/gazebo_sonar_plugin.dir/src/gazebo_sonar_plugin.cpp.o -MF CMakeFiles/gazebo_sonar_plugin.dir/src/gazebo_sonar_plugin.cpp.o.d -o CMakeFiles/gazebo_sonar_plugin.dir/src/gazebo_sonar_plugin.cpp.o -c /home/afrl/px4/Firmware/Tools/sitl_gazebo/src/gazebo_sonar_plugin.cpp
/home/afrl/px4/Firmware/Tools/sitl_gazebo/src/gazebo_sonar_plugin.cpp: In member function ‘virtual void gazebo::SonarPlugin::OnNewScans()’:
/home/afrl/px4/Firmware/Tools/sitl_gazebo/src/gazebo_sonar_plugin.cpp:226:18: error: ‘class sensor_msgs::msgs::Range’ has no member named ‘set_h_fov’
   sonar_message_.set_h_fov(2.0f * atan(parentSensor_->Radius() / parentSensor_->RangeMax()));
                  ^~~~~~~~~
/home/afrl/px4/Firmware/Tools/sitl_gazebo/src/gazebo_sonar_plugin.cpp:227:18: error: ‘class sensor_msgs::msgs::Range’ has no member named ‘set_v_fov’
   sonar_message_.set_v_fov(2.0f * atan(parentSensor_->Radius() / parentSensor_->RangeMax()));
                  ^~~~~~~~~
/home/afrl/px4/Firmware/Tools/sitl_gazebo/src/gazebo_sonar_plugin.cpp:228:18: error: ‘class sensor_msgs::msgs::Range’ has no member named ‘set_allocated_orientation’
   sonar_message_.set_allocated_orientation(new gazebo::msgs::Quaternion(orientation_));
```

Jaeyoung-Lim, 2021-02-16:
@weiyuhe Those are gazebo transports, not ROS topics. You need packages such as `gazebo_ros` in order to republish these topics or you need to add a ros interface/ publisher

weiyuhe, 2021-02-16:
@Jaeyoung-Lim Thanks for the clarification.

And yes, not I am using the sonar plugin from this repo, but now it is giving all these make errors.

Jaeyoung-Lim, 2021-02-16:
@weiyuhe 
```
make distclean
make submodulesupdate
make px4_sitl gazebo
```

weiyuhe, 2021-02-16:
Ok I did what you told me and it push me back to the "older" version. By that I mean:
```
#include "gazebo/physics/physics.hh"
#include "gazebo_sonar_plugin.h"

#include <gazebo/common/common.hh>
#include <gazebo/common/Plugin.hh>
#include <gazebo/gazebo.hh>
#include <gazebo/physics/physics.hh>
#include "gazebo/transport/transport.hh"
#include "gazebo/msgs/msgs.hh"

#include <chrono>
#include <cmath>
#include <iostream>
#include <memory>
#include <stdio.h>
#include <boost/algorithm/string.hpp>

using namespace gazebo;
using namespace std;

// Register this plugin with the simulator
GZ_REGISTER_SENSOR_PLUGIN(SonarPlugin)

/////////////////////////////////////////////////
SonarPlugin::SonarPlugin()
{
}

/////////////////////////////////////////////////
SonarPlugin::~SonarPlugin()
{
  this->parentSensor.reset();
  this->world.reset();
}

/////////////////////////////////////////////////
void SonarPlugin::Load(sensors::SensorPtr _parent, sdf::ElementPtr _sdf)
{
//  Get then name of the parent sensor
  this->parentSensor = std::dynamic_pointer_cast<sensors::SonarSensor>(_parent);

  if (!this->parentSensor)
    gzthrow("SonarPlugin requires a Sonar Sensor as its parent");

  this->world = physics::get_world(this->parentSensor->WorldName());

  this->parentSensor->SetActive(false);
  this->newScansConnection = this->parentSensor->ConnectUpdated(boost::bind(&SonarPlugin::OnNewScans, this));
  this->parentSensor->SetActive(true);

  if (_sdf->HasElement("robotNamespace"))
    namespace_ = _sdf->GetElement("robotNamespace")->Get<std::string>();
  else
    gzwarn << "[gazebo_sonar_plugin] Please specify a robotNamespace.\n";

  node_handle_ = transport::NodePtr(new transport::Node());
  node_handle_->Init(namespace_);

  const string scopedName = _parent->ParentName();
  string topicName = "~/" + scopedName + "/sonar";
  boost::replace_all(topicName, "::", "/");

  sonar_pub_ = node_handle_->Advertise<sensor_msgs::msgs::Range>(topicName, 10);
}

void SonarPlugin::OnNewScans()
{
  // Get the current simulation time.
[... 105 lines cut ...]
void SonarPlugin::OnNewScans()
{
  // Get the current simulation time.
#if GAZEBO_MAJOR_VERSION >= 9
  common::Time now = world_->SimTime();
#else
  common::Time now = world_->GetSimTime();
#endif

  sonar_message_.set_time_usec(now.Double() * 1e6);
  sonar_message_.set_min_distance(parentSensor_->RangeMin());
  sonar_message_.set_max_distance(parentSensor_->RangeMax());
  sonar_message_.set_current_distance(parentSensor_->Range());

  sonar_message_.set_h_fov(2.0f * atan(parentSensor_->Radius() / parentSensor_->RangeMax()));
  sonar_message_.set_v_fov(2.0f * atan(parentSensor_->Radius() / parentSensor_->RangeMax()));
  sonar_message_.set_allocated_orientation(new gazebo::msgs::Quaternion(orientation_));

  // For now, sending 0 as invalid. Might need a different signal modelling
  // than the lidar plugin
  sonar_message_.set_signal_quality(0);

  sonar_pub_->Publish(sonar_message_);
}
```

Jaeyoung-Lim, 2021-02-16:
@weiyuhe Why would a submodule update push you to an older version? Can you try with a more up to date firmware version?

Jaeyoung-Lim, 2021-02-22:
@weiyuhe Please reopen if this is a valid issue

closed 2021-02-22 by Jaeyoung-Lim