PX4/PX4-SITL_gazebo-classic #710: VTOL crashes after transition to fixed wing

Seiwert, 2021-02-17:
Hello,

I'm encountering an issue with all fixed wing style vehicles when trying to fly. The tailsitter, standard vtol, and plane models all go into a death spin and crash into the ground either after takeoff in the case of the plane or after transitioning to fixed wing mode in the case of the VTOL models. The VTOL models act totally normal up until the point at which they transition. An important distinction here though is I only encounter this issue when attempting to start the simulation with the roslaunch scripts provided. If I launch the vehicle with the _make px4_sitl gazebo_tailsitter_ or similar commands, the vehicle flies and performs as expected. 

I'm at a bit of a loss tracing this. It seems as though both methods of starting the simulation utilize the same SDF file so I'm unsure as to what the culprit could be. Any assistance would be greatly appreciated. Thanks in advance!

Jaeyoung-Lim, 2021-02-22:
@Seiwert If you are only seeing it when running with ROS, there is probably something wrong with your launchfile or how the simulation is being launched.

Could you elaborate on how you are running the sim?

Seiwert, 2021-02-22:
Hi @Jaeyoung-Lim , thanks for the response! So like I kind of mentioned before, when I launch the sim by utilizing the make files (e.g. **make px4_sitl gazebo_tailsitter**) everything works as I would expect. 

However, when I try to use any of the launch files that [are included with PX4 ](https://github.com/PX4/PX4-Autopilot/tree/master/launch) is when I run into issues. I've only been using the mavros ones however (primarily **mavros_posix_sitl.launch**) and the only thing I've changed is the vehicle type argument.

And to test that the system and overall movement is working I've just been using QGroundControl to run simple missions

ghfue, 2021-02-25:
I have seen this issue as well using plane and plane_catapult attempting to takeoff on a mission. I have reverted the PX4 branch to tag v1.11.3 (with the submodules), where it still works. 

This works on master:
```make px4_sitl gazebo_plane_catapult```

while this plane sim spirals and crashes on takeoff on master using the exact same mission but works on PX4 v1.11.3 and older:
```roslaunch px4 mavros_posix_sitl.launch vehicle:=plane_catapult```

To eliminate mavros (which shouldn't affect anything here), it is also unsuccessful taking off using this launch command:
```roslaunch px4 posix_sitl.launch vehicle:=plane_catapult```

Seiwert, 2021-02-26:
@ghfue I tried reverting to that tag as you suggested and it didn't quite work. Once again the make command works (although I'm seeing incredibly different flight behavior) the roslaunch based command does not work. However this time I'm getting a different error. When I start everything with any of the roslaunch files, I get an error saying the accelerometer is uncalibrated. To my understanding it is not possible to calibrate with the SITL and it should've came pre-calibrated.

ghfue, 2021-02-26:
@Seiwert Try doing a clean build and clear your eeprom.

Seiwert, 2021-02-26:
@ghfue That resolved my issue. I had to do some odd combinations of resetting all parameters and settings back to default and then doing a clean build but it eventually resolved the issue. I was unable to get this to work for the tailsitter but the standard vtol worked just fine ( the standard tailsitter will work for my needs). Glad to know this can be resolved by rolling back to a previous release but I'll keep this issue open for now in hopes of digging into the current root problem with current master

ghfue, 2021-03-03:
@Jaeyoung-Lim I think we can change the label from `question` to `bug`.

Jaeyoung-Lim, 2021-03-03:
@ghfue I am not sure, since I think this is coming from how the simulation is being launched from ROS, which is not part of this repo

Seiwert, 2021-03-04:
@Jaeyoung-Lim What repo would you think this issue would be better placed in? I can close this one and open in the appropriate repo and just link back to here for tracking

Jaeyoung-Lim, 2021-03-04:
@Seiwert You can keep this issue open, but the problem is probably in your roslaunch file

Seiwert, 2021-03-06:
@Jaeyoung-Lim I highly doubt it's the launch file as I've not changed the launch file at all and still encounter the issue. I can try to fix the problem but will need to do further research into how the SITL setup even works to begin to understand where the problem could lie

Jaeyoung-Lim, 2021-03-06:
@Seiwert Thanks! Please let us know how your investigation goes

But then what can it be? The only difference between the roslaunch and make target is that one uses a roslaunchfile and one uses a script. The rest of sdf/binaries that are being run are exactly the same

Seiwert, 2021-03-06:
@Jaeyoung-Lim That was my exact confusion. I found the script the make target uses and tried comparing outputs from it to the roslaunch and they seemed identical

Jaeyoung-Lim, 2021-03-19:
@Seiwert Can you check if https://github.com/PX4/PX4-SITL_gazebo/pull/730 fixed your problem?

ghfue, 2021-03-22:
Thanks @Jaeyoung-Lim , this appears to be fixed in my testing with that revert.

Jaeyoung-Lim, 2021-03-22:
@ghfue Thanks, not sure what exactly was the issue but closing this issue for now

closed 2021-03-22 by Jaeyoung-Lim

aoberai, 2022-06-08:
In case a change needs to be made to main, I have the exact same issue as op in the same situation (launch file broken, plain make px4... command works fine for transition to fw- will try the revert but I see that it was not merged to main - maybe something left to be fixed.

Jaeyoung-Lim, 2022-06-08:
@aoberai please post a separate issue with instructions on how to reproduce your problem

aoberai, 2022-06-08:
https://github.com/PX4/PX4-SITL_gazebo/issues/871