PX4/PX4-SITL_gazebo-classic #711: Multiple FC within the same vehicle - GPS linkage issue

LightningPORTO, 2021-02-19:
Hello there, I am trying to simulate multiple Flight controllers within the same vehicle.

I've created 3 IMU's all with different "imuSubTopic" names;
I'm using 3 times the mavlink_interface_plugin, with different ports to simulate 3 different Flight Controllers within the vehicle.

The problem I'm reaching is all the basic sensors except the GPS can be mapped with the "SubTopic" 

For example, I can map the IMU with the mavlink interface using this xacro macro:

<xacro:macro name="MAVLink_plugin_macro2" params="mavlink_udp_port mavlink_tcp_port qgc_udp_port sdk_udp_port number" >
        <plugin name='mavlink_interface${number}' filename='libgazebo_mavlink_interface.so'>
            <robotNamespace></robotNamespace>
            <imuSubTopic>/imu${number}</imuSubTopic>
            <magSubTopic>/mag</magSubTopic>
            <baroSubTopic>/baro</baroSubTopic>
            <mavlink_addr>INADDR_ANY</mavlink_addr>
            <mavlink_udp_port>${mavlink_udp_port}</mavlink_udp_port>
            <mavlink_tcp_port>${mavlink_tcp_port}</mavlink_tcp_port>
            <serialEnabled>false</serialEnabled>
            <serialDevice>/dev/ttyACM0</serialDevice>
            <baudRate>921600</baudRate>
            <qgc_addr>INADDR_ANY</qgc_addr>
            <qgc_udp_port>${qgc_udp_port}</qgc_udp_port>
            <sdk_addr>INADDR_ANY</sdk_addr>
            <sdk_udp_port>${sdk_udp_port}</sdk_udp_port>
            <hil_mode>false</hil_mode>
            <hil_state_level>false</hil_state_level>
            <send_odometry>false</send_odometry>
            <enable_lockstep>true</enable_lockstep>
            <use_tcp>true</use_tcp>
        </plugin>
    </xacro:macro>
    
   It works, but I can't seem to find a way to link the GPS to each FC.
   
   Previously the GPS was using the same logic to link it to the mavlink interface but not anymore, this was introduced in pr-gps_sensor_plugin by @TSC21 
   
   How do I link different GPS models to each Flight Controller?

Jaeyoung-Lim, 2021-02-19:
Xacro is deprecated from this repo and we use jinja templates instead

Which version of the firmware are you running?

LightningPORTO, 2021-02-19:
Hello, thanks for the reply. I know its deprecated but I'm keeping a custom updated version of it.

I'm running the stable version

Jaeyoung-Lim, 2021-02-22:
@LightningPORTO Some syntax that are used in sdf are not supported in xacro, therefore we had to migrate over to jinja templates.

You are correct that the mavlink interface assumes that you only have one gps joint for each vehicle. You would need to add support for having multiple FCU on a single vehicle.

While having multiple sensors (IMU/GPS/Airspeed sensors) are something that we [need](https://github.com/PX4/PX4-SITL_gazebo/pull/676), I have not seen anybody having multiple instances of an FMU on a real vehicle. Therefore I am not sure if this is a valid use case for other people in the community.

LightningPORTO, 2021-02-22:
Thanks for the reply.

Im closing the issue since the reply is satisfactory.

Wish you all the best!

closed 2021-02-22 by LightningPORTO