PX4/PX4-SITL_gazebo-classic #713: Plane airspeed failure on landing

ghfue, 2021-02-25:
When the plane model lands, the following warning is given:
```WARN [airspeed_selector] Airspeed sensor failure detected (1, -1)```.

The airspeed is then considered invalid, causing it to fail the pre-arm checks.
```WARN [PreFlightCheck] Preflight fail: Airspeed invalid.```

This means to refly a mission or to fly multiple missions, the PX4 sim has to be restarted.

Jaeyoung-Lim, 2021-02-26:
@ghfue Do you have a log? wondering what is happening to the airspeed sensor

ghfue, 2021-02-26:
[Log](https://logs.px4.io/plot_app?log=513574fe-35c7-49bb-b5ad-0a12441df5dc)

dayjaby, 2021-09-07:
Got that error on a real drone (with PX4 1.12.1) as well, so might not be related to simulations.

T3chDad, 2021-10-13:
I've had this problem too.  ZOHD Dart XL with the pitot tube on the left wing the airspeed sensor fails on every single landing and requires a power off to reset.  v1.12.3 on MRO Control Zero H7 FC.

T3chDad, 2021-10-13:
> @T3chDad Please post the issue on https://github.com/PX4/PX4-Autopilot/issues with details of your setup and failures

Done!

rrajnidhi, 2022-04-13:
Even i was facing same issue. I guess the reason was bcz of steady airspeed sensor value getting published after landing or when aircraft is not moving. PX4 considers this as a sensor fault. I have resolved it by changing the value of wait time from 1 second to 10second in /src/modules/airspeed_selector/airspeed_selector_main.cpp Try and let me know.

Jaeyoung-Lim, 2022-04-13:
@rrajnidhi Are you sure your firmware includes https://github.com/PX4/PX4-Autopilot/pull/18442 ? (Repeated comments from https://github.com/Auterion/px4-jsbsim-bridge/issues/49)