PX4/PX4-SITL_gazebo-classic #714: PX4-SITL Ignition gazebo

addy1997, 2021-02-26:
Can I use this to simulate drones in the ignition gazebo also?

Jaeyoung-Lim, 2021-02-26:
@addy1997 No, classical gazebo and ignition gazebo are not compatible

addy1997, 2021-02-26:
> @addy1997 No, classical gazebo and ignition gazebo are not compatible

Thanks @Jaeyoung-Lim. Could you guide on how to implement one for ign gazebo?

Jaeyoung-Lim, 2021-02-26:
@addy1997 You would need to re-implement the plugins in this repo into ignition gazebo

Writing plugins for ignition gazebo is well explained in there documentation, e.g.: https://ignitionrobotics.org/api/gazebo/2.10/createsystemplugins.html

closed 2021-02-26 by addy1997

Jaeyoung-Lim, 2021-10-05:
@addy1997 Ignition gazebo is now supported in https://github.com/Auterion/px4-simulation-ignition

addy1997, 2021-10-05:
> @addy1997 Ignition gazebo is now supported in https://github.com/Auterion/px4-simulation-ignition

@Jaeyoung-Lim, that's great! Is it open for contributions? I have worked on some plugins which can be useful for this.

Jaeyoung-Lim, 2021-10-05:
@addy1997 Of course! Please do!