PX4/PX4-SITL_gazebo-classic #721: Couldn't Find Child Link[stereo_camera::link] LoadJoint Failed

canberkgurel, 2021-03-05:
I'm dealing with a very strange error. I'm starting up my quadrotor vehicle with a stereo_camera model. **It used be working fine for me and it suddenly stopped working.** The error I'm seeing is below. Gazebo starts up fine, the vehicle is inserted, but I'm seeing a tiny white box instead of the stereo_camera model.
 

> [Err] [Joint.cc:293] EXCEPTION: Couldn't Find Child Link[stereo_camera::link]
> 
> [Err] [Model.cc:215] LoadJoint Failed
> 
>
> [drone1/vehicle_spawn_AERO_15_X9_15269_7794087773630503741-4] process has finished cleanly
> log file: $HOME/.ros/log/e2c1800c-7dc8-11eb-8e41-fc777466ce75/drone1-vehicle_spawn_AERO_15_X9_15269_7794087773630503741-4*.log

Here is the content of the log file:
```
~/.ros/log/e2c1800c-7dc8-11eb-8e41-fc777466ce75$ more drone1-vehicle_spawn_AERO_15_X9_15269_7794087773630503741-4.log 
[rospy.client][INFO] 2021-03-05 10:42:02,027: init_node, name[/drone1/vehicle_spawn_AERO_15_X9_15269_7794087773630503741], pid[15370]
[xmlrpc][INFO] 2021-03-05 10:42:02,028: XML-RPC server binding to 127.0.0.1:0
[xmlrpc][INFO] 2021-03-05 10:42:02,028: Started XML-RPC server [http://127.0.0.1:44551/]
[rospy.init][INFO] 2021-03-05 10:42:02,028: ROS Slave URI: [http://127.0.0.1:44551/]
[rospy.impl.masterslave][INFO] 2021-03-05 10:42:02,028: _ready: http://127.0.0.1:44551/
[rospy.registration][INFO] 2021-03-05 10:42:02,029: Registering with master node http://127.0.0.1:11311
[xmlrpc][INFO] 2021-03-05 10:42:02,029: xml rpc node: starting XML-RPC server
[rospy.init][INFO] 2021-03-05 10:42:02,129: registered with master
[rospy.rosout][INFO] 2021-03-05 10:42:02,129: initializing /rosout core topic
[rospy.rosout][INFO] 2021-03-05 10:42:02,130: connected to core topic /rosout
[rospy.simtime][INFO] 2021-03-05 10:42:02,131: initializing /clock core topic
[rospy.simtime][INFO] 2021-03-05 10:42:02,132: connected to core topic /clock
[rosout][INFO] 2021-03-05 10:42:02,133: Loading model XML from Gazebo Model Database
[rosout][INFO] 2021-03-05 10:42:02,134: Waiting for service /gazebo/spawn_sdf_model
[rospy.internal][INFO] 2021-03-05 10:42:02,388: topic[/rosout] adding connection to [/rosout], count 0
[rospy.internal][INFO] 2021-03-05 10:42:02,908: topic[/clock] adding connection to [http://127.0.0.1:34031/], count 0
[rosout][INFO] 2021-03-05 10:42:03,039: Calling service /gazebo/spawn_sdf_model
[rosout][INFO] 2021-03-05 10:42:03,558: Spawn status: SpawnModel: Successfully spawned entity
[rospy.core][INFO] 2021-03-05 10:42:03,559: signal_shutdown [atexit]
[rospy.internal][INFO] 2021-03-05 10:42:03,561: topic[/rosout] removing connection to /rosout
[rospy.internal][INFO] 2021-03-05 10:42:03,561: topic[/clock] removing connection to http://127.0.0.1:34031/
[rospy.impl.masterslave][INFO] 2021-03-05 10:42:03,562: atexit
```

The error is related to `<child>stereo_camera::link</child>` line in the stereo_camera.sdf file. It looks good to me, I don't know why it's not working. 

```
<?xml version="1.0" ?>
<sdf version='1.5'>
  <model name='iris_stereo_camera'>
    <include>
      <uri>model://iris</uri>
    </include>

    <include>
      <uri>model://stereo_camera</uri>
      <pose>0.1 0 0 0 0 0</pose>
    </include>
    <joint name="stereo_camera_joint" type="revolute">
      <child>stereo_camera::link</child>
      <parent>iris::base_link</parent>
      <axis>
        <xyz>0 0 1</xyz>
        <limit>
          <upper>0</upper>
          <lower>0</lower>
        </limit>
      </axis>
    </joint>

  </model>
</sdf>
```
### Appendix:

[image]

Environment:

    OS: Ubuntu 18.04.5 LTS
    Gazebo version: Gazebo 9.16.0
    ROS version: Melodic

Jaeyoung-Lim, 2021-03-05:
> I'm dealing with a very strange error. I'm starting up my quadrotor vehicle with a stereo_camera model. It used be working fine for me and it suddenly stopped working. The error I'm seeing is below. [...]

What does this mean? Are you saying that something changed on your system and it stopped working? or did you switch which version of this repo you are using?

addy1997, 2021-03-06:
This is a common problem with the ROS simulator. I can be sure that it has nothing to do with your file. Did you give all the path variables correctly while installation?

canberkgurel, 2021-03-08:
@Jaeyoung-Lim I didn't change anything in my setup. My system is still the same, all of the paths are appended in the bashrc, so my paths are still the same too. I noticed that the CAD model that is supposed to be the stereo-camera appears as "link" in the Models pane in Gazebo. The link name wasn't matching with what was defined in the .sdf files. I don't know why it wasn't matching, I would expect the simulation to look this information up from the .sdf file, no idea where the incorrect information is coming from. But I changed the link names in the sdf files to match with the ones in Gazebo. Doing so solved the `Couldn't Find Child Link[stereo_camera::link] LoadJoint Failed` problem. However, I'm still seeing the following error and the camera model is visualized with the white box in Gazebo.

```
[vehicle_spawn_AERO_15_X9_16300_4744574030477120007-4] process has finished cleanly
log file: /home/eia/.ros/log/671b4364-8018-11eb-a240-fc777466ce75/vehicle_spawn_AERO_15_X9_16300_4744574030477120007-4*.log
```

[image]

Jaeyoung-Lim, 2021-03-08:
@canberkgurel I am still confused. So what changed between your working version and not working version?

canberkgurel, 2021-03-09:
We solved the problem by `sudo rm -rd ~/.gazebo/models/stereo_camera/` after noticing that the changes made in `PX4-Autopilot/Tools/sitl_gazebo/models/stereo_camera/stereo_camera.sdf` don't reflect to the simulation. We assumed that the erroneous model must be coming from somewhere else., some digging revealed the problem that there's a cached one in `~/.gazebo/models` folder.

closed 2021-03-09 by canberkgurel