PX4/PX4-SITL_gazebo-classic #722: Compilation is slow

julianoes, 2021-03-06:
We need to figure out why the build of these libraries takes such a long time. I suspect it has to do with all the dependencies and headers that need to be parsed every. Maybe pre-compiled headers would help?

TSC21, 2021-03-06:
One reason: https://github.com/PX4/PX4-Autopilot/blob/master/platforms/posix/cmake/sitl_target.cmake#L47 - limiting the build cores.

dagar, 2021-03-06:
There are some relatively easy ways to get the time spent in each step with ninja-build. 
 - https://github.com/ninja-build/ninja/issues/1080
 - https://github.com/nico/ninjatracing

I've also been looking for an excuse to try `-ftime-report` with newer clang. 
 - https://aras-p.info/blog/2019/01/12/Investigating-compile-times-and-Clang-ftime-report/

julianoes, 2021-05-10:
This has been improved a bunch by https://github.com/PX4/PX4-Autopilot/commit/52bb99688f4c828aa95e5e337269880cfca94f50, closing for now.

closed 2021-05-10 by julianoes