PX4/PX4-SITL_gazebo-classic #723: Multiple PX4 on interacting UAV

Inonyx, 2021-03-07:
Hi, for a project I'm working on multiple UAV in Gazebo controlled by PX4. The issue I'm running into is that the setup is different from the gazebo_sitl_multiple_run example since the UAV need to be connected using joints and controlled by multiple PX4 ports/vehicles in QGC. Currently I've created a python script to create an .sdf file including the UAV and joints with different ports for mavlink. So far no succes where the second UAV PX4 terminates the first UAV PX4 start.
Question, would it even be possible to use one .sdf with multiple (nested) UAV and joints controlled by multiple PX4 mavlinks? If so would there be an example or tips to get this to work? Thanks.

Jaeyoung-Lim, 2021-03-19:
@Inonyx As long as the mavlink interface plugin in gazebo is configured to have the right TCP port, it should just work: https://github.com/PX4/PX4-SITL_gazebo/blob/1b1afca3440df678167818eaf8e38f301b16bcbd/models/iris/iris.sdf.jinja#L492

`gazebo_sitl_multiple_run.sh` is basically doing exactly that: https://github.com/PX4/PX4-Autopilot/blob/2360c5071308bf861b26d373ef5b60ed155d2ce3/Tools/gazebo_sitl_multiple_run.sh#L38

Jaeyoung-Lim, 2021-03-29:
@Inonyx Please reopen if you feel like this is still an issue

closed 2021-03-29 by Jaeyoung-Lim

yanedwar, 2024-08-13:
@Jaeyoung-Lim Hi, I'm developing something very similar to @Inonyx , where I have a nested model using two drones because they need to be able to interact with each other. I am planning on developing a double sls system, like two drones stabilizing one load, so both drones need to be in the same sdf file, but also need separate PX4 instances. I am using nested models for the two drones, but PX4 is having issues connecting to the drone models, and I keep getting invalid local position estimate errors in QGC and PX4 continuously requests the home position endlessly. Using the multi-vehicle launch files provided is not what I am looking for since they create separate sdf files for each drone with different mavlink ids, and so I cannot create a physical link between the two in gazebo. Could you help me find a work around for this? I am using gazebo-classic and on the main branch of PX4-autopilot.

Here is what my sdf file looks like right now for the nested drone models:

    <model name = "drones">
    
      <model name="drone1">
        <include>
          <uri>model://px4vision</uri>
          <name>drone1</name> 
          <pose>0 0 0 0 0 0</pose> 
        </include>
      </model>

      <model name="drone2">
        <include>
          <uri>model://px4vision2</uri>
          <name>drone2</name> 
          <pose>0 2 0 0 0 0</pose> 
        </include>
      </model>      

    </model>